bluez/bluez #185: bluetoothctl and other bluez_utils not found

honoju, 2021-08-11:
Hello,

I'm using buuildroot 2020.02 to build an embedded linux target. I want to use bluetooth so I included bluez-tools and bluez-utils 5.x but when I try to use on my embedded target the `bluetoothctl` tool, it is not found. I searched in /bin and /usr/bin/ but there is no bluetoothctl, neither are other bluez tools found.

Can anyone help me with this?

Regards,

Vudentz, 2021-08-11:
@honoju not sure what this has to do with BlueZ, or are you saying the tar.gz of a release is not including those binaries? Anyway we don't actually have those tools split into bluez-tools and bluez-utils 5.x, so these must be coming from other sources.

honoju, 2021-08-13:
@Vudentz I'm not getting blueZ from a release but from buildroot so that blueZ can be installed in my linux embedded system. As I read on the net, bluetoothctl is included in the blueZ CLI client package. I installed that package in buildroot but with no success. I am getting the following error :

/bin/sh: bluetoothctl: not found

rawatsaurabh, 2021-08-25:
can you please check in : /usr/libexec/bluetooth/bluetoothd -v

honoju, 2021-09-10:
Hi rawatsaurabh,

I'm really sorry for the late reply!

I tried your line command and it ouputs 5.54 (the blueZ version used I suppose)

When I tried `/usr/libexec/bluetooth/bluetoothd` : first time, the prompt looped infinitely and second time I got this **D-Bus setup failed: Name already in use**

Any extra info or help will be much appreciated!

Vudentz, 2021-09-10:
> Hi rawatsaurabh,
>
> I'm really sorry for the late reply!
>
> I tried your line command and it ouputs 5.54 (the blueZ version used I suppose)
>
> When I tried `/usr/libexec/bluetooth/bluetoothd` : first [...]

That means there is another instance already running, try stop and disable bluetooth.service first.