bluez/bluez #186: Failed to generate UUID array from 53696C6162734465762D59AE66818E58

carminee, 2021-08-13:
Hello,
I downloaded and installed bluez on the ubuntu21.04,the download address is http://www.bluez.org/, and the version is 5.60.
Then I started to use mesh-cfgclient and provision my mesh node but an exception occurred:

**chzfmx@PC:** mesh-cfgclient 
Mesh configuration loaded from /home/chzfmx/.config/meshcfg/config_db.json
Proxy added: org.bluez.mesh.Node1 (/org/bluez/mesh/node8ae08231732845aaa8dff969f1497f32)
Proxy added: org.bluez.mesh.Management1 (/org/bluez/mesh/node8ae08231732845aaa8dff969f1497f32)
Attached with path /org/bluez/mesh/node8ae08231732845aaa8dff969f1497f32
**[mesh-cfgclient]#** discover-unprovisioned on 10
Unprovisioned scan started
Scan result:
	rssi = -64
	UUID = 53696C6162734465762D59AE66818E58
	OOB = 0000
Scan result:
	rssi = -54
	UUID = 53696C6162734465762D59AE66818E58
	OOB = 0000
**[mesh-cfgclient]#** provision 53696C6162734465**76**2D59AE66818E58
Failed to generate UUID array from 53696C6162734465762D59AE66818E58

Finally, I located the function: bool l_uuid_is_valid(const uint8_t uuid[16]) according to the error message in the source code,
/**
 * l_uuid_is_valid:
 * @uuid: UUID to check.
 *
 * Checks whether the given UUID is valid according to RFC 4122.  This function
 * checks that the version field is set properly and the variant of the UUID
 * is set to RFC 4122.
 *
 * Returns: Whether the UUID is valid
 **/
LIB_EXPORT bool l_uuid_is_valid(const uint8_t uuid[16])
{
	unsigned int version;
	unsigned int variant;

	if (!uuid)
		return false;

	variant = uuid[8] >> 6;         _**// uuid[8]=0x76**_
	if (variant != 2)                     _**// if(variant != 1)**_       
		return false;

	version = uuid[6] >> 4;
	if (version < 1 || version > 5)
		return false;

	return true;
}

Micah2015, 2021-11-26:
After do some change to let l_uuid_is_valid return true and make install, I found provision passed, test-mesh can control mesh light.
bluez git: 44789fb8d94839d9c2b847d71faa6d649138b128
ell git: 3940a1014631c0a51ddf2ec2e5f1db634e68cee5

inga-s, 2022-01-27:
Just to clarify:
The UUID check l_uuid_is_valid() in ell is correct in accordance with RFC4122:
i.e., the MSB bits of octet 8 should be set 1 and 0 (0x2)
The issue lies not with ell, but with the entity generating the problematic UUID as it is not compliant.