bluez/bluez #187: [5.60] cannot pair with a2dp sink

gtalusan, 2021-08-13:
Hello, I have a Divoom Timebox Evo.  I've been having trouble pairing it with BlueZ 5.60 (also running BlueAlsa on top).  I read through #157, which seems similar, but I'm unable to pair with an empty cache.

Here's bluetoothd's log with an empty cache:

```
Aug 13 10:50:10 bluetoothd[1818]: Bluetooth daemon 5.60
Aug 13 10:50:10 bluetoothd[1818]: Starting SDP server
Aug 13 10:50:10 bluetoothd[1818]: Bluetooth management interface 1.14 initialized
Aug 13 10:50:12 bluetoothd[1818]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Unknown Command (0x01)
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/a2dp.c:media_server_probe() path /org/bluez/hci0
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/avrcp.c:avrcp_controller_server_probe() path /org/bluez/hci0
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/avrcp.c:avrcp_target_server_probe() path /org/bluez/hci0
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/a2dp.c:a2dp_sink_server_probe() path /org/bluez/hci0
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/a2dp.c:a2dp_source_server_probe() path /org/bluez/hci0
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x17822c0 registered: type:0 codec:0 seid:1
Aug 13 10:50:12 bluetoothd[1818]: Endpoint registered: sender=:1.2 path=/org/bluez/hci0/A2DP/SBC/Source/1
Aug 13 10:50:12 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x1787dd8 registered: type:0 codec:0 seid:2
Aug 13 10:50:12 bluetoothd[1818]: Endpoint registered: sender=:1.2 path=/org/bluez/hci0/A2DP/SBC/Source/2
Aug 13 10:50:19 bluetoothd[1818]: profiles/audio/avrcp.c:avrcp_controller_probe() path /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:19 bluetoothd[1818]: profiles/audio/control.c:control_init() Registered interface org.bluez.MediaControl1 on path /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:19 bluetoothd[1818]: profiles/audio/avrcp.c:avrcp_target_probe() path /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:19 bluetoothd[1818]: profiles/audio/a2dp.c:a2dp_sink_probe() path /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:19 bluetoothd[1818]: profiles/audio/sink.c:sink_init() /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_11_75_58_90_37_72
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_ref() 0x1790200: ref=1
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_ref() 0x1790200: ref=2
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/a2dp.c:setup_ref() 0x1790100: ref=1
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_11_75_58_90_37_72: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
Aug 13 10:50:35 bluetoothd[1818]: profiles/audio/sink.c:sink_connect() stream creation in progress
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 11:75:58:90:37:72: Host is down (112)
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_ref() 0x1790200: ref=3
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:connection_lost() Disconnected from 11:75:58:90:37:72
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err 0xbe919ae4
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_unref() 0x1790200: ref=2
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/a2dp.c:setup_unref() 0x1790100: ref=0
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/a2dp.c:setup_free() 0x1790100
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_unref() 0x1790200: ref=1
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_11_75_58_90_37_72: SINK_STATE_CONNECTING -> SINK_STATE_DISCONNECTED
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/a2dp.c:channel_remove() chan 0x17a1fa8
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_unref() 0x1790200: ref=0
Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_free() 0x1790200
```

Here's the subsequent cache entry:

```
[General]
Name=TimeBox-Evo-audio

[ServiceRecords]
0x00010009=35380900000A00010009090001350319110B0900043510350619010009001935061900190901030900093508350619110D090103090311090001
0x00010002=353B0900000A00010002090001350619110E19110F0900043510350619010009001735061900170901040900093508350619110E090106090311090001
0x00010005=35380900000A00010005090001350319110C0900043510350619010009001735061900170901040900093508350619110E090106090311090002
0x00010003=355A0900000A00010003090001350619111E191203090004350C350319010035051900030803090006350909656E09006A0901000900093508350619111E090105090100251048616E64732D4672656520756E697420090311090010
0x00010004=353B0900000A000100040900013503191101090004350C3503190100350519000308010900053503191002090100250D53657269616C20506F72742031

[Attributes]
0001=2800:0005:00001800-0000-1000-8000-00805f9b34fb
0002=2803:0003:02:00002a00-0000-1000-8000-00805f9b34fb
0004=2803:0005:02:00002a01-0000-1000-8000-00805f9b34fb
0006=2800:0016:0000180a-0000-1000-8000-00805f9b34fb
0007=2803:0008:02:00002a29-0000-1000-8000-00805f9b34fb
0009=2803:000a:02:00002a24-0000-1000-8000-00805f9b34fb
000b=2803:000c:02:00002a25-0000-1000-8000-00805f9b34fb
000d=2803:000e:02:00002a27-0000-1000-8000-00805f9b34fb
000f=2803:0010:02:00002a26-0000-1000-8000-00805f9b34fb
0011=2803:0012:02:00002a28-0000-1000-8000-00805f9b34fb
0013=2803:0014:02:00002a23-0000-1000-8000-00805f9b34fb
0015=2803:0016:02:00002a2a-0000-1000-8000-00805f9b34fb
0017=2800:0022:49535343-fe7d-4ae5-8fa9-9fafd205e455
[... 55 lines cut ...]
Aug 13 10:51:11 bluetoothd[1818]: profiles/audio/a2dp.c:setup_ref() 0x178fef0: ref=2
Aug 13 10:51:11 bluetoothd[1818]: profiles/audio/a2dp.c:setup_unref() 0x178fef0: ref=1
Aug 13 10:51:11 bluetoothd[1818]: profiles/audio/avdtp.c:session_cb() Collision detected
Aug 13 10:51:11 bluetoothd[1818]: profiles/audio/avdtp.c:session_cb() 
Aug 13 10:51:11 bluetoothd[1818]: profiles/audio/avdtp.c:session_cb() No pending request, ignoring message
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:stream_open_timeout() Timed out waiting for peer to open the transport channel
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: CONFIGURED -> ABORTING
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:session_cb() 
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_parse_resp() ABORT request succeeded
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state change failed: Input/output error
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/a2dp.c:abort_cfm() Source 0x17822c0: Abort_Cfm
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/a2dp.c:setup_unref() 0x178fef0: ref=0
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/a2dp.c:setup_free() 0x178fef0
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_unref() 0x178e6b0: ref=1
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: ABORTING -> IDLE
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:handle_unanswered_req() handle_unanswered_req: Abort req, returning
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_unref() 0x178e6b0: ref=0
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_ref() 0x178e6b0: ref=1
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 0
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_ref() 0x178e6b0: ref=2
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/avdtp.c:connection_lost() Disconnected from 11:75:58:90:37:72
Aug 13 10:51:18 bluetoothd[1818]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_11_75_58_90_37_72: SINK_STATE_CONNECTING -> SINK_STATE_DISCONNECTED
```

Pairing with BlueZ 5.55 and empty cache with this device previously worked, however I would intermittently receive "Timed out waiting for peer to open the transport channel", which led me to #157 and 5.60 which seems to have made the problem more reproducible.

Vudentz, 2021-08-13:
`Aug 13 10:50:43 bluetoothd[1818]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 11:75:58:90:37:72: Host is down (112)`

This usually indicates there is a page timeout, if please add the HCI trace using btmon.

closed 2021-09-10 by Vudentz

gtalusan, 2021-10-04:
Hi @Vudentz,

My apologies in the huge delay in responding.  I bisected my way to the commit where my speaker stopped working.

This is the commit:
```
commit 60135a7f83dd87570f3b44d0221a32e169e254c7 (refs/bisect/bad)
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Feb 16 14:55:45 2021 -0800

    avdtp: Remove use of G_PRIORITY_LOW
    
    G_PRIORITY_LOW was used in order to prioritize the AVDTP media transport
    channel over the signalling channel but this has the side effect of
    delaying the dispatching of other conditions such as HUP/NVAL, so now
    that BtIO use G_PRIORITY_HIGH for its watches we no longer need to
    deprioritize session_cb.

diff --git a/profiles/audio/avdtp.c b/profiles/audio/avdtp.c
index 9ddcd6464..088ca58b3 100644
--- a/profiles/audio/avdtp.c
+++ b/profiles/audio/avdtp.c
@@ -2378,19 +2378,10 @@ static void avdtp_connect_cb(GIOChannel *chan, GError *err, gpointer user_data)
                if (session->io_id)
                        g_source_remove(session->io_id);
 
-               /* This watch should be low priority since otherwise the
-                * connect callback might be dispatched before the session
-                * callback if the kernel wakes us up at the same time for
-                * them. This could happen if a headset is very quick in
-                * sending the Start command after connecting the stream
-                * transport channel.
-                */
-               session->io_id = g_io_add_watch_full(chan,
-                                               G_PRIORITY_LOW,
+               session->io_id = g_io_add_watch(chan,
                                                G_IO_IN | G_IO_ERR | G_IO_HUP
                                                | G_IO_NVAL,
-                                               (GIOFunc) session_cb, session,
-                                               NULL);
+                                               (GIOFunc) session_cb, session);
 
                if (session->stream_setup)
                        set_disconnect_timer(session);
```

I've attached attached a btmon trace where connecting/pairing works as expected after using `git bisect`.  And a btmon trace from 5.61 where the speaker doesn't work.

Let me know if more information is needed, or if you'd like me to open up a new issue.  Thanks!

[btmon-divoom-timebox.zip](https://github.com/bluez/bluez/files/7274748/btmon-divoom-timebox.zip)