bluez/bluez #189: Can't control volume from headset using AVRCP

JPabloLassala, 2021-08-19:
I'm using Bluez 5.60 and I still can't control the volume nor the play/pause gestures from my WF-1000XM4 earbuds.
Aside from controls, sound from the headset works flawlessly.

Here's my latest bluetoothd log:
```
Aug 19 09:10:10 archwork bluetoothd[46127]: profiles/audio/avctp.c:avctp_server_socket() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: avrcp-controller: Protocol not supported (93)
Aug 19 09:10:10 archwork bluetoothd[46127]: profiles/audio/avctp.c:avctp_server_socket() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: audio-avrcp-target: Protocol not supported (93)
Aug 19 09:10:10 archwork bluetoothd[46127]: src/profile.c:ext_start_servers() L2CAP server failed for Message Notification: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: src/profile.c:ext_start_servers() L2CAP server failed for Message Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: src/profile.c:ext_start_servers() L2CAP server failed for Phone Book Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: src/profile.c:ext_start_servers() L2CAP server failed for File Transfer: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: src/profile.c:ext_start_servers() L2CAP server failed for Object Push: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSink/sbc
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSource/sbc
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Aug 19 09:10:10 archwork bluetoothd[46127]: Endpoint registered: sender=:1.72 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Aug 19 09:10:15 archwork bluetoothd[46127]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for 14:3F:A6:BC:B0:83: Input/output error
Aug 19 09:10:15 archwork bluetoothd[46127]: /org/bluez/hci0/dev_14_3F_A6_BC_B0_83/sep1/fd0: fd(29) ready
Aug 19 09:24:03 archwork bluetoothd[46127]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for 14:3F:A6:BC:B0:83: Input/output error
Aug 19 09:24:03 archwork bluetoothd[46127]: /org/bluez/hci0/dev_14_3F_A6_BC_B0_83/sep1/fd1: fd(29) ready
```

[pa-info.txt](https://github.com/bluez/bluez/files/7014327/pa-info.txt)

Vudentz, 2021-08-19:
https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/l2cap_sock.c#n293

Most likely you have ERTM disabled in your kernel which is why these profiles are failing.

JPabloLassala, 2021-08-19:
> https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/l2cap_sock.c#n293
> 
> Most likely you have ERTM disabled in your kernel which is why these profiles are failing.

Thank you. I had ERTM disabled because of the Xbox Gamepad driver

I solved it by doing
`echo 0 > /sys/module/bluetooth/parameters/disable_ertm

systemctl restart bluetooth`

(If someone comes in the future looking for this solution)

closed 2021-08-20 by JPabloLassala