bluez/bluez #191: bluetoothctl remove XX:XX:XX:XX:XX:XX does not remove cache file

dlech, 2021-08-23:
While investigating https://github.com/pybricks/support/issues/374, we came across a problematic BlueZ behavior.

We created a custom firmware for a BLE device. This device was connected using BlueZ. Later, the firmware was upgraded where a new characteristic was added so the attribute handles for subsequent attributes were renumbered. After this, it broke client programs the use the BlueZ D-Bus API including the Bleak Python library and WebBluetooth in the Chromium web browser.

Normally `bluetoothctl remove XX:XX:XX:XX:XX:XX` would fix issue like this, but in this case it did not.

After the firmware upgrade, when we connect to a device, we see this in `bluetoothctl`:

```
[CHG] Device 90:84:2B:00:A8:8D Connected: yes
[DEL] Characteristic (Handle 0xd097)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000a
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[DEL] Characteristic (Handle 0xd097)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000c
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
[DEL] Characteristic (Handle 0xd097)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000e
	00002a50-0000-1000-8000-00805f9b34fb
	PnP ID
[DEL] Secondary Service (Handle 0x0073)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[DEL] Descriptor (Handle 0x0e6c)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011/desc0013
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[DEL] Characteristic (Handle 0xd097)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011
	c5f50002-8280-46da-89f4-6d8051e4aeef
	Vendor specific
[DEL] Secondary Service (Handle 0x0073)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010
	c5f50001-8280-46da-89f4-6d8051e4aeef
	Vendor specific
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[NEW] Secondary Service (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[NEW] Characteristic (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000a
	00002a26-0000-1000-8000-00805f9b34fb
	Firmware Revision String
[NEW] Characteristic (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000c
	00002a28-0000-1000-8000-00805f9b34fb
	Software Revision String
[NEW] Characteristic (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0009/char000e
	00002a50-0000-1000-8000-00805f9b34fb
	PnP ID
[NEW] Secondary Service (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010
	c5f50001-8280-46da-89f4-6d8051e4aeef
	Vendor specific
[NEW] Characteristic (Handle 0x8461)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011
	c5f50002-8280-46da-89f4-6d8051e4aeef
	Vendor specific
[NEW] Descriptor (Handle 0xf4f4)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011/desc0013
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device 90:84:2B:00:A8:8D UUIDs: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
[CHG] Device 90:84:2B:00:A8:8D UUIDs: c5f50001-8280-46da-89f4-6d8051e4aeef
[CHG] Device 90:84:2B:00:A8:8D ServicesResolved: yes
```

As you can see, after the device connects, the services/characteristics/descriptors are all removed and then immediately added back. This causes issues with the client programs because they don't expect this removal to happen.

We found that if we manually delete `/var/lib/bluetooth/YY:YY:YY:YY:YY:YY/cache/XX:XX:XX:XX:XX:XX`, then this problem goes away.

We would expect that `bluetoothctl remove XX:XX:XX:XX:XX:XX` should completely forget everything about the device so that next time the device is connected it behaves as if this is the first time BlueZ ever saw the device. However the cache file remaining and having to be manually deleted means BlueZ is not doing this.

Vudentz, 2021-08-23:
@dlech is the system configure Cache=always:

```
[GATT]
# GATT attribute cache.
# Possible values:
# always: Always cache attributes even for devices not paired, this is
# recommended as it is best for interoperability, with more consistent
# reconnection times and enables proper tracking of notifications for all
# devices.
# yes: Only cache attributes of paired devices.
# no: Never cache attributes
# Default: always
#Cache = always
```

If you want the daemon to only remember paired devices service then change the value to Cache=yes, that said doesn't the device generate ServiceChanged or changes its hash so the daemon is able to properly update the attributes? Or the client is no able to handle runtime changes of the services?

Anyway if setting Cache=yes doesn't work perhaps we also need the following change:

```
diff --git a/src/device.c b/src/device.c
index 807106812..998485be7 100644
--- a/src/device.c
+++ b/src/device.c
@@ -4511,6 +4511,7 @@ static void device_remove_stored(struct btd_device *device)
        key_file = g_key_file_new();
        g_key_file_load_from_file(key_file, filename, 0, NULL);
        g_key_file_remove_group(key_file, "ServiceRecords", NULL);
+       g_key_file_remove_group(key_file, "Attributes", NULL);
 
        data = g_key_file_to_data(key_file, &length, NULL);
        if (length > 0) {
```

kakra, 2021-08-24:
If firmware updates to a device could cause such problems, maybe `Cache=always` isn't a good default choice for the configuration file? Xbox controllers have similar issues where a `remove XX:XX:...` is not enough but users need to manually purge the cache.

dlech, 2021-08-24:
The problem still happens with `Cache=yes`.

The device does not generate ServicesChanged because the change occurs when the firmware is upgraded (which requires a reboot of the device), not during runtime. I'm not sure what you mean when you say "or changes its hash". Could you elaborate?

The Bleak client currently doesn't properly handle the InterfacesRemoved D-Bus signal, but the Chromium web browser does have code to handle this. But it doesn't matter since BlueZ is adding back the old values instead of the new values. In the logs I gave, there is only one descriptor, so lets focus on that:

```
[DEL] Descriptor (Handle 0x0e6c)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011/desc0013
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
...
[NEW] Descriptor (Handle 0xf4f4)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0010/char0011/desc0013
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
```

We can see that `service0010/char0011/desc0013` didn't change. This is wrong because the descriptor has actually moved. After manually deleting the cache file, we get the correct new values:

```
[NEW] Descriptor (Handle 0x6d64)
	/org/bluez/hci0/dev_90_84_2B_00_A8_8D/service0014/char0017/desc0019
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
```

The proposed patch does make `bluetoothctl remove XX:XX:XX:XX:XX:XX` work as expected. Wouldn't it be better to just delete the entire file though?

dlech, 2021-08-24:
> I'm not sure what you mean when you say "or changes its hash". Could you elaborate?

Answering my own question, I have found the GATT _Database Hash_ characteristic. I assume that if we add this characteristic, then if we again change the layout we have to be careful that this characteristic does not move?

dlech, 2021-08-24:
> maybe `Cache=always` isn't a good default choice for the configuration file?

The Bluetooth spec says (sorry, copying as text didn't work):

[image]

So it does seem like `Cache=yes` should be the default to comply with the spec.

But it seems like BlueZ isn't respecting this option and still uses the cached values.

dlech, 2021-08-24:
> But it seems like BlueZ isn't respecting this option and still uses the cached values.

After some more testing, I think I did something wrong earlier. Setting `Cache = yes` does fix the problem of getting cached attributes after they have changed on the device. However, when `Cache = yes`, the device can no longer be discovered by WebBluetooth until it is removed from BlueZ, so there is a different issue (I'll open a new issue when I have more info).

Vudentz, 2021-08-24:
> The Bluetooth spec says (sorry, copying as text didn't work):
>
> [image]
>
> So it does seem like `Cache=yes` should be the default to comply with the spec.
>
> But it seems like BlueZ isn't respecting this [...]

We do support reading the Database Hash however, thus why we use always as default, that said the always settings, and other for the same matter, seems to not be affecting the device_remove_stored which is responsible for wiping the device services once the device is removed.

dlech, 2021-08-24:
> We do support reading the Database Hash however, thus why we use always as default,

I read the spec as saying "if the device is not paired and does not have the Database Hash characteristic, then attributes should not be cached". This sounds like the yes option rather than the always option.

Vudentz, 2021-08-24:
> > We do support reading the Database Hash however, thus why we use always as default,
> 
> I read the spec as saying "if the device is not paired and does not have the Database Hash characteristic, then attributes should not be cached". This sounds like the yes option rather than the always option.

In BlueZ we do have a the concept of Trusted device thought, and anyway the spec doesn't say it must not be cached, anyway we depend on the server announcing the changes via ServiceChanged even after it being reflased, this is actually the very reason code like the following exists on BlueZ:

https://github.com/bluez/bluez/blob/master/src/gatt-database.c#L3874

So whenever the database is changed it must send ServiceChanged when reconnecting to device which had subscribed to it:

```
The «Service Changed» characteristic is a control-point attribute (as defined in
[Vol 3] Part F, Section 3.2.6) that **shall** be used to indicate to connected devices
that services have changed (i.e., added, removed or modified).
```

Is would only not be valid if you wipe the bonding as well, but that would break the bonding so the device would have to be removed whenever there is a new firmware.

dlech, 2021-08-24:
> anyway the spec doesn't say it must not be cached,

Technically true, it doesn't say that it can't be cached. But for the condition of "the device is not trusted" and "the device does not support Database Hash characteristic" it does say that the cache is only valid for the duration of the connection. So when a device with these conditions is reconnected, BlueZ is using an invalid cache in this case, which is the source of the problem. 

So I don't understand why BlueZ would require that either "the device is trusted" or "the device supports the Database Hash characteristic" when supporting neither should also be a valid use case.

Vudentz, 2021-08-24:
> > anyway the spec doesn't say it must not be cached,
> 
> Technically true, it doesn't say that it can't be cached. But for the condition of "the device is not trusted" and "the device does not support Database Hash characteristic" it does say that the cache is only valid for the duration of the connection. So when a device with these conditions is reconnected, BlueZ is using an invalid cache in this case, which is the source of the problem.
> 
> So I guess I don't understand why we would want BlueZ to require that either "the device is trusted" or "the device supports the Database Hash characteristic" when supporting neither should also be a valid use case.

Because over the years we had a lot o problem with intermittent connections, so if we had to wipe the cache every single time a connection is interrupted it caused way more problems than always caching the services and then validating like:

https://github.com/bluez/bluez/blob/master/src/shared/gatt-db.c#L708

So we still perform the service discovery on every re-connection but we skip the service that do match the cache, so I suspect this is not working because:

1. The device doesn't emit ServiceChanged
2. There is no Database Hash to compare to
3. Service UUID, service handle and number of attributes haven't changed

The number 3 is condition is not exactly described in the spec but shall work in most cases, except if a service is changed in a way that it breaks backward compatibility, meaning attributes are changed without affecting of the above conditions.

dlech, 2021-08-24:
Thanks for the explanation. We will experiment with the available options.

> Service UUID, service handle and number of attributes haven't changed

In the case I documented in the original comment here, the number of attributes did change, so perhaps something we could look into there too.