bluez/bluez #192: Advertising data does not populate UUIDs when [GATT] Cached = yes

dlech, 2021-08-24:
This is a followup to https://github.com/bluez/bluez/issues/191#issuecomment-90482308.

After adding the following setting to `/etc/bluetooth/main.conf`:

```
[GATT]
Cached = yes
```

BlueZ D-Bus client programs (Bleak Python library and Chromium web browser) were no longer able to discover the device. In both cases, they are filtering on a 128-bit UUID from the advertising data. Previously, when using the default (`Cached = always`), BlueZ would populate the `UUIDs` property with UUIDs from the advertising data. But after this, `UUIDs` is not longer populated and therefore the device cannot be discovered.

Here is a snippet of the output of the D-Bus GetManagedObjects method for the device in question. We can see that `UUIDs` is empty, but `ServiceData` is populated. So BlueZ did get the advertising data, but just didn't add the UUIDs for some reason.

```python
{
	'/org/bluez/hci0/dev_90_84_2B_00_A8_8D': {
		'Address': '90:84:2B:00:A8:8D',
		'AddressType': 'public',
		'Name': 'Pybricks Hub',
		'Alias': 'Pybricks Hub',
		'Paired': False,
		'Trusted': True,
		'Blocked': False,
		'LegacyPairing': False,
		'Connected': False,
		'UUIDs': [],
		'Modalias': 'bluetooth:v0397p0041d0000',
		'Adapter': '/org/bluez/hci0',
		'ServiceData': {
			'00002a50-0000-1000-8000-00805f9b34fb': b'\x01\x97\x03A\x00\x00\x00'
		},
		'ServicesResolved': False
	},
```

dlech, 2021-08-24:
If we remove the device from BlueZ with `bluetoothctl remove XX:XX:XX:XX:XX:XX`, then try to connect it works correctly. But if we disconnect and try to connect again, the device cannot be discovered and we have to remove it from BlueZ again in order to connect.

dlech, 2021-08-24:
Patch has been sent to the mailing list.

Vudentz, 2021-08-24:
The UUIDs is populated with services that have been probed, so Im not sure populating it from the advertising is the right thing to do.

dlech, 2021-08-24:
How else can we get the UUIDs from the advertising data without connecting to the device?

Vudentz, 2021-08-25:
> How else can we get the UUIDs from the advertising data without connecting to the device?

Ive replied to your patch in the mailing list, we should probably do that but I suggested some changes.