bluez/bluez #197: bluez update on Raspbian Buster makes Bluetooth unavailable

mibcat, 2021-08-28:
Hi!
I run into trouble when updating my system with apt upgrade.
After the update the bluetooth MAC address changed to aa:aa:aa:aa:aa:aa:aa and I can isolate the problem to the **bluez** package.

before the update:

```bash
 pi@dc00015:~ $ hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: DC:A6:32:C6:5A:6D  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN 
	RX bytes:8085 acl:74 sco:0 events:382 errors:0
	TX bytes:13188 acl:73 sco:0 commands:285 errors:0
```

updating only bluez:
```bash
pi@dc00015:~ $ sudo apt install --only-upgrade bluez
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  python-colorzero
Use 'sudo apt autoremove' to remove it.
Suggested packages:
  pulseaudio-module-bluetooth
The following packages will be upgraded:
  bluez
1 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.
Need to get 764 kB of archives.
After this operation, 91.1 kB disk space will be freed.
Get:1 https://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian buster/main armhf bluez armhf 5.50-1.2~deb10u2 [764 kB]
Fetched 764 kB in 0s (2,222 kB/s)
Reading changelogs... Done
(Reading database ... 133145 files and directories currently installed.)
Preparing to unpack .../bluez_5.50-1.2~deb10u2_armhf.deb ...
Unpacking bluez (5.50-1.2~deb10u2) over (5.50-1.2~deb10u1+rpt2) ...
Setting up bluez (5.50-1.2~deb10u2) ...
Processing triggers for systemd (241-7~deb10u7+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for dbus (1.12.20-0+deb10u1) ...
```

after the update:
```bash
pi@dc00015:~ $ hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1606 acl:0 sco:0 events:100 errors:0
	TX bytes:3323 acl:0 sco:0 commands:98 errors:0
```

maybe the missing appendix **+rpt2** is the key ???

Callenberg, 2021-08-30:
I can confirm that. Started the day with updating with apt install of 5.50-1.2~deb10u2 version. No luck...

Compiled the latest version 5.61 and tried it out it, still got the AA:AA:AA... mac address after boot.
Also tried with the restart of bluetooth.service and the correct mac address got in place. For me, unfortunately, bluetoothctl do not behave as I am used to.  It is better now but still there are problems for me with pairing and connecting and disconnects. 

Did things work better for you with the 5.50-1.2~deb10u1+rpt2 version, did you try that?

 Kind regards, 
 Callenberg

mibcat, 2021-08-31:
Yes, I went back to version 5.50-1.2~deb10u1+rpt2 and bluetooth is working as expected again.

After that I fixed the version with
`apt-mark bluez`

Callenberg, 2021-08-31:
Thank you, I downgraded to 5.50-1.2~deb10u1+rpt2 and it works better. I can actually connect my gear, but only from the command line - for example connecting from my phone does not work. It get paired and then nothing more   ...ah well, I have to live with that for the time being. Thanks again!

Vudentz, 2021-10-05:
It doesn't looks like bluetoothd ever had set the Public Address of the adapter, or are we talking about LE static address here? It would be great if you guys can send the HCI trace to see when the address is changed when it works to compare to when it doesn't.

Callenberg, 2021-10-06:
Hello,
Thank you Vudentz for reaching out. I appreciate that a lot.

The problem is that I can not connect an Android or an iPhone to the Raspberry Pi 4 anymore. I can get them paired, but not connected.

This is how Bluetoothctl look like for me. 
1. First there is the "Show" command to just show you how the controller is set up.
2. Then I try to pair.  The Bluetoothctl confirms the pairing and says it is connected...   for about three seconds and then "Connected: no".
3. On the smart phone side it says "Not connected" (but paired). If I try to connect from the phone Bluetoothctl connects and disconnects immediately. No connection is established.
---
1) show command:
[bluetooth]# show
Controller E4:5F:01:33:AA:41 (public)
        Name: Player
        Alias: Player
        Class: 0x002c041c
        Powered: yes
        Discoverable: yes
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no
2) Pair attempt from the smart phone:
[NEW] Device 8C:DE:E6:25:C5:8C Galaxy S21 Ultra 5G
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C Modalias: bluetooth:v0075p0100d0201
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001112-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001115-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device 8C:DE:E6:25:C5:8C ServicesResolved: yes
[CHG] Device 8C:DE:E6:25:C5:8C Paired: yes
[CHG] Device 8C:DE:E6:25:C5:8C ServicesResolved: no
[CHG] Device 8C:DE:E6:25:C5:8C Connected: no
2) continued - this ends  the pairing sequence, the devices are paired here
3) the smart phone is not connected and I make two connect attempts
[CHG] Device 8C:DE:E6:25:C5:8C Connected: yes
[CHG] Device 8C:DE:E6:25:C5:8C Connected: no
3) continued: after a few seconds "Connected: no", next attempt below - yields the same result...
[CHG] Device 8C:DE:E6:25:C5:8C Connected: yes
[CHG] Device 8C:DE:E6:25:C5:8C Connected: no
---

**Workaround**:
So what I do is that I use some code that listens to Bluetoothctl events and if a "Paired: yes" is discovered the program immediately does a bluetoothctl connect.

Like this:
[bluetooth]# paired-devices
Device 8C:DE:E6:25:C5:8C Galaxy S21 Ultra 5G
[bluetooth]# connect 8C:DE:E6:25:C5:8C
Attempting to connect to 8C:DE:E6:25:C5:8C
[CHG] Device 8C:DE:E6:25:C5:8C Connected: yes
Connection successful
[CHG] Device 8C:DE:E6:25:C5:8C ServicesResolved: yes
[Galaxy S21 Ultra 5G]#

This is the version I use:
`apt list bluez -a
Listing... Done
bluez/testing,now 5.50-1.2~deb10u2+rpt1 armhf [installed,automatic]
bluez/stable 5.50-1.2~deb10u1 armhf
`

Kernel:
```
 uname -r
5.10.60-v7l+
```

Callenberg, 2021-10-06:
Vudentz, I guess that you are more interested in the public address failure.

I am beyond that now. It disappeared after i installed `5.50-1.2~deb10u2+rpt1 armhf`, but the `bluez/stable 5.50-1.2~deb10u1 armhf` has that problem with the public address, at least for me.

closed 2025-08-25 by github-actions[bot]