bluez/bluez #198: Does ./profiles/input/sixaxis.h only support sony ps3 and SHANWAN ps3  controller?

Cooky-12, 2021-08-30:
```c
{
	.name = "Sony PLAYSTATION(R)3 Controller",
	.source = 0x0002,
	.vid = 0x054c,
	.pid = 0x0268,
	.version = 0x0000,
	.type = CABLE_PAIRING_SIXAXIS,
},
{
	.name = "SHANWAN PS3 GamePad",
	.source = 0x0002,
	.vid = 0x054c,
	.pid = 0x0268,
	.version = 0x0000,
	.type = CABLE_PAIRING_SIXAXIS,
},
```
If a fake PS3 controller has a different hid name, sixaxis plugin will not work.
I have made a patch to treat my fake PS3 controller as SHANWAN, sixaxis plugin worked and it works in bluetooth mode.

kakra, 2021-09-05:
Detailing on https://github.com/bluez/bluez/pull/200, What is the different HID name?

Cooky-12, 2021-09-05:
> Detailing on #200, what is the different HID name?

It's just "PS3 controller".

kakra, 2021-09-05:
It probably has a different MAC OUI than the original controllers?

GramThanos, 2023-10-03:
I have the [VOYEE - HY-2208](https://www.amazon.com/VOYEE-Controller-Wireless-Dualshock-Playstation/dp/B07ZRSYWNB/), which is identified as `GUO HUA PS3 GamePad`.  I patched the driver to support it on commit e2749ccd5cd90f0184ca5023c03ebeb1e6798ae6. I tested it and it works quite well.

Related issues/discussions:
https://github.com/batocera-linux/batocera.linux/issues/6774
https://gitlab.com/recalbox/recalbox/-/issues/2211
https://aur.archlinux.org/packages/bluez-plugins-ps3#comment-895838

---

I also have 2 more unbranded PS3 controllers, but I failed to successfully connect them.

The first one is identified as `Gasia Co.,Ltd PS(R) Gamepad` when connected over USB but is identified as `PLAYSTATION(R)3 Controller` when connected over Bluetooth. After pairing it, if you remove the USB it seems that it connects successfully over Bluetooth but is never assigned a number and turns off.

The second one is identified as `SHENGHIC 2009/0708ZXW-V1Inc. PLAYSTATION(R)3Conteroller` (no spelling errors from my side). I have problems setting the master Bluetooth on this one and I never got it to connect over Bluetooth.

Cooky-12, 2023-10-05:
> I have the [VOYEE - HY-2208](https://www.amazon.com/VOYEE-Controller-Wireless-Dualshock-Playstation/dp/B07ZRSYWNB/), which is identified as `GUO HUA PS3 GamePad`. I patched the driver to support it on commit [e2749cc](https://github.com/bluez/bluez/commit/e2749ccd5cd90f0184ca5023c03ebeb1e6798ae6). I tested it and it works quite well.

There are all sorts of fake PS3 controllers with different HID names. Can't add them all to sixaxis.h, the easiest way is to just treat any other fakes as Shanwan controllers.

GramThanos, 2023-10-05:
> There are all sorts of fake PS3 controllers with different HID names. Can't add them all to sixaxis.h, the easiest way is to just treat any other fakes as Shanwan controllers.

I think the problem is that even if we allow all devices based on their VID and PID to be treated as a PS3 controllers, it doesn't mean that they will work. At some point there has to be a testing or debugging to fix problems/incompatibilities. Furthermore, even if the Bluez is patched, then the controller may face problems that has to be addressed on [hid-sony.c ](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-sony.c), even the Shanwan controllers have exceptions there.

I guess HID name matching was added for a reason, but I am not sure what that reason is. `hid-sony.c` also have HID matching to resolve incompatibilities.
Maybe we can do a simpler matching like checking for keywords like `PS3` or `PS(R)` or `PLAYSTATION(R)3` inside the name?
Like:
```c
	for (i = 0; i < G_N_ELEMENTS(devices); i++) {
		if (devices[i].vid != vid)
			continue;
		if (devices[i].pid != pid)
			continue;
		if (name && strcmp(name, devices[i].name))
			continue;
		return &devices[i];
	}

	// General rule for unofficial PS3 Controllers
	if (
		vid == 0x054c && vid == 0x054c &&
		(
			strstr(name, "PS3") != NULL ||
			strstr(name, "PS(R)") != NULL ||
			strstr(name, "PLAYSTATION(R)3") != NULL
		)
	) {
			return devices[1];
		}
	}
```
Or even improve it to keep its HID name.