bluez/bluez #199: Please add MTU property

weliem, 2021-08-30:
I working with devices where 'messages' are sent over BLE. One message can be up to +/-60 bytes long so I need to spread them over several writes, depending on the MTU size. I specifically don't want a long write to happen since my device doesn't support it. 

However, in Bluez I cannot retrieve the current 'write-MTU'. But I can easily get that in Android and iOS!

Can we have a MTU property in Bluez? Please?

Vudentz, 2021-09-01:
We have been discussing a few options on this, perhaps we should actually switch to something similar to a write syscall where the number of bytes written is returned in the response, that said this sort of solution is not very performant since D-Bus round-trips are not that fast usually I recommend to use AcquireWrite and then perform the writes locally via file descriptor.

weliem, 2021-09-02:
I have seen the AcquireWrite method but in my case the characteristic does not have the required 'write-without-response'. It is a 'with-response' characteristic. So I can't use it. 

Besides that, the normal WriteValue method is working fine for me and doing it via a file descriptor brings in additional complexity I am not waiting for. On top of that, the Java DBUS binding library I am using seems to have issues with file descriptors.

Why not simply add a MTU property? It seems to me that it would be the simplest solution that would solve the issue. MTU is a basic concept in BLE so I don't understand why it is not made available.

Simplicity is always preferred!

Vudentz, 2021-09-03:
@weliem for starter the MTU can change, also with EATT each and every channel can have different MTUs and on top of that different tx and rx MTUs, can you perhaps explain how the peripheral is behaving when we write, because write request operation don't have a offset if should interprete as starting over from 0, anyway we could perhaps just have a force flag to use multiple writes instead if it doesn't fit on a single MTU.

aknabi, 2021-09-03:
I would say +1 to having a straight-forward, complex-free "getMTU" (ala iOS/Android). From experience the "normal" case is to get/request the MTU size then roll. Though in theory there could be  a benefit to having a flag to force multiple writes under the hood (reducing the complexity for the developer) I have a case where we're using the segment feature based on the User Data Service where we'd need the actual MTU (involved in BT standards for medical). But glad to see this issue raised and hopefully sorted soon!

dlech, 2021-09-03:
> for starter the MTU can change

Wouldn't a D-Bus property change address this?

> perhaps explain how the peripheral is behaving when we write, because write request operation don't have a offset if should interprete as starting over from 0

1. The nRF UART service treats data as a stream. There is no "offset" in the stream. To maximize throughput, we need to know what is the most bytes that can be used in a Write Without Response message.
2. LEGO Powered Up Smart Hubs have a firmware update protocol that uses Write Without Response. Same need to maximize throughput. This is a mass-produced device so the protocol can't be changed.

---

I sent a similar reply on the mailing list, but paraphrasing it here so it is all in one place:

Basically, what we are asking for is primarily for cross-platform compatibility. We want the same function that is already on other OSes. A device-level property/method/signal/whatever to get the MTU or the maximum size that can be used with Write Without Response.

- https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattsession.maxpdusize?view=winrt-19041
- https://developer.apple.com/documentation/corebluetooth/cbperipheral/1620312-maximumwritevaluelength
- https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback.html#onMtuChanged(android.bluetooth.BluetoothGatt,%20int,%20int)

---

>  also with EATT each and every channel can have different MTUs and on top of that different tx and rx MTUs

If there is need to access this information, I think it is independent of what we are asking for and can probably be considered separately.

weliem, 2021-09-06:
Indeed, the needs is not to know everything. We just need to know how many bytes we can write. Our protocol takes care of chunking bytes and adding a sequence number, similar to the DFU protocol. So we are not looking for some clever functionality. We just need to know the 'write MTU'....

I know the MTU can change, so a property would be perfect. That way we will get a propertiesChanged signal when the value changes...

Vudentz, 2021-09-10:
@weliem @dlech There is another aspect that don't seem to be covered on these APIs and is the reason why Acquire* was introduced, these attributes that act as control points, serial port, channel, etc, they need exclusive access to its handle otherwise if there are multiple application writing to the same attribute it would stop working properly in most cases. This is quite easy to exploit since one can watch/subscribe for notification and trigger writes independently which can potentially brick the peripherals in case of firmware update, so I wouldn't be surprised if the next addition to the other OS is a way to lock the attributes just like we did, so while Im fine with addition an MTU on a per attribute basis, Im pretty sure user of it will eventually have to move to the likes of AcquireWrite or similar interface on other OS with proper locking.

dlech, 2021-09-10:
We seem to be an in impasse then. While I appreciate the various suggestions of the ideal way of doing things, none of them really solve current developer needs for existing hardware and software.

weliem, 2021-09-10:
@Vudentz , we are working on software that is basically a gateway that connects to over a 100 different medical devices, most of which are off-the-shelve and made by various manufactures. We have no control over these devices nor can we change their BLE firmware. Some of your suggestions are definitely valid in general, but since we have no control over the peripheral, it doesn't help. We do have full control over the gateway and our application is the only application that does BLE. So no need for locking...

Fact of life is that without knowing the MTU for writing, we simply **cannot implement support** for some of these devices....

Emill, 2021-09-20:
Even though the AcquireWrite API is a nice feature and should exist on all platforms, this is not really the case how it looks right now. Anyway, you should not be forced to use the AcquireWrite API. There are plenty of devices where a "locking" mechanism is not needed. Either because different writes are uncorrelated, or because the application layer protocol has some way of distinguishing writes from different "app clients". See for example how in https://github.com/50ButtonsEach/flic2-documentation/wiki/Flic-2-Protocol-Specification where the first byte is a client identifier.

closed 2021-09-29 by BluezTestBot

weliem, 2021-09-30:
Thank you very much!