bluez/bluez #201: After a restart, it is no longer possible to connect to bonded-devices (more than 14 bonded devices)

NPlaten, 2021-09-08:
Hi there,
we have been experiencing an issue in bonded devices related to a reboot.

in detail:
Many Bluetooth devices will be connected and paired / bonded. The pairing works and a connection can then be established to all devices. As soon as **more than 14** devices are paired and the system is **restarted** (or a Bluetooth dongle is removed and reinserted), no connection can be established to any device.
My investigation has shown that the LTKs are no longer there (see log).
If the list of paid devices is reduced to <= 14 again and the Bluetooth service is restarted, it is possible to re-establish a connection to the remaining devices.

In order to reproduce the bug, the following is necessary:

- [ ] Pair 15 devices ``` bluetoothctl: connect xx:xx:xx:xx:xx:xx & pair ```
- [ ] reboot the system or reconnect the bluetooth dongle  ``` reboot ```
- [ ] try to connect (not possible, new pairing is required) ``` bluetoothctl: connect xx:xx:xx:xx:xx:xx ```
- [ ] delete from a paired device so that only 14 are in the list ``` bluetoothctl: remove yy:yy:yy:yy:yy:yy ```
- [ ] bluetooth service restart ``` service bluetooth restart ```
- [ ] connect possible again ``` bluetoothctl: connect xx:xx:xx:xx:xx:xx ```

Here are a few more logs and information:
Log of btmon, if connect is not possible:
```
Bluetooth monitor ver 5.53
= Note: Linux version 5.11.0-27-generic (x86_64)                       0.137088
= Note: Bluetooth subsystem version 2.22                               0.137091
@ MGMT Open: bluetoothd (privileged) version 1.19             {0x0001} 0.137092
@ MGMT Open: btmon (privileged) version 1.19                  {0x0002} 0.137295
= New Index: 00:00:00:00:00:00 (Primary,USB,hci0)               [hci0] 7.192592

...

@ MGMT Event: Command Complete (0x0001) plen 6         {0x0001} [hci0] 7.317799
      Add UUID (0x0010) plen 3
        Status: Success (0x00)
        Class: 0x000000
          Major class: Miscellaneous
          Minor class: 0x00
@ MGMT Command: Unblock Device (0x0027) plen 7         {0x0001} [hci0] 7.317825
        BR/EDR Address: 00:00:00:00:00:00 (OUI 00-00-00)
@ MGMT Event: Device Unblocked (0x0015) plen 7         {0x0002} [hci0] 7.317828
        BR/EDR Address: 00:00:00:00:00:00 (OUI 00-00-00)
@ MGMT Event: Command Complete (0x0001) plen 10        {0x0001} [hci0] 7.317831
      Unblock Device (0x0027) plen 7
        Status: Success (0x00)
        BR/EDR Address: 00:00:00:00:00:00 (OUI 00-00-00)
= bluetoothd: Failed to load LTKs for hci0: Failed (0x03)       [hci0] 7.319059
@ MGMT Command: Load Identity Res.. (0x0030) plen 347  {0x0001} [hci0] 7.317890

...

> ACL Data RX: Handle 0 flags 0x02 dlen 6                 #113 [hci0] 16.027478
      SMP: Pairing Failed (0x05) len 1
        Reason: Pairing not supported (0x05)
@ MGMT Event: Authentication Failed (0x0011) plen 8   {0x0002} [hci0] 16.027659
        LE Address: E3:50:E1:8C:BF:D5 (Static)
        Status: Authentication Failed (0x05)
@ MGMT Event: Authentication Failed (0x0011) plen 8   {0x0001} [hci0] 16.027659
        LE Address: E3:50:E1:8C:BF:D5 (Static)
        Status: Authentication Failed (0x05)
< HCI Command: Disconnect (0x01|0x0006) plen 3            #114 [hci0] 16.027730
        Handle: 0
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                 #115 [hci0] 16.030424
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4            #116 [hci0] 16.127382
        Status: Success (0x00)
        Handle: 0
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8     {0x0002} [hci0] 16.127484
        LE Address: E3:50:E1:8C:BF:D5 (Static)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8     {0x0001} [hci0] 16.127484
        LE Address: E3:50:E1:8C:BF:D5 (Static)
        Reason: Connection terminated by local host (0x02)
```

If the full log is required, just ask, but I think these are the most important points.

List of paired-devices:
```
[bluetooth]# paired-devices
Device E3:50:E1:8C:BF:CD
Device E3:50:E1:8C:BF:CB
Device E3:50:E1:8C:BF:C6
Device E3:50:E1:8C:BF:CF
Device E3:50:E1:8C:BF:CA
Device E3:50:E1:8C:BF:C9
Device E3:50:E1:8C:BF:C8
Device E3:50:E1:8C:BF:CE
Device E3:50:E1:8C:BF:D1
Device E3:50:E1:8C:BF:D4
Device E3:50:E1:8C:BF:D0
Device E3:50:E1:8C:BF:D3
Device E3:50:E1:8C:BF:D2
Device E3:50:E1:8C:BF:C7
Device E3:50:E1:8C:BF:CC
```

Trying to connect (To make troubleshooting easier, the Bluetooth device was configured in a way that it only allows one connection/pair.):
```
[bluetooth]# connect E3:50:E1:8C:BF:D4
Attempting to connect to E3:50:E1:8C:BF:D4
[CHG] Device E3:50:E1:8C:BF:D4 Connected: yes
Connection successful
[CHG] Device E3:50:E1:8C:BF:D4 Connected: no
```

Test on System 1 (PC): 
- Linux ubuntu 5.11.0-34-generic
- bluez 5.53
- dongles  (chips nRF52840 and RTL8822CU),

Test on System 2 (Raspberry Pi 4): 
- Linux raspberrypi 5.10.17-v7l+
- bluez 5.50 and bluez 5.61

I hope that is now all the information that is needed.

Vudentz, 2021-09-08:
Is there any error being printed on dmesg?

NPlaten, 2021-09-09:
> Is there any error being printed on dmesg?

No there is only the normal log for a new device

```
usb 2-2.1: new full-speed USB device number 8 using uhci_hcd
usb 2-2.1: New USB device found, idVendor=2fe3, idProduct=000b, bcdDevice= 2.04
usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 2-2.1: Product: USB-DEV
usb 2-2.1: Manufacturer: ZEPHYR
usb 2-2.1: SerialNumber: 60E0C85A20FC4A64
````

Vudentz, 2021-09-09:
@NPlaten weird I don't see any code returning status 0x03 for Load LTK, can you attach the full logs when loading LTKs happens.

Vudentz, 2021-09-09:
@NPlaten looks like it is not the kernel after all:

https://patchwork.kernel.org/project/bluetooth/list/?series=544659

NPlaten, 2021-09-10:
> @NPlaten weird I don't see any code returning status 0x03 for Load LTK, can you attach the full logs when loading LTKs happens.

here is the entire btmon output from the dongle's init process:
[load_ltks_error.log](https://github.com/bluez/bluez/files/7141486/load_ltks_error.log)
[hcitrace.zip](https://github.com/bluez/bluez/files/7141530/hcitrace.zip)
And here the log with debug information:
[journalctl.log](https://github.com/bluez/bluez/files/7141568/journalctl.log)

An error occurs with this dongle when transferring the name, but this was not the case with the other dongle, so no reason for the incorrect transfer of the LTKs

> @NPlaten looks like it is not the kernel after all:
> 
> https://patchwork.kernel.org/project/bluetooth/list/?series=544659

Ok, that means more than 14 keys are provided, but cannot be used.

What is interesting for me is that working with more than 14 devices works well until the dongle is restarted.
After the restart it works as long as there are only 14 participants in "/var/lib/bluetooth/\<adapter\>/".

To me it sounds like something is going wrong when too many LTKs are transmitted (during initialization). Unfortunately, I couldn't find anything about how exactly the LTKs are transferred, but I suspect differently than with the pairing process, because more than 14 work there.

Vudentz, 2021-09-10:
I suspect we are not even able to send the command for some reason, try setting MGMT_DEBUG=1 when lauching bluetoothd:

```
bluetoothd[2971852]: mgmt: [0x0000] command 0x0013
bluetoothd[2971852]: mgmt: < 13 00 00 00 26 00 01 00 29 32 93 cf 09 98 01 03  ....&...)2......
bluetoothd[2971852]: mgmt:   00 10 00 00 00 00 00 00 00 00 00 00 b2 d1 68 64  ..............hd
bluetoothd[2971852]: mgmt:   2d de 3d 40 6d 82 f4 b5 2d 82 ec 97              -.=@m...-...    
bluetoothd[2971852]: mgmt: > 01 00 00 00 03 00 13 00 00                       .........       
bluetoothd[2971852]: mgmt: [0x0000] command 0x0013 complete: 0x00
bluetoothd[2971852]: src/adapter.c:load_ltks_complete() LTKs loaded for hci0
```

NPlaten, 2021-09-14:
Here is the start up log with the debug mode for the mgmt.
with error (with 15 devices)::
[bluetoothd.log](https://github.com/bluez/bluez/files/7159833/bluetoothd.log)
working (only 14 devices):
[bluetoothd-working.log](https://github.com/bluez/bluez/files/7159926/bluetoothd-working.log)

Vudentz, 2021-09-14:
```
       EINVAL The sum of the iov_len values overflows an ssize_t value.

       EINVAL The vector count, iovcnt, is less than zero or greater than the permitted maximum.

```

Ive tried emulating this by creating dummy devices in the storage, with 17 it works but it looks like you have actually 30 LTKs so I bumped to 33 and then I managed to reproduce:

```
bluetoothd[3741945]: src/adapter.c:load_ltks() hci0 keys 33
bluetoothd[3741945]: Failed to load LTKs for hci0: Failed (0x03)

```

Vudentz, 2021-09-14:
It is actually a kernel limitation, currently it support up to HCI_MAX_FRAME_SIZE (1024 + 4) which limits the amount of data that we can send in one MGMT packet, I will be fixing that to enable the userspace to change that.

NPlaten, 2021-09-15:
Thank you, so we now know where exactly it comes from and can hopefully look forward to a solution soon. Will it then be a bluez solution that takes into account that the maximum packet length is limited?

I am still trying to find out why we have 30 keys even though there are only 15 devices. Thanks for the hint.

Vudentz, 2021-09-15:
@NPlaten don't think there is much we can do since it is in the kernel, we could perhaps attempt to truncate the message so it loads as many LTKs as possible but we would have to discard the ones that didn't fit.

@holtmann do you have any other ideas?

closed 2021-09-21 by BluezTestBot

mrhm0000, 2021-11-09:
Hi @Vudentz.

I had a question regarding the fix provided for this issue. Since we have been using old kernel (5.4.70), so we had to backport the commit  [[Bluetooth: hci_sock: Add support for BT_{SND,RCV}BUF]](https://github.com/BluezTestBot/bluetooth-next/pull/506/commits/82c139f94813db8f10760c6c68da4a13a8cee4e3) for 5.4.70. The intent is to integrate the core logic of the fix and avoid the above issue. We also used BlueZ5-5.62. Unfortunately it seems not to resolve the issue, so was wondering if the backport is correct or whether it depends (maybe impacted) by not having the in-between changes from the BT module on our kernel? If you can kindly provide us any hint - would be appreciated.

Let me know if there's any further question.

Below is the dump of the mentioned backported fix for 5.4.70. The only difference on this backport is how contents are copied from userspace. On 5.4.70, sockptr helper wasn't available. Otherwise, I believe it keeps the core logic intact.

```
diff --git a/net/bluetooth/hci_sock.c b/net/bluetooth/hci_sock.c
index 8159b344d..782ed9134 100644
--- a/net/bluetooth/hci_sock.c
+++ b/net/bluetooth/hci_sock.c
@@ -57,6 +57,7 @@ struct hci_pinfo {
 	unsigned long     flags;
 	__u32             cookie;
 	char              comm[TASK_COMM_LEN];
+	__u16		  mtu;
 };
 
 void hci_sock_set_flag(struct sock *sk, int nr)
@@ -1334,6 +1335,10 @@ static int hci_sock_bind(struct socket *sock, struct sockaddr *addr,
 		break;
 	}
 
+	/* Default MTU to HCI_MAX_FRAME_SIZE if not set */
+	if (!hci_pi(sk)->mtu)
+		hci_pi(sk)->mtu = HCI_MAX_FRAME_SIZE;
+
 	sk->sk_state = BT_BOUND;
 
 done:
@@ -1691,7 +1696,7 @@ static int hci_sock_sendmsg(struct socket *sock, struct msghdr *msg,
 			       MSG_CMSG_COMPAT))
 		return -EINVAL;
 
-	if (len < 4 || len > HCI_MAX_FRAME_SIZE)
+	if (len < 4 || len > hci_pi(sk)->mtu)
 		return -EINVAL;
 
 	lock_sock(sk);
@@ -1813,7 +1818,7 @@ static int hci_sock_sendmsg(struct socket *sock, struct msghdr *msg,
 	goto done;
 }
 
-static int hci_sock_setsockopt(struct socket *sock, int level, int optname,
+static int hci_sock_setsockopt_old(struct socket *sock, int level, int optname,
 			       char __user *optval, unsigned int len)
 {
 	struct hci_ufilter uf = { .opcode = 0 };
@@ -1822,9 +1827,6 @@ static int hci_sock_setsockopt(struct socket *sock, int level, int optname,
 
 	BT_DBG("sk %p, opt %d", sk, optname);
 
-	if (level != SOL_HCI)
-		return -ENOPROTOOPT;
-
 	lock_sock(sk);
 
 	if (hci_pi(sk)->channel != HCI_CHANNEL_RAW) {
@@ -1899,7 +1901,55 @@ static int hci_sock_setsockopt(struct socket *sock, int level, int optname,
 	return err;
 }
 
-static int hci_sock_getsockopt(struct socket *sock, int level, int optname,
+static int hci_sock_setsockopt(struct socket *sock, int level, int optname,
+                               char __user *optval, unsigned int len)
+{
+        struct sock *sk = sock->sk;
+        int err = 0, opt = 0;
+
+        BT_DBG("sk %p, opt %d", sk, optname);
+
+        if (level == SOL_HCI)
+                return hci_sock_setsockopt_old(sock, level, optname, optval,
+                                               len);
+
+        if (level != SOL_BLUETOOTH)
+                return -ENOPROTOOPT;
+
+        lock_sock(sk);
+
+        switch (optname) {
+        case BT_SNDMTU:
+        case BT_RCVMTU:
+                switch (hci_pi(sk)->channel) {
+                /* Don't allow changing MTU for channels that are meant for HCI
+                 * traffic only.
+                 */
+                case HCI_CHANNEL_RAW:
+                case HCI_CHANNEL_USER:
+                        err = -ENOPROTOOPT;
+                        goto done;
+                }
+
+                if (copy_from_user(&opt, optval, sizeof(u16))) {
+                        err = -EFAULT;
+                        break;
+                }
+
+                hci_pi(sk)->mtu = opt;
+                break;
+
+        default:
+                err = -ENOPROTOOPT;
+                break;
+        }
+
+done:
+        release_sock(sk);
+        return err;
+}
+
+static int hci_sock_getsockopt_old(struct socket *sock, int level, int optname,
 			       char __user *optval, int __user *optlen)
 {
 	struct hci_ufilter uf;
@@ -1908,9 +1958,6 @@ static int hci_sock_getsockopt(struct socket *sock, int level, int optname,
 
 	BT_DBG("sk %p, opt %d", sk, optname);
 
-	if (level != SOL_HCI)
-		return -ENOPROTOOPT;
-
 	if (get_user(len, optlen))
 		return -EFAULT;
 
@@ -1968,6 +2015,39 @@ static int hci_sock_getsockopt(struct socket *sock, int level, int optname,
 	return err;
 }
 
+static int hci_sock_getsockopt(struct socket *sock, int level, int optname,
+                               char __user *optval, int __user *optlen)
+{
+        struct sock *sk = sock->sk;
+        int err = 0;
+
+        BT_DBG("sk %p, opt %d", sk, optname);
+
+        if (level == SOL_HCI)
+                return hci_sock_getsockopt_old(sock, level, optname, optval,
+                                               optlen);
+
+        if (level != SOL_BLUETOOTH)
+                return -ENOPROTOOPT;
+
+        lock_sock(sk);
+
+        switch (optname) {
+        case BT_SNDMTU:
+        case BT_RCVMTU:
+                if (put_user(hci_pi(sk)->mtu, (u16 __user *)optval))
+                        err = -EFAULT;
+                break;
+
+        default:
+                err = -ENOPROTOOPT;
+                break;
+        }
+
+        release_sock(sk);
+        return err;
+}
+
 static const struct proto_ops hci_sock_ops = {
 	.family		= PF_BLUETOOTH,
 	.owner		= THIS_MODULE,
```

ayushkevich, 2025-01-21:
It seems that I faced the same issue in Linux Mint Cinnamon. I found that it works fine on Steam Deck. After the investigation I discovered that Cinnamon uses blueman and it doesn't trigger pairing during connection. In addition hid-sony + hid-playstation drivers should be used. So my solution:

- sudo modprobe hid-sony
- sudo modprobe hid-playstation
- Remove the device
- Trust the device
- Pair the device
- Connect it after ALL these steps