bluez/bluez #202: [5.61] mpris-proxy cannot register mpris players

Pagwin-Fedora, 2021-09-09:
as the title says whenever a new mpris player is opened or even when mpris-proxy tries to register players already open when it starts up it always gives an error of 
```
Can't register player
Operation is not supported
``` 
this also seems to lead to a double free if the mpris player in question is closed which causes a segfault which might be related to #114

Bluez/mpris-proxy version: 5.61
Dbus version: 1.12.20
Linux distro: Arch
Wm: i3 (4.19.1)

Vudentz, 2021-09-10:
Cannot reproduce it:

```
tools/mpris-proxy 
org.bluez appeared
Bluetooth Adapter /org/bluez/hci0 found
player org.mpris.MediaPlayer2.chromium.instance2860 at :1.253 found
player org.mpris.MediaPlayer2.chromium.instance4633 at :1.262 found
player org.mpris.MediaPlayer2.chromium.instance55563 at :1.473 found
player org.mpris.MediaPlayer2.rhythmbox at :1.5380 found
player org.mpris.MediaPlayer2.rhythmbox at :1.5380 disappear

```

```
bluetoothd[2779807]: Player registered: sender=:1.105233 path=/_1_473
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() CanSeek not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() MaximumRate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_metadata() Album=
bluetoothd[2779807]: profiles/audio/media.c:set_metadata() Artist=
bluetoothd[2779807]: profiles/audio/media.c:set_metadata() Title=
bluetoothd[2779807]: profiles/audio/media.c:get_uid() 0x63acf70
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() MinimumRate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_status() Status=Stopped
bluetoothd[2779807]: profiles/audio/media.c:set_position() Position=0
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() Rate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() Volume not supported, ignoring
bluetoothd[2779807]: Player registered: sender=:1.105233 path=/_1_5380
bluetoothd[2779807]: profiles/audio/media.c:set_status() Status=Stopped
bluetoothd[2779807]: profiles/audio/media.c:set_property() Repeat=off
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() Rate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_property() Shuffle=off
bluetoothd[2779807]: profiles/audio/media.c:get_uid() 0x63d5560
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() Volume not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() MinimumRate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() MaximumRate not supported, ignoring
bluetoothd[2779807]: profiles/audio/media.c:set_player_property() CanSeek not supported, ignoring
bluetoothd[2779807]: Player unregistered: sender=:1.105233 path=/_1_5380
bluetoothd[2779807]: profiles/audio/media.c:media_player_destroy() sender=:1.105233 path=/_1_5380
```

@Pagwin-Fedora perhaps the problem is that either AVRCP plugin is disabled or you have L2CAP ERTM disabled in the kernel?

Pagwin-Fedora, 2021-09-10:
> Cannot reproduce it:
>
> ```
> tools/mpris-proxy
> org.bluez appeared
> Bluetooth Adapter /org/bluez/hci0 found
> player org.mpris.MediaPlayer2.chromium.instance2860 at :1.253 found
> player [...]

is there a good way of checking if either or both of those are the case?

Vudentz, 2021-09-10:
@Pagwin-Fedora check with your distribution if there how they are loading the Bluetooth modules, I suspect this is a case where L2CAP ERTM is disabled because some people had a problem with it when pluggin so gaming controllers.

closed 2021-09-12 by Pagwin-Fedora