bluez/bluez #206: [Mesh] bluetooth-mesh daemon stops sending hci commands

isakwestin, 2021-09-16:
I'm using the bluetooth-mesh daemon to create a custom Bluetoooth Mesh application.

After some time, the host (daemon) stops sending HCI commands to the bluetooth controller. The last send command from the host is a "LE Set Advertise Enable," enabling advertising (see btmon log below). After that, no more commands are sent. This leaves the controller in a "advertise forever"-state (verified with a bluetooth sniffer) and no more mesh messages can be sent by the application. Mesh messages are however received and correctly delivered to the application with the "MessageReceived" DBUS api call. Restarting the bluetooth-mesh daemon gets it working again, until the same scenario happens again. Typically the problem occurs sooner when there are several (perhaps 100) mesh messages sent in a sequence by the application.

Another interesting behavior before the above occurs, is that "LE Set Advertising Parameters" is sometimes being sent after enabling advertising, making the controller return "Command Disallowed". Can also be seen in the btmon log.

I'm having a hard time to figure this one out. Even if the root cause is in the application I would appriciate your help to troubleshoot it.

Let me know if additional information is needed. Below the btmon log is also the output from dbus-monitor.

My setup:
Host: aarch64 GNU/Linux
bluez version: 5.55 (I also tried with 5.61 and got the same behavior)
bluetooth controller: nrf52832 chip running Zephyr hci_uart sample

The application is only using the DBUS-apis for mesh functionality. 

btmon output:
```
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                      #5931 [hci0] 41.898621
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                                      #5932 [hci0] 41.900241
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1                      #5933 [hci0] 41.900827
        Advertising: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                                      #5934 [hci0] 41.902620
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Parameters (0x08|0x0006) plen 15               #5935 [hci0] 41.902770
        Min advertising interval: 20.000 msec (0x0020)
        Max advertising interval: 20.000 msec (0x0020)
        Type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Own address type: Random (0x01)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from White List Only, Allow Connect Request from White List Only (0x03)
> HCI Event: Command Complete (0x0e) plen 4                                      #5936 [hci0] 41.905582
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32                     #5937 [hci0] 41.905741
        Length: 23
        Mesh Data: 0fc2f35002bbefcb6484b28ea45d6381f4349bbcab
          IVI: 0
          NID: 0x0f
        c2 f3 50 02 bb ef cb 64 84 b2 8e a4 5d 63 81 f4  ..P....d....]c..
        34 9b bc ab                                      4...            
> HCI Event: LE Meta Event (0x3e) plen 35                                        #5938 [hci0] 41.911374
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Address type: Random (0x01)
        Address: 17:CE:5D:D6:F7:D7 (Non-Resolvable)
        Data length: 23
        Mesh Data: 0fe5df4baca30e5eb017694cb3988c88dc2ea71231
          IVI: 0
          NID: 0x0f
        e5 df 4b ac a3 0e 5e b0 17 69 4c b3 98 8c 88 dc  ..K...^..iL.....
        2e a7 12 31                                      ...1            
        RSSI: -49 dBm (0xcf)
> HCI Event: LE Meta Event (0x3e) plen 35                                        #5939 [hci0] 41.914207
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Address type: Random (0x01)
        Address: 23:4E:50:CD:10:F6 (Non-Resolvable)
        Data length: 23
        Mesh Data: 0f5149b86f6f25a45d715ba2750b70388cc9e2b143
          IVI: 0
[... 1353 lines cut ...]
      52 c4 09 f4 c4 05
   ]
method call time=1631794781.434218 sender=:1.11 -> destination=:1.12 serial=312 path=/bt_mesh/0/app/elem0; interface=org.bluez.mesh.Element1; member=MessageReceived
   uint16 91
   uint16 0
   variant       uint16 49152
   array of bytes [
      82 4e 00 00
   ]
method call time=1631794781.516351 sender=:1.11 -> destination=:1.12 serial=313 path=/bt_mesh/0/app/elem0; interface=org.bluez.mesh.Element1; member=MessageReceived
   uint16 108
   uint16 0
   variant       uint16 49152
   array of bytes [
      52 a0 09 00
   ]
method call time=1631794781.592111 sender=:1.11 -> destination=:1.12 serial=314 path=/bt_mesh/0/app/elem0; interface=org.bluez.mesh.Element1; member=MessageReceived
   uint16 110
   uint16 0
   variant       uint16 49152
   array of bytes [
      52 c4 09 60 36 05
   ]

```

csilva-trd, 2022-02-02:
Hi @isakwestin,
Did you manage to fix this issue ? We are facing the same!

BR,

isakwestin, 2022-02-09:
Hi @csilva-trd,
No, I'm using Bluez version 5.62 now and it is still an issue.
For now, I have restricted the "network transmit interval steps" for the local node (i.e. the mesh-daemon) so that it has a re-transmit interval of at least 50ms for outgoing messages. This way, the system is a bit slower, but the above issue seldom occurs.

However, it does occur once in a while. So it would be good if we could get some attention to this issue since it makes the system quite unreliable :)

isakwestin, 2022-08-12:
I believe that this commit fixed my issue: 71560e12863ff1b133e421ef7dd25d20c8d83acc

closed 2022-08-12 by isakwestin