bluez/bluez #208: Have to (suspend then resume) everytime [Headset randomly stop playing sound]

nellowly, 2021-09-18:
Sound stopped playing at 10:19:19
Suspend after 10:19:19 and resume at 10:19:46
Successfully connected at 10:20:09
```
-- Boot 77c1997c669a4a858735eae077a99878 --
Sep 18 10:19:03 arch systemd[1]: Starting Bluetooth service...
Sep 18 10:19:03 arch bluetoothd[569]: Bluetooth daemon 5.61
Sep 18 10:19:03 arch systemd[1]: Started Bluetooth service.
Sep 18 10:19:03 arch bluetoothd[569]: Starting SDP server
Sep 18 10:19:03 arch bluetoothd[569]: Bluetooth management interface 1.21 initialized
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/ldac
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSink/aptx_hd
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx_hd
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSink/aptx
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aac
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSink/sbc
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/sbc
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSink/sbc_xq
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/sbc_xq
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/faststream
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/faststream_duplex
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Sep 18 10:19:12 arch bluetoothd[569]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Sep 18 10:19:19 arch bluetoothd[569]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Sep 18 10:19:46 arch bluetoothd[569]: Device is already marked as connected
Sep 18 10:20:09 arch bluetoothd[569]: /org/bluez/hci0/dev_94_DB_56_17_9E_C2/sep5/fd0: fd(31) ready
```

Vudentz, 2021-10-04:
Please add the HCI traces when the suspend timeout, it looks like the headset is not responding for some reason. Btw, what is the manufacturer and model of your headset?

nellowly, 2021-10-05:
Hi! Manufacturer is sony, model WH-XB900N

I will start the HCI capture now. after I type the command `$ btmon -i hci0 -w hcidump.log`

When do I stop the capturing.. Right away or wait for x seconds?

Just to make sure because hcidump.log is very long and the error doesn't appear in systemctl status for a few moment...

nellowly, 2021-10-05:
So I forgot to mention that recently I sometimes can only hear sound from one side after a while

In this case I can still hear it but the volume is lower

Log in this case:

```
< SCO Data TX: Handle 4 flags 0x00 dlen 72                                                                                                      #124032 [hci0] 829.863521
> SCO Data RX: Handle 4 flags 0x00 dlen 72                                                                                                      #124033 [hci0] 829.871525
> SCO Data RX: Handle 4 flags 0x00 dlen 72                                                                                                      #124034 [hci0] 829.871542
< SCO Data TX: Handle 4 flags 0x00 dlen 72                                                                                                      #124035 [hci0] 829.879300
> SCO Data RX: Handle 4 flags 0x00 dlen 72                                                                                                      #124036 [hci0] 829.881513
< SCO Data TX: Handle 4 flags 0x00 dlen 72                                                                                                      #124037 [hci0] 829.886028
< SCO Data TX: Handle 4 flags 0x00 dlen 72                                                                                                      #124038 [hci0] 829.894401
> SCO Data RX: Handle 4 flags 0x00 dlen 72                                                                                                      #124039 [hci0] 829.901520
> SCO Data RX: Handle 4 flags 0x00 dlen 72                                                                                                      #124040 [hci0] 829.901530
< SCO Data TX: Handle 4 flags 0x00 dlen 72                                                                                                      #124041 [hci0] 829.901814
```

I uploaded the full log file in case you need them [https://drive.google.com/drive/u/0/folders/1Tf5tHXoeEIFRNfRwHsVXU4527TiYvsy8](https://drive.google.com/drive/u/0/folders/1Tf5tHXoeEIFRNfRwHsVXU4527TiYvsy8)

Both of the case can be solved with disconnect then reconnect or just wait for a while and it will be back at the correct volume

The third case with completely no sound which can't be disconnect then reconnect.

nellowly, 2021-10-05:
Here is the HCI traces when the suspend timeout !

```
< ACL Data TX: Handle 3 flags 0x02 dlen 784
      Channel: 1861 len 780 [PSM 0 mode Basic (0x00)] {chan 65535}
        80 60 23 4b 27 7f ec 00 00 00 00 00 f7 00 6c f7  .`#K'.........l.
        1b fd 18 83 c8 19 66 0b 1e fa 33 1e eb ef ef 12  ......f...3.....
        05 f8 8e 65 c6 25 ea e7 8c 40 f6 92 2b e0 98 3a  ...e.%...@..+..:
        2f 73 e9 f8 7c 44 ef ee 38 d0 08 17 cf fe a2 ef  /s..|D..8.......
        60 71 09 70 2c 18 0c 1f e7 11 d5 d6 eb a4 06 d2  `q.p,...........
        32 10 68 0b ed 93 67 ff 7d bd d7 fb 92 d1 0c 6b  2.h...g.}......k
        17 7a 12 f6 88 1d d1 82 64 ea 05 cb 0e f0 40 4e  .z......d.....@N
        62 10 20 7d fe 17 8d be 2f f3 a3 0f 85 4c e6 05  b. }..../....L..
        b5 37 6b ed d7 f4 28 21 76 6f 3f 22 34 3f 18 29  .7k...(!vo?"4?.)
        e0 98 0e e0 8e 71 17 ec 69 4f e1 e3 20 5c 5d 27  .....q..iO.. \]'
        f8 6f 00 74 63 f7 86 4d 28 03 91 1f 82 07 10 94  .o.tc..M(.......
        68 11 8c a1 0e f2 48 ff f2 10 28 23 86 f8 9f c7  h.....H...(#....
        39 19 fc 38 00 15 0f 0d f3 06 75 ef f9 15 ac 09  9..8......u.....
        99 58 ce 03 ec ce f6 39 10 89 8b 30 85 84 40 f9  .X.....9...0..@.
        a7 17 8f c1 f8 0a 5b df 97 f7 d8 08 19 e8 15 a0  ......[.........
        00 0e 08 10 74 82 08 79 d7 f1 18 34 fe 9c 07 cf  ....t..y...4....
        10 61 18 eb f8 f8 9d af 1e fd eb fe 77 e2 d8 ec  .a..........w...
        86 cf ff cf d8 13 68 0f 05 cc d6 84 14 e7 e2 2c  ......h........,
        c8 90 8e c9 16 40 07 ec 64 0f 76 a9 c1 6c 04 d0  .....@..d.v..l..
        7c 1e 11 04 7d 20 f6 5b fe ef f0 ee 95 fc c7 f9  |...} .[........
        98 d7 77 af e6 6e 42 05 9b c4 00 7c 1e f1 6a 05  ..w..nB....|..j.
        1e 84 40 f8 0a 15 00 f1 d8 09 07 f3 d8 fa 13 08  ..@.............
        5e 3e ef fe 02 0f f9 ec 1f 0a 16 ef 80 1c 08 09  ^>..............
        b6 0e 9f e8 08 1f f6 3f f4 96 c0 82 1f f1 05 b3  .......?........
        27 1a 97 20 7a 0c 11 76 16 e9 0c 7a ef 1b b7 1f  '.. z..v...z....
        01 7d ff fe 80 09 05 f6 fe f3 b0 d8 19 cd 08 f9  .}..............
        d4 e9 7c 22 f0 9c a3 e0 9a 5f 17 05 ed f7 95 cd  ..|"....._......
        29 4f 6b d9 5f c7 0e 02 0d e6 76 1e 28 fd bc 18  )Ok._.....v.(...
        f1 8b 46 1d d9 36 05 ac f1 6d ac f1 72 45 ff 19  ..F..6...m..rE..
        d1 1e 89 5d b9 83 b4 b8 07 83 f7 7a 15 06 f9 b3  ...].......z....
        f7 e6 45 e0 ef c7 c7 61 db ff d3 e3 10 ed 91 01  ..E....a........
        73 f3 df 10 13 b8 02 37 10 04 05 10 9b fb f7 7f  s......7........
        eb e7 13 dd 10 f8 5e 10 09 d2 10 69 b4 e7 07 d1  ......^....i....
        1f 88 3e 1f fd fa f7 ef c1 19 6b ec ff f9 b7 f9  ..>.......k.....
        78 6f ef 07 b3 08 08 52 ff 98 05 0f 01 d3 f0 77  xo.....R.......w
        8a 17 81 cc 27 7f 83 27 e2 18 21 71 ac 20 8f cb  ....'..'..!q. ..
        2e 71 ae 2f 5c 44 ff 12 19 11 09 d4 d8 89 c4 d8  .q./\D..........
        23 d4 30 87 af 26 76 2d 29 0e a0 2f 07 ea d0 1c  #.0..&v-)../....
        4d c7 03 91 f7 94 00 c7 a3 ec 20 fe 74 20 fc 8f  M......... .t ..
        06 5e 55 f7 f4 1a fe a2 73 e1 92 7a ef f0 3a df  .^U.....s..z..:.
        fc a7 e0 98 0b 27 10 38 ff 6c 5a 08 5e 5a 1f 7b  .....'.8.lZ.^Z.{
        bd e9 00 2a 08 08 08 d8 8c 8f d8 74 72 e8 68 60  ...*.......tr.h`
        00 e4 5a 11 65 bf c8 19 cc d7 98 26 00 7c 09 07  ..Z.e......&.|..
        64 26 07 f3 a1 e0 07 a9 0d ee 05 ef f3 dd 38 06  d&............8.
        14 f9 98 01 20 77 b8 07 fd b2 19 f6 4b f8 6e 04  .... w......K.n.
        e7 91 e2 e7 89 75 be 83 d9 07 01 94 e6 9b 78 17  .....u........x.
        01 bf 06 dd ef f7 69 c9 c9 82 02 b9 86 69 d7 10  ......i......i..
        40 f6 12 3f e1 ea 88 11 6e 31 37 f8 55 37 69 f5  @..?....n17.U7i.
        18 61 b9 0f e2 04 f8 15 ed f7 8c 04              .a..........

< HCI Command: Read Clock Offset (0x01|0x001f) plen 2
                                                                #2570573 17062.925594
        Handle: 3
< HCI Command: Disconnect (0x01|0x0006) plen 3
                                                                #2570574 17064.957816
        Handle: 3
        Reason: Remote User Terminated Connection (0x13)
```

Full trace: [https://drive.google.com/drive/u/0/folders/1Tf5tHXoeEIFRNfRwHsVXU4527TiYvsy8](https://drive.google.com/drive/u/0/folders/1Tf5tHXoeEIFRNfRwHsVXU4527TiYvsy8)

seamusdemora, 2024-04-09:
I know this isn't a helpful comment, but may be useful for setting expectations. If you spend a week or so chasing Bluetooth problems on a Linux system, you begin to wonder: "Does Bluetooth on Linux just suck?" Unfortunately, I think the answer may be, "Yes, it does suck." I've played audio thru a BT speaker (JBL Flip 5) using my iPhone6 as the music source & speaker control continuously for days without an interrupt. I can occasionally manage an hour or two using Linux with the same Flip 5 speaker. It just sucks - so deal with it.

Vudentz, 2024-04-09:
> I know this isn't a helpful comment, but may be useful for setting expectations. If you spend a week or so chasing Bluetooth problems on a Linux system, you begin to wonder: "Does Bluetooth on Linux [...]

Or you turn around as ask, does JBL cares to test any of their peripherals with Linux, no they don't, do they care to test with iPhone, absolutely yes, would they even go as far as changing their own stack to make it work with iPhone, yes again... So if you expect it to just works, I doubt we can help, but we can perhaps help if there is a little bit more information.

seamusdemora, 2024-04-10:
How nice of you! What information could I supply to help with this?