bluez/bluez #209: Media control not working

Wiedzmin89, 2021-09-22:
I have my headphones connected via bluetooth but media keys not working.

My device connected:
[image]

But media control does not:
[image]

And also volume level sync not working. If i understand properly it relates on AVRCP.

Versions:
bluez-git 1:5.61.r50.g9ee79f67b-1
bluez-qt 5.86.0-1

Operating System: Arch Linux
KDE Plasma Version: 5.22.5
KDE Frameworks Version: 5.86.0
Qt Version: 5.15.2
Kernel Version: 5.10.67-1-lts (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i5-8250U CPU @ 1.60GHz
Memory: 15.5 ГиБ of RAM
Graphics Processor: Mesa Intel® UHD Graphics 620

Vudentz, 2021-09-23:
Please add the HCI traces and daemon logs.

Wiedzmin89, 2021-09-23:
[bluetooth.txt](https://github.com/bluez/bluez/files/7219667/bluetooth.txt)
[btmon.txt](https://github.com/bluez/bluez/files/7219668/btmon.txt)

Vudentz, 2021-09-23:
Weird PSM 23 doesn't appear in your logs, I suspect there is something else at play try running bluetoothd from a shell with -dn to see if there any reason why the following is happening:

`сен 23 21:06:45 wiedzmin bluetoothd[448]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for 94:DB:56:7E:1E:54: Input/output error
`

Wiedzmin89, 2021-09-23:
[bluetoothd.txt](https://github.com/bluez/bluez/files/7220795/bluetoothd.txt)

Vudentz, 2021-09-23:
You will need to stop the bluetooth.service first, e.g: sudo systemctl stop bluetooth.service, then attempt to connect again.

Wiedzmin89, 2021-09-23:
[bluetoothd.txt](https://github.com/bluez/bluez/files/7220911/bluetoothd.txt)

Vudentz, 2021-09-24:
```
bluetoothd[168838]: profiles/audio/avrcp.c:avrcp_controller_server_probe() path /org/bluez/hci0
bluetoothd[168838]: profiles/audio/avctp.c:avctp_server_socket() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
bluetoothd[168838]: avrcp-controller: Protocol not supported (93)
bluetoothd[168838]: profiles/audio/avrcp.c:avrcp_target_server_probe() path /org/bluez/hci0
bluetoothd[168838]: profiles/audio/avctp.c:avctp_server_socket() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
bluetoothd[168838]: audio-avrcp-target: Protocol not supported (93)
```

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/l2cap_core.c#n8517

Check if your distro is using disable_ertm that would explain why the protocol is not available.

Vudentz, 2021-09-24:
@Wiedzmin89 would you able to test with the above changes?

Wiedzmin89, 2021-09-25:
@Vudentz I tested this version: [https://github.com/tedd-an/bluez/pull/1251](https://github.com/tedd-an/bluez/pull/1251)
Media connected successfully, thanks everyone

Wiedzmin89, 2021-09-25:
And, yes. It was disable_ertm Y
[image]

closed 2021-09-27 by BluezTestBot