bluez/bluez #210: mSBC missing after A2DP profile switching

bdaase, 2021-09-23:
* Fedora 35
* bluez 5.61
* Liberty Air 2 Pro

On some connects, mSBC is randomly missing after connecting my Liberty Air 2 Pro.
I can also reliably reproduce this by switching the profile between SBC and AAC (or vice versa), which makes mSBC disappear. Whenever this happens, there is the following line in the journal
```
Sep 23 11:41:44 p14s-gen2-amd bluetoothd[18491]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
```

The issue appears independent of the adapter, both with a Killer 1435 and an AX210.

Pipewire issue: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/810

Vudentz, 2021-09-23:
Please add the HCI traces using btmon, if this is a clean disconnect Im afraid there is nothing much we can do.

bdaase, 2021-09-23:
> Please add the HCI traces using btmon,

Here you go, I hope that's what you were asking for: [hcitrace.txt](https://github.com/bluez/bluez/files/7218946/hcitrace.txt).
It should only contain the profile switching and no further interaction with the device before or after.

> if this is a clean disconnect Im afraid there is nothing much we can do.

Would you mind explaining why? I do not yet understand why after a clean disconnect/connect HSP/HFP codecs should not be available anymore.

Vudentz, 2021-09-23:
> > Please add the HCI traces using btmon,
> 
> Here you go, I hope that's what you were asking for: [hcitrace.txt](https://github.com/bluez/bluez/files/7218946/hcitrace.txt).
> It should only contain the profile switching and no further interaction with the device before or after.

You probably need to start recording from the start otherwise btmon wont be able to decode the RFCOMM commands, anyway I suspect the remote is sending a disconnect.

> > if this is a clean disconnect Im afraid there is nothing much we can do.
> 
> Would you mind explaining why? I do not yet understand why after a clean disconnect/connect HSP/HFP codecs should not be available anymore.

If it is intended behavior (e.g. due to some policy) reconnecting would just end up in an endless loop thus way most stack don't reconnect, that said it apparently is a direct RFCOMM disconnect but the daemon has no control over the AT command it can't determine if that was indeed a clean disconnection or not.

Another thing to keep in mind is that the headset may not have enough memory to have both SBC and AAC loaded at the same time, so when you change the codec to AAC it has to unload the SBC(mSBC) causing the disconnection, if that would be the case then probably when HFP/HSP reconnects there won't be support for mSBC anymore.

bdaase, 2021-09-23:
>  You probably need to start recording from the start otherwise btmon wont be able to decode the RFCOMM commands, anyway I suspect the remote is sending a disconnect.

Alright, here is a new HCI trace including the initial connect. The headset initially connected with AAC and then I changed the profile to SBC -> mSBC disappeared. Is that recoding better?

> If it is intended behavior (e.g. due to some policy) reconnecting would just end up in an endless loop thus way most stack don't reconnect, that said it apparently is a direct RFCOMM disconnect but [...]

Why exactly do we get stuck in an endless loop? I do not yet get the difference between an initial connect (where mSBC is available) and a disconnect/reconnect, (where mSBC is not available).
Maybe I am just missing some basic Bluetooth knowledge here :)

> Another thing to keep in mind is that the headset may not have enough memory to have both SBC and AAC loaded at the same time, so when you change the codec to AAC it has to unload the SBC(mSBC) [...]

I actually do not think that is the case. As shown in the trace above, the problem also occurs when switching from AAC to SBC, where mSBC is available while using AAC but not after the switch to SBC.

bdaase, 2021-09-27:
Additionally, here is the trace of a normal connect which didn't have mSBC available: 
[hcitrace.txt](https://github.com/bluez/bluez/files/7234395/hcitrace.txt)

bdaase, 2022-02-08:
FWIW, this still happens with a 5.17.rc3 kernel.

closed 2025-08-25 by github-actions[bot]