bluez/bluez #211: Failure to register notifications for headset battery level

brutester, 2021-09-26:
I have a Sennheiser HD 450 BT headset. I am trying to make Gnome settings display the battery level.

I found that I can extract the level using
```
# gatttool -i hci0 -b 00:1B:66:C7:BF:E0 --char-read -a 0x002d
Characteristic value/descriptor: 64  #  == 100%
## repeat after a couple of hours
Characteristic value/descriptor: 5a  # == 90%
```

I then found that if I manually connect in LE mode, the attributes get populated and upower can report battery level for around 30 seconds. After that it is no longer reported.
```
# bluetoothctl
[HD 450BT]# connect 00:1B:66:C7:BF:E0 
[LE-HD 450BT]# menu gatt
[LE-HD 450BT]# attribute-info /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/service002b/char002c
Characteristic - Battery Level
	UUID: 00002a19-0000-1000-8000-00805f9b34fb
	Service: /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/service002b
	Notifying: no
	Flags: read
	Flags: notify

# upower -d
Device: /org/freedesktop/UPower/devices/unknown_dev_00_1B_66_C7_BF_E0
  native-path:          /org/bluez/hci0/dev_00_1B_66_C7_BF_E0
  model:                LE-HD 450BT
  serial:               00:1B:66:C7:BF:E0
  power supply:         no
  updated:              Thu Jan  1 02:00:00 1970 (1632647811 seconds ago)
  has history:          yes
  has statistics:       no
  unknown
    warning-level:       none
    icon-name:          'battery-missing-symbolic'
```

I am no expert, but it seems that the root cause of the problem is that the device doesn't support notifications. Bluez code tries to configure one, thus the errors in logs:
bluetoothd[13686]: src/device.c:gatt_debug() (chan 0x55914d530d30) Operation timed out: 0x12
bluetoothd[13686]: src/device.c:gatt_debug() (chan 0x55914d530d30) Failed to obtain disconnect error: Bad file descriptor

Here are the debug logs. I had correlated them for T+0s == `profiles/battery/battery.c:parse_battery_level() Battery Level updated: 90%`:
```
# journalctl -u bluetooth.service -f
T-5s bluetoothd[13686]: profiles/audio/transport.c:media_owner_create() Owner created: sender=:1.71
T-5s bluetoothd[13686]: profiles/audio/a2dp.c:a2dp_sep_lock() SEP 0x55914d54b250 locked
T-5s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_ref() 0x55914d5559c0: ref=3
T-5s bluetoothd[13686]: profiles/audio/a2dp.c:setup_ref() 0x55914d540a00: ref=1
T-5s bluetoothd[13686]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/sep5/fd0: TRANSPORT_STATE_IDLE -> TRANSPORT_STATE_REQUESTING
T-5s bluetoothd[13686]: profiles/audio/transport.c:media_request_create() Request created: method=Acquire id=42
T-5s bluetoothd[13686]: profiles/audio/transport.c:media_owner_add() Owner :1.71 Request Acquire
T-5s bluetoothd[13686]: profiles/audio/transport.c:media_transport_set_owner() Transport /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/sep5/fd0 Owner :1.71
T-5s bluetoothd[13686]: profiles/audio/avdtp.c:session_cb()
T-5s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_parse_resp() START request succeeded
T-5s bluetoothd[13686]: profiles/audio/a2dp.c:start_cfm() Source 0x55914d54b250: Start_Cfm
T-5s bluetoothd[13686]: profiles/audio/transport.c:media_owner_remove() Owner :1.71 Request Acquire
T-5s bluetoothd[13686]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/sep5/fd0: TRANSPORT_STATE_REQUESTING -> TRANSPORT_STATE_ACTIVE
T-5s bluetoothd[13686]: profiles/audio/a2dp.c:setup_unref() 0x55914d540a00: ref=0
T-5s bluetoothd[13686]: profiles/audio/a2dp.c:setup_free() 0x55914d540a00
T-5s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_unref() 0x55914d5559c0: ref=2
T-5s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: OPEN -> STREAMING
T-5s bluetoothd[13686]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_00_1B_66_C7_BF_E0: SINK_STATE_CONNECTED -> SINK_STATE_PLAYING
T-5s bluetoothd[13686]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/sep5/fd0 State=TRANSPORT_STATE_ACTIVE Playing=1
T-2s bluetoothd[13686]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_00_1B_66_C7_BF_E0/sep5/fd0: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_SUSPENDING
T-2s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_ref() 0x55914d5559c0: ref=3
T-2s bluetoothd[13686]: profiles/audio/a2dp.c:setup_ref() 0x55914d53bab0: ref=1
T-2s bluetoothd[13686]: profiles/audio/transport.c:media_request_create() Request created: method=Release id=43
T-2s bluetoothd[13686]: profiles/audio/transport.c:media_owner_add() Owner :1.71 Request Release
T-2s bluetoothd[13686]: profiles/audio/avdtp.c:session_cb()
T-2s bluetoothd[13686]: profiles/audio/avdtp.c:avdtp_parse_resp() SUSPEND request succeeded
[... 73 lines cut ...]
T+30s bluetoothd[13686]: src/device.c:gatt_debug() (chan 0x55914d530d30) Failed to obtain disconnect error: Bad file descriptor
T+30s bluetoothd[13686]: src/device.c:gatt_debug() Channel 0x55914d530d30 disconnected: Success
T+30s bluetoothd[13686]: src/device.c:gatt_debug() Failed to register handler for "Service Changed"
T+30s bluetoothd[13686]: src/device.c:att_disconnected_cb()
T+30s bluetoothd[13686]: src/device.c:att_disconnected_cb() Success (0)
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d54e410: device 00:1B:66:C7:BF:E0 profile batt-profile state changed: connected -> disconnecting (0)
T+30s bluetoothd[13686]: src/battery.c:btd_battery_unregister() path = /org/bluez/hci0/dev_00_1B_66_C7_BF_E0
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d54e410: device 00:1B:66:C7:BF:E0 profile batt-profile state changed: disconnecting -> disconnected (0)
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d555900: device 00:1B:66:C7:BF:E0 profile deviceinfo state changed: connected -> disconnecting (0)
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d555900: device 00:1B:66:C7:BF:E0 profile deviceinfo state changed: disconnecting -> disconnected (0)
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d555880: device 00:1B:66:C7:BF:E0 profile gap-profile state changed: connected -> disconnecting (0)
T+30s bluetoothd[13686]: src/service.c:change_state() 0x55914d555880: device 00:1B:66:C7:BF:E0 profile gap-profile state changed: disconnecting -> disconnected (0)
T+30s bluetoothd[13686]: src/gatt-client.c:btd_gatt_client_disconnected() Device disconnected. Cleaning up.
T+30s bluetoothd[13686]: src/device.c:att_disconnected_cb() Automatic connection disabled
T+30s bluetoothd[13686]: src/gatt-database.c:btd_gatt_database_att_disconnected()
T+30s bluetoothd[13686]: src/gatt-database.c:att_disconnected()
T+30s bluetoothd[13686]: attrib/gattrib.c:g_attrib_unref() 0x55914d55e180: g_attrib_unref=0
T+33s bluetoothd[13686]: src/adapter.c:dev_disconnected() Device 00:1B:66:C7:BF:E0 disconnected, reason 2
T+33s bluetoothd[13686]: src/adapter.c:adapter_remove_connection()
T+33s bluetoothd[13686]: plugins/policy.c:disconnect_cb() reason 2
T+33s bluetoothd[13686]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:1B:66:C7:BF:E0 type 1 status 0xe
T+33s bluetoothd[13686]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
T+33s bluetoothd[13686]: src/device.c:device_bonding_failed() status 14
T+33s bluetoothd[13686]: src/adapter.c:resume_discovery()
```

Vudentz, 2021-09-27:
`T+30s bluetoothd[13686]: src/device.c:gatt_debug() Failed to register handler for "Service Changed"`

Looks like it couldn't subscribe to Service Changed, try collecting the HCI trace when connecting (using btmon).

brutester, 2021-09-28:
Attaching the hci trace.
[hcitrace-full.zip](https://github.com/bluez/bluez/files/7241903/hcitrace-full.zip)

I see the battery level being reported in frame 611 and 30.05 seconds later at frame 652.  But the "Operation timed out" is on frame 632, which is 27ms prior to frame 652.

eternal-sorrow, 2022-09-20:
Any news about this issue?

valeth, 2023-02-16:
Recently the battery state appeared reliably in the KDE Plasma bluetooth widget.
I'm not sure if that was because of a `bluez` or `upower` update, but it just started working.

But I think it still has problems refreshing the battery percentage without a reconnect.
Could also just be an UI bug, needs further testing.