bluez/bluez #212: Sending ADV_DIRECT_IND

hexpwn, 2021-09-26:
I can't find a way to specify the pdu type for advertising packets, to send `ADV_DIRECT_IND`. Is it even possible to customize this in BlueZ?

I'm using `dbus` to interact with BlueZ.

Vudentz, 2021-09-27:
@hexpwn it is currently not supported over D-Bus, we would probably need to add another type to:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt#n37

That said one would also need to provide a Device object, or initiate a Device.Connect in order for the kernel to know it should use directed advertising.

hexpwn, 2021-09-27:
I appreciate the clarifications, @Vudentz! I'm not familiar enough with BlueZ to attempt a patch but it's something I might give a shot at in the future.

closed 2021-09-27 by hexpwn