bluez/bluez #213: bluetoothctl | 'power on' fails to unblock soft blocked Buetooth module

Ricky-Tigg, 2021-10-04:
bluez version: **5.61**; Tested on: **Fedora**; Kernel version: **5.14.7**

Original state of Bluetooth module:
```
$ rfkill list | grep -A 2 'Bluetooth$'
1: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
```
Note: util-linux's '`rfkill unblock <port_number>`' and tlp's '`bluetooth on`' are able to overcome Bluetooth module soft restriction, _blocked_,  systematically and instantaneously.

To reproduce:
```
$ bluetoothctl
Agent registered
[CHG] Controller xx:xx:xx:xx:xx:xx Pairable: yes
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Busy
```

Vudentz, 2021-10-04:
@Ricky-Tigg rfkill acts as a master lock so as far I remember this works as intended, or are you saying that power on should unblock itself? There would probably need to be a request for the user to approve it if we would go in that direction so Id expect there should be an application managing the rfkill settings (e.g. gnome bluetooth settings, etc).

Ricky-Tigg, 2021-10-05:
In graphical environment, which is the case in _Fedora Desktop_ release, whose default is _Gnome_ – now in version 41 – ,  "Powering on" function in Gnome menu does achieve two operations: unlocking Bluez controller then powering on the _Bluetooth_ module. That did not come to mind as a valid suggestion for improvement. Nevertheless since your last comment, i could conceive that it would be advantageous.

Here is the event that has been at the source of my motivation to post the present report; '`bluetoothctl`''s '`unblock'` had **no effect**, Then this operation had to rely on a third-party software function, such as '`rfkill`'. _Note_: MAC address of my computer has been replaced in afterwards by "_xx:xx:xx:xx:xx:xx"_. 
```
$ bluetoothctl
Agent registered
[CHG] Controller xx:xx:xx:xx:xx:xx Pairable: yes
[bluetooth]# unblock xx:xx:xx:xx:xx:xx
Device xx:xx:xx:xx:xx:xx not available
```

Ricky-Tigg, 2021-10-05:
In a graphical environment, which is the case in _Fedora Desktop_ release, whose default is _Gnome_ – now in version 41 – ,  "_Poweri on_" function in Gnome menu does achieve two operations: unlocking BlueZ controller then powering on the _Bluetooth_ module. That did not come to mind as a valid suggestion for improvement. Nevertheless since your last comment, i could conceive that it would be advantageous.

Here is the event that has been at the source of my motivation to post the present report; '`bluetoothctl`''s '`unblock'` had **no effect**, Then i had to rely on a third-party software function, such as '`rfkill`' in order to achieve operation. _Note_: MAC address of my computer has been replaced in afterwards by "_xx:xx:xx:xx:xx:xx"_. 
```
$ bluetoothctl
Agent registered
[CHG] Controller xx:xx:xx:xx:xx:xx Pairable: yes
[bluetooth]# unblock xx:xx:xx:xx:xx:xx
Device xx:xx:xx:xx:xx:xx not available
```

Ricky-Tigg, 2021-10-05:
In a graphical environment, which is the case in _Fedora Desktop_ release, whose default is _Gnome_ – now in version 41 – ,  "_Poweri on_" function in Gnome menu does achieve two operations: unlocking BlueZ controller then powering on the _Bluetooth_ module. That did not come to mind as a valid suggestion for improvement. Nevertheless since your last comment, i could conceive that it would be advantageous.

Here is the event that has been at the source of my motivation to post the present report; '`bluetoothctl`''s '`unblock'` had **no effect**, Then i had to rely on a third-party software function, such as '`rfkill`' in order to achieve this operation. _Note_: MAC address of my computer has been replaced in afterwards by "_xx:xx:xx:xx:xx:xx"_. 
```
$ bluetoothctl
Agent registered
[CHG] Controller xx:xx:xx:xx:xx:xx Pairable: yes
[bluetooth]# unblock xx:xx:xx:xx:xx:xx
Device xx:xx:xx:xx:xx:xx not available
```

Ricky-Tigg, 2021-10-11:
@qiurui144 . That's relevant information does emphasize a inconsistency in the present project rarely observable among software projects and whicht cannot be missed, The _Wiki_-named [section](https://github.com/bluez/bluez/wiki), which is empty,  redirects to a [page](http://www.bluez.org/) with no Wiki-manual!. No surprise that valuable information ends up here randomly. To all maintainers; feel free to close any time, unless you intend to improve something