bluez/bluez #214: Cannot connect to device with random address type

weliem, 2021-10-05:
I am trying to connect to a device that has a random address. It is the Beurer FT95 thermometer.
When I scan for it I see it:

`2021-10-05 08:03:55:723 DEBUG [Main] Device{name='FT95', address='FF:0A:3D:05:5E:26', address_type=random, rssi=-66, uuids=[00001809-0000-1000-8000-00805f9b34fb], manufacturer_data=[0611 -> 01], service_data=[], paired=0, txpower=0 path='/org/bluez/hci0/dev_FF_0A_3D_05_5E_26' }`

After the device has been discovered I stop the discovery and try to connect to it. But then I get the following error:

`error 41: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "Connect" with signature "" on interface "org.bluez.Device1" doesn't exist`

After some investigation I found that after stopping the discovery, **all devices with a random address are immediately removed from the DBus**, as opposed to devices with a static address that stay on the DBus for some seconds. Hence the device is not there anymore when I try to connect.

So I tried connecting to the device while discovery is still running and then the connection works. But that gives me a different problem because some adapters don't like it when you connect while a discovery is running!

Any thoughts on this? Can I indeed only connect to devices with a random address while discovery is running? That would be quite inconvenient...

Vudentz, 2021-10-05:
There are only remove if temporary which should be cleared once you have connected, what version are you running?

weliem, 2021-10-05:
I am running Bluez 5.56

Vudentz, 2021-10-05:
@weliem try using the latest, there might some fixes in this regard and temporary devices now stay for longer.

weliem, 2021-10-30:
Also tried 5.62 and the issue remains:

```
2021-10-30 17:11:15:523 DEBUG [Main] Device{name='(null)', address='48:65:BB:B7:CB:B7', address_type=random, rssi=-83, uuids=[], manufacturer_data=[004C -> 10050118b6afa3], service_data=[], paired=0, txpower=0 path='/org/bluez/hci0/dev_48_65_BB_B7_CB_B7' }
2021-10-30 17:11:16:810 DEBUG [Main] Device{name='FT95', address='FF:0A:3D:05:5E:26', address_type=random, rssi=-69, uuids=[00001809-0000-1000-8000-00805f9b34fb], manufacturer_data=[0611 -> 01], service_data=[], paired=0, txpower=0 path='/org/bluez/hci0/dev_FF_0A_3D_05_5E_26' }
2021-10-30 17:11:16:826 DEBUG [Adapter] Device /org/bluez/hci0/dev_37_66_C1_39_63_2B removed
2021-10-30 17:11:16:826 DEBUG [Device] freeing /org/bluez/hci0/dev_37_66_C1_39_63_2B
2021-10-30 17:11:16:827 DEBUG [Adapter] Device /org/bluez/hci0/dev_0F_F1_CE_14_A3_CE removed
2021-10-30 17:11:16:827 DEBUG [Device] freeing /org/bluez/hci0/dev_0F_F1_CE_14_A3_CE
2021-10-30 17:11:16:828 DEBUG [Adapter] Device /org/bluez/hci0/dev_D2_A3_05_BA_A1_76 removed
2021-10-30 17:11:16:828 DEBUG [Device] freeing /org/bluez/hci0/dev_D2_A3_05_BA_A1_76
2021-10-30 17:11:16:831 DEBUG [Adapter] Device /org/bluez/hci0/dev_48_65_BB_B7_CB_B7 removed
2021-10-30 17:11:16:831 DEBUG [Device] freeing /org/bluez/hci0/dev_48_65_BB_B7_CB_B7
2021-10-30 17:11:16:833 DEBUG [Adapter] Device /org/bluez/hci0/dev_FF_0A_3D_05_5E_26 removed
2021-10-30 17:11:16:834 DEBUG [Device] freeing /org/bluez/hci0/dev_FF_0A_3D_05_5E_26
2021-10-30 17:11:16:841 DEBUG [Main] discovery 'off'
```

My device is the 'FT95'. Directly after finding it, I stop the scan. And as you can see it is immediately removed from the DBUS.

weliem, 2021-11-09:
Any thoughts on this bug?

Vishal8523, 2025-05-15:
Yes, seeing this issue on my end as well. Using BlueZ 5.79
@Vudentz : Any thoughts ?

closed 2025-08-25 by github-actions[bot]