bluez/bluez #228: Fuzzing by way of OSS-Fuzz

DavidKorczynski, 2021-11-04:
Hi!

I was wondering if you would be interested in getting Bluez fuzzed by OSS-Fuzz? Fuzzing is a technique for stress-testing applications and OSS-Fuzz is a service run by Google for fuzzing important open source projects. It would be great to get Bluez integrated with OSS-Fuzz. OSS-Fuzz will run the fuzzers continuously and report back with bug reports when bugs are found. These reports will have full stack traces, input triggers and more (i.e. Sanitizer reports). If you are interested, then the only thing needed is an email(s) connected to a Google account that can be used for receiving the bugs reports, which can be put in the configuration file (project.yaml) over at the OSS-Fuzz repository.

I have set up an initial integration with OSS-Fuzz here: https://github.com/google/oss-fuzz/pull/6752 which includes a simple initial fuzzer targetting sdp code. We can always extend with more fuzzers if you are happy to integrate and I am happy to put efforts into doing this to increase code coverage. I would just need an email and then the OSS-Fuzz should get merged, following that the fuzzing should be starting shortly after.

Vudentz, 2021-12-17:
@DavidKorczynski that would be great, but I wonder if we could integrate something like that directly to our CI so we can run the fuzzing test on each patch submitted, what do you think @tedd-an?

tedd-an, 2021-12-17:
@DavidKorczynski what's the normal running time for one full cycle? It will be nice to integrate with our CI but if not, still running it independently can be beneficial to the project. 
please add "bluez.test.bot@gmail.com" and "hj.tedd.an@gmail.com".

DavidKorczynski, 2021-12-17:
Great! 

I can add integration with the CIFuzz right after integrating with OSS-Fuzz: OSS-Fuzz has built-in capabilities to make this super easy. 
The normal routine is to run both continuously and also during CI!

tedd-an, 2021-12-21:
@DavidKorczynski  Hi David, I have a couple of questions about this. At first, how can I integrate the OSS-Fuzz into the existing BlueZ CI(https://github.com/BluezTestBot/bluez/actions)? I want to run with the patch set submitted to the mailing list if OSS-Fuzz can run from another repository and the execution time is not too long.
Also, I wonder if you can add the test cases for MGMT APIs(https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/mgmt-api.txt) when you have a chance.

DavidKorczynski, 2023-01-02:
Hi @tedd-an -- I realised I never replied here! I have added some fuzzers on the OSS-Fuzz throughout 2022, did you by any chance receive some emails about them?

The easiest may be to add a Github action in https://github.com/BluezTestBot/bluez/tree/workflow/.github/workflows that will run the fuzzers on a pull or push. There is a standard Github action for this that we can use (https://google.github.io/oss-fuzz/getting-started/continuous-integration/)

tedd-an, 2023-01-03:
Hi @DavidKorczynski, I don't remember receiving any email about it but glad to hear from you and will try to enable it.

DavidKorczynski, 2023-01-03:
Strange as I listed your email here: https://github.com/google/oss-fuzz/blob/49c2379eab4dadcacaa2cdb3650cb744774d7dd8/projects/bluez/project.yaml#L6

The emails are usually sent from something e.g. `monorail+v2.1950284618@chromium.org` and the subject starts with something like "Issue XXXX in oss-fuzz: bluez" Can you confirm you did not receive any?

tedd-an, 2023-01-03:
@DavidKorczynski Aha, that's why.
Now I can see the emails.

@Vudentz  Are you ok to send those emails to the linux-bluetooth mailing list instead?

Vudentz, 2023-01-04:
> @DavidKorczynski Aha, that's why. Now I can see the emails.
> 
> @Vudentz Are you ok to send those emails to the linux-bluetooth mailing list instead?

Yes, I think we shall sent them to linux-bluetooth if possible.

tedd-an, 2023-01-04:
@DavidKorczynski Could you update the email to linux-bluetooth@vger.kernel.org?