bluez/bluez #230: Bluetooth headset connection occasionally fails: br-connection-page-timeout

LVH-27, 2021-11-08:
I'm using Arch `5.14.12-arch1-1` on a ThinkPad X1 Carbon. I am connecting my headset (Sennheiser BT 4.40) via `bluetoothctl`, by running `connect <my device>`. 

However, recently I've started getting an error message occasionally when trying to connect. This is the output of `connect`:

    Attempting to connect to 00:16:94:3B:11:A4  
    Failed to connect: org.bluez.Error.Failed br-connection-page-timeout

Usually just restarting the service (`systemctl restart bluetoothd.service`) fixes the problem, but sometimes I have to try a couple of times. It seems that an update to the `bluez` packages (`bluez`, `bluez-libs`, and `bluez-utils`) broke something. I downgraded from `5.62-1` to `5.61-1` a couple of days ago, and I indeed haven't run into the problem since.

But I have no idea what is causing the problem. I've been surprisingly ineffective at googling it. I've yet to notice any patterns among the occurrences, but I've had some issues with suspending this laptop, so it may be possible that it happens only after a sleep or something. But not consistently, so there must be something else at play. Any ideas?

Vudentz, 2021-11-08:
@LVH-27 The page timeout error is down at the Link Layer which is part of the controller firmware so I wonder how downgrading would have made any difference. Anyway it can be that the controller is acting up and needs a reset which happens to be what done when you do a service restart.

LVH-27, 2021-11-10:
@Vudentz I see. I do not know enough to comment on that yet, but the downgrade really seems to have helped. In the amount of time and use since the downgrade, I surely would have hit an error by now, judging from past `5.62-1` behavior. I would hate to be stuck with a specific version of `bluez`. Can you suggest how I can diagnose this further? I'll switch back to `5.62-1` and take a closer look at the logs when I get the time.

LVH-27, 2021-11-11:
After re-enabling `5.62-1`, I soon got the page timeout error, so I took a look at the logs of `bluetooth.service`:

```
src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
profiles/audio/media.c:parse_int64_metadata() expected DBUS_TYPE_INT64 got DBUS_TYPE_UINT64
```

Does this tell you anything? I don't recall seeing this before `5.62-1`, though admittedly I wasn't really combing the logs when everything was working.

Vudentz, 2021-11-30:
Host is down is just the result of Page Timeout error, the metadata is just a warning that some players don't set the data type properly.

kendoori, 2022-03-27:
I too have this issue with Bluez 5.64 on Linux Mint 20.3 5.16.10-051610-generic

I am connecting via Broadcom Corp. BCM20702A0 Bluetooth 4.0 to Costco KS10 hearing aids and they work really well when connected. Sometimes if I walk out of range and try to reconnect, I get this message:
```
kenny@kenny-ThinkPad-T480:~$ bluetoothctl connect 70:66:1B:C1:AE:53
Attempting to connect to 70:66:1B:C1:AE:53
Failed to connect: org.bluez.Error.Failed br-connection-page-timeout
```
The only reliable way to make this message go away is to unpair and reboot the hearing aids and pair again. At that point, they reconnect without any issue or error. I had previously been a much older version of Bluez that shipped with Mint. I had some instability issues then, but not this error.

LVH-27, 2022-03-28:
That sounds very similar to my issue. Once connected, everything is okay, but when I break the connection for whatever reason (suspend, out-of-range, turning off the headset), sometimes this occurs and I have to mess around to get it running.

It was good on `5.61-1`, then it was good on `5.63-1`, and it seems it broke again now on `5.64-2` (or maybe `5.63-2`, judging by the dates in `/var/loc/pacman.log`), as I started seeing the error again recently. Maybe try downgrading to `5.63-1`?

kendoori, 2022-03-28:
> Maybe try downgrading to `5.63-1`?
I had to compile/configure Bluez for Mint because the version that ships with Mint is really old. Any guidance on downgrading?

These are the instructions I used: https://budimir.cc/2022/01/08/compile-bluez-from-source/

Do we think it's just a question of doing this step:
`Now we have all ready to compile BlueZ`

And will that just overwrite the version of Bluez I have installed?

kendoori, 2022-03-28:
> And will that just overwrite the version of Bluez I have installed?

In case anyone needs this, I just recompiled and that was it. I downgraded to 5.63.

LVH-27, 2022-03-29:
Cool! Please report back in a couple of days, I'm interested if the issue persists.

kendoori, 2022-03-29:
> I'm interested if the issue persists.

I tried the version of 5.63 that is here: http://www.bluez.org/release-of-bluez-5-63/ but the issue persisted, so I'm now on 5.62. I will report back.

unixbhaskar, 2022-04-03:
Well, I have bumped onto this very problem and googling doesn't help either. Plus, the lack of knowledge about the underlying stuff makes it even harder to debug. 

I have to face this on Gentoo and Arch ....but as other people pointed it is more of a bluez problem than some distro-specific issue. 

I am trying to connect my wireless earbuds (Boat) and it spits out the error "br-connection-page-timeout" ...

`
[ebuild   R    ] net-wireless/bluez-5.62-r3:0/3::gentoo  USE="-btpclient -cups -debug -deprecated -doc -experimental -extra-tools mesh -midi obex readline (-selinux) -systemd -test -test-programs udev" ABI_X86="-32 (64) (-x32)" PYTHON_SINGLE_TARGET="-python3_8 python3_9 -python3_10" 0 KiB
`

kendoori, 2022-04-03:
@unixbhaskar FWIW, I normally run Linux Mint, but tried on Ubuntu 21.10 and it seemed much more stable. Also, I think a lot depends on your BT card. I also found that an external USB BT 4.0 dongle is more stable than my inbuilt combo WiFi/BT 5 Intel AX210 (pretty new). Right now, Bluez 5.64 with the BT 4 dongle + 5.16.10-051610-generic works well.

unixbhaskar, 2022-04-03:
@kendoori  Well, now the earbuds get connected but there is no sound...bemused! ....Oh btw, I am using x250 Lenovo thinkpad. 

Wondering what stopping the sound get into the earbuds when it is showing everywhere that it is connected?? 

```
bhaskar@Gentoo_03:09:14_Mon Apr 04: :~>bluetoothctl
Agent registered
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 5C:E0:C5:B2:98:7C Discovering: yes
[NEW] Device C8:3D:DC:FE:23:49 C8-4D-DC-FE-23-49
[CHG] Device A9:66:D4:40:B8:77 RSSI: -42
[CHG] Device A9:66:D4:40:B8:77 ManufacturerData Key: 0x7262
[CHG] Device A9:66:D4:40:B8:77 ManufacturerData Value:
  32 32 78 78 11 22 33 44 55 66 aa bb 00 00        22xx."3DUf....
[CHG] Device A9:66:D4:40:B8:77 Connected: yes
```

LVH-27, 2022-04-04:
@unixbhaskar  Hm, that's weird. I suppose you already confirmed that your audio server (Pulseaudio/Pipewire/whatever) is indeed sending sound to the correct output?

A (not-so-)recent `linux-firmware` update broke A2DP, one of Bluetooth audio profiles, so my headset would work, and occasionally all sound would disappear, and after a while (like half a minute), I would get disconnected and then reconnected, and the sound would be back. It would happen every couple of minutes. It's not the same as what you're seeing, but maybe you've hit some weird edge case of that. Here is a [reddit thread](https://www.reddit.com/r/archlinux/comments/sh6eqs/bluetooth_headset_disconnecting_intermittently/) discussing the issue.

I somehow doubt that's your problem, but maybe try switching to a different audio profile (in `pavucontrol` under _Configuration_ or in Blueman) and see what happens.

unixbhaskar, 2022-04-05:
> 

Thanks. In fact, I am using bluealsa and pluseaudio compiled with bluetooth.  For the time being , I have found a dirty fix to this annoyance by doing ......

Bounce bluetooth service 

Bounce bluealsa service 

The above steps are safety checks and reassurance.

Remove device---> add device --> Re pair with the device -->trust the device 

....and stays without getting disconnected or losing sounds.  

Okay, to be precise, I haven't lost sound for a bit almost about 1 hr 15 min of listening with it . 

pavucontrol is of absolutely no help. It doesn't show earbuds integration or other thing related to that. I have checked the stuff from bluemen service manager window.

bryankaplan, 2022-07-25:
As of my Arch upgrade yesterday, I am experiencing this same bug. It is not intermittent.

rsyring, 2022-08-13:
I have the same issue on a fresh upgrade to Ubuntu 22.04.  Happy to provide diagnostic info if helpful.  Thanks.

Vudentz, 2022-08-15:
> I have the same issue on a fresh upgrade to Ubuntu 22.04. Happy to provide diagnostic info if helpful. Thanks.

Please provide the HCI trace (btmon), also what kernel version as well.

rsyring, 2022-08-19:
> Please provide the HCI trace (btmon), also what kernel version as well.

Thanks for your help.  Please see below.

Kernel:
`Linux meld 5.15.0-43-generic #46-Ubuntu SMP Tue Jul 12 10:30:17 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`

Trace [is attached](https://github.com/bluez/bluez/files/9384216/btmon.zip).  The steps I was taking when making the trace are just below.  The numbers on the left correspond to the elapsed seconds numbers in the text output.  Also, the numbers below could be off by +/ ~2 seconds.

> 0: speaker already on, BT device graphics show connected (orange, green, and blue icons) but no audio device from the speaker
> 22: turn off speaker
> 47: turn on speaker
> 73: manual disconnect
> 79: re-connects itself, br-connection-page-timeout error
> 103: another manual disconnect and then it reconnects itself
> 125: manual disconnect and then manual connect, audio device shows up, connected as HFP which isn't normal.  I may have clicked the wrong connection item in the BT Devices menu.  Switched it to A2DP Sink without problem (after the capture stopped).

Vudentz, 2022-08-19:
@rsyring strange, Im not able to find the page timeout error on the logs, perhaps this is self generated by bluetoothd somehow, 5.15 is a bit old as well which perhaps could be causing some compatibility issue with userspace, we should anyway fix the issue since userspace should be backwards compatible.