bluez/bluez #231: Assign Handles for Characteristics

pmysore, 2021-11-08:
According to https://github.com/bluez/bluez/blob/master/doc/gatt-api.txt#L275, we should able to assign handles to Characteristics on the server side. 

uint16 Handle [read-write, optional] (Server Only)

			Characteristic handle. When available in the server it
			would attempt to use to allocate into the database
			which may fail, to auto allocate the value 0x0000
			shall be used which will cause the allocated handle to
			be set once registered.

But the supplied handles don't seem to take effect (or are overridden by bluez). Is there another way to provide a handle to characteristics?

Vudentz, 2021-11-08:
@pmysore Do you have any logs? Is the Handle really visible when doing GetManagedObjects, if it is not the daemon would assume that it should auto assign the first handle available.

Vudentz, 2021-11-08:
It seems there a bug in bluetoothctl that doesn't accept the handle to be entered as hexadecimal but after fixing that it seems to work as expected:

```
[bluetooth]# register-service 0xffff 0x0100
[NEW] Primary Service (Handle 0x0100)
	/org/bluez/app/service0
	0xffff
	Unknown
[/org/bluez/app/service0] Primary (yes/no): yes
[bluetooth]# register-application
bluetoothd[886969]: src/gatt-database.c:database_add_service() handle 0x0100 UUID 0000ffff-0000-1000-8000-00805f9b34fb
```

pmysore, 2021-11-09:
@Vudentz Thanks for looking into it. Was using version 5.5 which didn't have the Add Handle Commit https://github.com/bluez/bluez/commit/0c81f3733eaaa9c18e5b53bf35a7973bf5c9641b#diff-1b74118d7a7d36c799a0a55ca3c886bc2256bf0e86effa6ef83409a3a80e673c

Curiously, I went up to Versions 5.53 which has the above commit but the handles were still getting overridden. Anyway, I upgraded to Version 5.61 and it works fine. Appreciate your help. 

Closing the ticket.

closed 2021-11-09 by pmysore