bluez/bluez #235: bluetoothd is stopping after starting bluetoothctl (bluez 5.62)

NPlaten, 2021-11-12:
Hi there,
after we would like to use the new bluez 5.62, we made an update to it and had to adjust the bluetoothd keeps crashing.

System setup:
Raspberry Pi 4:
- Kernel 5.10
- bluez 5.62

It doesn't take much to reproduce this crash:
- call bluetoothctl
- connect <device>
- disconnect
- exit
- call bluetoothctl (bluetoothd now crashes)

The problem occurs with a Yocto build (Linux Kernel 5.4) and with the raspberry pi. To make it easier, I'll describe how I updated the bluez on the Pi.
New Pi image from October 30th 2021 loaded and the following commands executed:
````shell sudo apt-get install libdbus-1-dev libglib2.0-dev libudev-dev libical-dev libreadline-dev -y
sudo apt-get install python3-docutils -y
wget https://mirrors.edge.kernel.org/pub/linux/bluetooth/bluez-5.62.tar.xz
cd bluez-5.62
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-experimental
make -j4
sudo make install
sudo reboot
````

When I look at the following logs and the post I think these issues might be similar.
Probably related Issue:
[Bluez version 5.62 - Spontaneous restart of the bluetooth daemon after previous successful connection](https://github.com/bluez/bluez/issues/219)

Logs:
[btmon.log](https://github.com/bluez/bluez/files/7527610/btmon.log)
[var_log_messages.log](https://github.com/bluez/bluez/files/7527611/var_log_messages.log)
[bluetoothd-adapter.log](https://github.com/bluez/bluez/files/7527612/bluetoothd-adapter.log)

For me this area sounds very interesting, which only appears when start bluetoothctl again.
```
= bluetoothd: Disconnected from D-Bus. Exiting.                       28.947983
@ MGMT Close: bluetoothd                                     {0x0001} 28.956988
= bluetoothd: Stopping SDP server                                     28.957285
= bluetoothd: Exit                                                    28.957424
```

Vudentz, 2021-11-15:
`= bluetoothd: Disconnected from D-Bus. Exiting.                       28.947983`

This is usually due to some security policy e.g. cannot transfer fd, see:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/HACKING#n68

NPlaten, 2021-11-16:
Hello,
I also tried it with the parameter.

1. remove old bluez and install dependencies
2. `./configure --enable-maintainer-mode`
2. `make` *(1)
3. `cp ./src/bluetooth.conf /etc/dbus-1/system.d/`
4. `./src/bluetoothd -n -d -f ./src/main.conf`

the log from bluetoothd now looks like this:
[src_bluetoothd(2).log](https://github.com/bluez/bluez/files/7545152/src_bluetoothd.2.log)

Unfortunately no change when restarting bluetoothctl, the bluetoothd continues to crash.

---

*(1)
An error occurred while building:
```c
src/adapter.c: In function ‘bt_uuid_hash’:
src/adapter.c:3614:8: error: cast increases required alignment of target type [-Werror=cast-align]
 3614 |  val = (uint64_t *)&uuid_128.value.u128;
````
I quickly repaced this for testing. But I think that shouldn't be the problem here.
```c
uint64_t val_temp = 0;
val = &val_temp;
````

Vudentz, 2021-11-16:
@NPlaten did you disable SELinux with sudo setenforce 0?

hannes-ma, 2021-11-19:
I have been digging a bit into this. 
It seems that the dbus call to GetManagedObjects() after a disconnect leads to the crash:

```
method call time=1637341489.016807 sender=:1.78 -> destination=org.bluez serial=7 path=/; interface=org.freedesktop.DBus.ObjectManager; member=GetManagedObjects
signal time=1637341489.022747 sender=org.freedesktop.DBus -> destination=:1.76 serial=6 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameLost
   string "org.bluez"
signal time=1637341489.022876 sender=org.freedesktop.DBus -> destination=(null destination) serial=8 path=/org/freedesktop/DBus; interface=org.freedesktop.DBus; member=NameOwnerChanged
   string "org.bluez"
   string ":1.76"
   string ""
```

bluetoothctl is executing that call at startup. We have a custom application which is comunicating via dbus to bluez and it happens the exact same thing too when GetManagedObjects() is called after a disconnect.
Just to mention, since there were comments that this could be a problem with dbus message parsing I updated also the dbus package to version 1.12.20 (from previously 1.12.10) but that did not solve anything.

I presume bluez is returning an invalid signature on that call and then dbus-daemon is dropping the connection. But I'm not an expert on dbus...
So any ideas on how to further debug this?

javispedro, 2021-11-25:
Hello, I am hitting exactly the same problem on OpenSUSE (which uses Apparmor rather than SELinux) so I can say it's most definitely not a selinux issue).  :)

I can reproduce it 100% with a custom python script that we have that connects to a specific GATT peripheral, queries a few attributes, and then eventually disconnects and exits.
The first time I invoke the script it runs without problem; however the SECOND time I run the script, and right on the call to GetManagedObjects() on / object,  bluez terminates with "Disconnected from D-Bus. Exiting.".
I see the same dbus-monitor trace as the above comment. 

The problem is definitely a 5.62 regression; reverting to a bluetoothd binary from 5.61 fixes it.

Unfortunately reducing our custom python script to a minimal testcase seems complicated, since it seems that to actually reproduce it we need to read at least some of our peripheral's attributes.  

However I have run a git bisect, and the first faulty commit is:
commit aaa0c4996ae9638be5b97de916acfcf68782430a
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Tue Sep 28 15:45:05 2021 -0700

    gatt: Add implementation of GattCharacteristic1.MTU
    
    This implements MTU property in GattCharacteristic1 interface.

javispedro, 2021-11-25:
Yep, gdb shows that during the first run of the script, whenever characteristic_get_mtu is called, it returns TRUE. 
However, on the second run of the script, bt_gatt_client_get_att(gatt) returns NULL (looks like the dbus object is cached but we are not actually connected yet, so there is no bt_att yet), thus characteristic_get_mtu returns FALSE. We then proceed to send an empty method return to dbus and dbus-daemon kicks us. 

Looks like property getters in gdbus are not supposed to return FALSE. 
The return value is completely ignored and a dbus error cannot be replied by the time the getter is called anyway since gdbus will always send a method reply anyway.

Just returning mtu = 0 on disconnected clients would seem to do the trick. However not sure whether 0 or the default mtu makes more sense.

hannes-ma, 2021-11-26:
@javispedro Perfect analysis!

I've applied this patch and bluez does not terminate anymore. Tested it for a couple of hours with lots of connects/disconnects.

```
diff --git a/src/gatt-client.c b/src/gatt-client.c
index de18bea..e68df8a 100644
--- a/src/gatt-client.c
+++ b/src/gatt-client.c
@@ -884,9 +884,9 @@ static gboolean characteristic_get_mtu(const GDBusPropertyTable *property,
 
 	att = bt_gatt_client_get_att(gatt);
 	if (!att)
-		return FALSE;
-
-	mtu = bt_att_get_mtu(att);
+		mtu = BT_ATT_DEFAULT_LE_MTU;
+	else
+		mtu = bt_att_get_mtu(att);
 
 	dbus_message_iter_append_basic(iter, DBUS_TYPE_UINT16, &mtu);
 
```

closed 2021-11-29 by BluezTestBot