bluez/bluez #236: TrackPoint II Keyboard not autoconnecting after shutdown/resume

l3s2d, 2021-11-13:
- The device is trusted. Other trusted devices autoconnect as expected
- I can reliably trigger a connect by running `scan on` from bluetoothctl

Vudentz, 2021-11-15:
It would be great to have the btmon and bluetoothd traces if possible. Also it could be that the device is using directed advertising using our Public/Identity address which would not match the address we use to passive scan if privacy is enabled.

l3s2d, 2021-11-16:
Thanks for your help! It looks like this was caused by having `Experimental = true` in `/etc/bluetooth/main.conf`. Setting it to `false` has fixed the issue for me.

With `Experimental = true` the behavior was the same between `Privacy` set to `off` or `device`. Would traces still be helpful?

Vudentz, 2021-11-16:
@l3s2d Could this be LL Privacy acting up then, check the btmon logs what address type is being used when passive scan is used since that needs to match the address type if the device is using Directed Advertising.

closed 2021-11-24 by Vudentz

tigerjack, 2022-01-04:
Hi @Vudentz, I had exactly the same problem with my Inphic mouse. I disabled the experimental features (removing `-E` from the systemd unit file). However, in this way, I'm not able to get the battery percentage of my headphones.

If you can expand on what do you need to investigate the issue, I'll be happy to help.

borreeero, 2022-01-19:
Since bluez 5.62, experimental features can be enabled individually from command line or from the config file:
```properties
[General]
# ...

# Enables experimental features and interfaces, alternatively a list of UUIDs
# can be given.
# Possible values: true,false,<UUID List>
# Possible UUIDS:
# d4992530-b9ec-469f-ab01-6c481c47da1c (BlueZ Experimental Debug)
# 671b10b5-42c0-4696-9227-eb28d1b049d6 (BlueZ Experimental Simultaneous Central and Peripheral)
# 15c0a148-c273-11ea-b3de-0242ac130004 (BlueZ Experimental LL privacy)
# 330859bc-7506-492d-9370-9a6f0614037f (BlueZ Experimental Bluetooth Quality Report)
# a6695ace-ee7f-4fb9-881a-5fac66c629af (BlueZ Experimental Offload Codecs)
# Defaults to false.
#Experimental = false
```

The DBus battery reporting gets enabled if there is at least one experimental feature set. Some of those must cause the issue with HID devices not autoconnecting (I've also encountered it with my Razer mouse with all experimental features turned on). My solution/workaround is to only enable one of the features. I've tested only with the Experimental Bluetooth Quality Report and did the trick.
```properties
Experimental = 330859bc-7506-492d-9370-9a6f0614037f
```

Don't forget to restart the bluetooth service afterwards.
```shell
sudo systemctl restart bluetooth.service
```

tigerjack, 2022-01-20:
Hi @borreeero , thanks for the suggestion. I had the experimental enabled only for the battery percentage of the headphones. Do you know if any of them can be used for this purpose?

borreeero, 2022-01-20:
> Hi @borreeero , thanks for the suggestion. I had the experimental enabled only for the battery percentage of the headphones. Do you know if any of them can be used for this purpose?

Yes, try with "BlueZ Experimental Bluetooth Quality Report". You only need to either set Experimental to it's UUID in `main.conf` or start `bluetoothd` with `-E <uuid>`. Hope it works for you.

If I have some time I'll try each one of the experimental options and check which one breaks the Bluetooth keyboard/mice, and then open an issue. Also, I don't know the maturity status of the DBus battery reporting, it would be nice to permanently enable it.

tigerjack, 2022-01-20:
@borreeero even in this case, I always get the same errors. From the log of `bluetooth.service`

```
Jan 20 22:36:01 azenb bluetoothd[383542]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 04:CB:88:C3:1C:76: Host is down (112)
Jan 20 22:36:06 azenb bluetoothd[383542]: sdp_extract_attr: Unknown data descriptor : 0x2 terminating
Jan 20 22:36:08 azenb bluetoothd[383542]: profiles/battery/battery.c:parse_battery_level() Trying to update an unregistered battery
```

while from the output of `blueman-applet`

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/blueman/main/Applet.py", line 73, in _on_device_property_changed
    plugin.on_device_property_changed(path, key, value)
  File "/usr/lib/python3.10/site-packages/blueman/plugins/applet/Battery.py", line 20, in on_device_property_changed
    text = "%d%%" % BluezBattery(obj_path=path)["Percentage"]
  File "/usr/lib/python3.10/site-packages/blueman/bluez/Base.py", line 135, in __getitem__
    return self.get(key)
  File "/usr/lib/python3.10/site-packages/blueman/bluez/Base.py", line 106, in get
    raise parse_dbus_error(e)
blueman.bluez.errors.BluezDBusException: org.freedesktop.DBus.Error.InvalidArgs No such property 'Percentage'
```