bluez/bluez #238: Bluetooth devices get disconnected when Chromium based apps start

elman22, 2021-11-20:
SUMMARY
Sometimes when I start Chromium based apps (Chromium, Chrome) or Electron based app (Typora), my bluetooth devices (speaker and mouse) get disconnected for a short while and then they are reconnected. Issue doesn't occur with Electron app in snap.

STEPS TO REPRODUCE
1. Start Chromium based application.

OBSERVED RESULT
Bluetooth gets disconnected and the reconnected

SOFTWARE/OS VERSIONS
Operating System: Manjaro Linux
KDE Plasma Version: 5.23.3
KDE Frameworks Version: 5.88.0
Qt Version: 5.15.2
Kernel Version: 5.15.2-2-MANJARO (64-bit)
Graphics Platform: X11

ADDITIONAL INFORMATION
This is what I see in logs:

```
nov 19 19:19:05 pc bluetoothd[997]: Disconnected from D-Bus. Exiting.
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/ldac
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSink/aptx_hd
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx_hd
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSink/aptx
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aac
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSink/sbc
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSink/sbc_xq
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/sbc_xq
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/faststream
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/faststream_duplex
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx_ll_1
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx_ll_0
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
nov 19 19:19:05 pc bluetoothd[997]: Endpoint unregistered: sender=:1.79 path=/MediaEndpoint/A2DPSource/sbc
nov 19 19:19:05 pc bluetoothd[997]: dbus[997]: arguments to dbus_message_new_method_call() were incorrect, assertion "destination == NULL || _dbus_check_is_valid_bus_name (destination)" failed in file dbus-message.c line 1364.
nov 19 19:19:05 pc bluetoothd[997]: This is normally a bug in some application using the D-Bus library.
nov 19 19:19:05 pc bluetoothd[997]: profiles/audio/media.c:release_endpoint() Couldn't allocate D-Bus message
nov 19 19:19:05 pc bluetoothd[997]: Stopping SDP server
nov 19 19:19:05 pc bluetoothd[997]: Exit
nov 19 19:19:05 pc systemd[1]: bluetooth.service: Deactivated successfully.
nov 19 19:19:05 pc systemd[1]: bluetooth.service: Consumed 7.585s CPU time.
nov 19 19:19:05 pc systemd[1]: Starting Bluetooth service...
nov 19 19:19:05 pc bluetoothd[184289]: Bluetooth daemon 5.62
nov 19 19:19:05 pc systemd[1]: Started Bluetooth service.
nov 19 19:19:05 pc bluetoothd[184289]: Starting SDP server
nov 19 19:19:05 pc bluetoothd[184289]: Bluetooth management interface 1.21 initialized
```

adriancaruana, 2021-11-23:
I'm seeing a similar issue. I haven't made the association with chromium apps, but sometimes my Bluetooth randomly stops working. This only started happening with the update from `5.61 -> 5.62`. 

SOFTWARE/OS VERSIONS
Operating System: Manjaro Linux
Gnome Version: 41.1
Kernel Version: 5.10.79-1-MANJARO (64-bit)
Graphics Platform: Wayland
Bluez Version: 5.62

Logs:
```
-- Boot cd436f0f5e3744058924464ef40af607 --
Nov 23 19:05:52 aclaptop systemd[1]: Starting Bluetooth service...
Nov 23 19:05:52 aclaptop bluetoothd[1553]: Bluetooth daemon 5.62
Nov 23 19:05:52 aclaptop systemd[1]: Started Bluetooth service.
Nov 23 19:05:52 aclaptop bluetoothd[1553]: Starting SDP server
Nov 23 19:05:52 aclaptop bluetoothd[1553]: Bluetooth management interface 1.18 initialized
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Nov 23 19:06:20 aclaptop bluetoothd[1553]: Endpoint registered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Nov 23 19:17:34 aclaptop bluetoothd[1553]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Device or resource >  ...
Nov 23 19:17:38 aclaptop bluetoothd[1553]: /org/bluez/hci0/dev_38_18_4C_11_69_24/sep1/fd0: fd(42) ready
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Disconnected from D-Bus. Exiting.
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Nov 23 20:36:36 aclaptop bluetoothd[1553]: Endpoint unregistered: sender=:1.77 path=/MediaEndpoint/A2DPSource/sbc
Nov 23 20:36:37 aclaptop bluetoothd[1553]: dbus[1553]: arguments to dbus_message_new_method_call() were incorrect, assertion "destination =>  ...
Nov 23 20:36:37 aclaptop bluetoothd[1553]: This is normally a bug in some application using the D-Bus library.
Nov 23 20:36:37 aclaptop bluetoothd[1553]: profiles/audio/media.c:release_endpoint() Couldn't allocate D-Bus message
Nov 23 20:36:37 aclaptop bluetoothd[1553]: Stopping SDP server
Nov 23 20:36:37 aclaptop bluetoothd[1553]: Exit
Nov 23 20:36:37 aclaptop systemd[1]: bluetooth.service: Deactivated successfully.
Nov 23 20:36:37 aclaptop systemd[1]: bluetooth.service: Consumed 2.271s CPU time.
```

Vudentz, 2021-11-30:
This shall have been fixed by the patch above.

closed 2021-11-30 by Vudentz