bluez/bluez #239: Mesh test join example not

LarsLangholm, 2021-11-23:
I'm trying to run the mesh-test example on a raspberry pi 4 with raspbian OS. However, when I try to run the "join" command, I can't find it with any provisioner apps (meshctl, nRF Mesh etc).

I have enabled
```
    CONFIG_CRYPTO_USER
    CONFIG_CRYPTO_USER_API
    CONFIG_CRYPTO_USER_API_AEAD
    CONFIG_CRYPTO_USER_API_HASH

    CONFIG_CRYPTO_AES
    CONFIG_CRYPTO_CCM
    CONFIG_CRYPTO_AEAD
    CONFIG_CRYPTO_CMAC
```
and rebuilt the kernel (version 5.10.79). 
Bluez is at version 5.62. Configured with:
`./configure --enable-mesh --enable-testing --enable-tools --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var`

Is there something I've missed?

pjp009, 2021-11-24:
The same issue occurred on my end also.
I've tried with [test_join.py](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test/test-join) example.
But can not display the name of the device in the provisioner app (nrf mesh app).
kindly provide us solution.

LLangh, 2021-11-30:
I'm guessing this directly relates to issue #240. Closing this, as I consider the issue answered.

closed 2021-11-30 by LLangh