bluez/bluez #240: from "test-mesh" python script, "Join" command does not work

niravagrawal552, 2021-11-24:
Hello folks,

I am working with Bluez5.62 where I need to implement python code for ble-mesh node.

While executing "join" command in test-mesh script, I get below log,

> ./test-mesh    
> Register OnOff Server model on element 0
> OnOff Server 
> State is OFF
> Register Vendor model on element 0
> Register OnOff Client model on element 1
> OnOff Client
> *** MAIN MENU ***
> token  - set node ID (token)
> join  - join mesh network
> attach  - attach mesh node
> remove  - delete node
> dest  - set destination address
> uuid  - set remote uuid
> app-index  - set AppKey index
> vendor-send  - send raw vendor message
> client-menu  - On/Off client menu
> quit  - exit the test
> 
**> join
> Joining with UUID 76d309be5da54d07ad6d4f872a34cd0c
> Join procedure started**

It says like joining started but there is no advertising beacon found in scanner application (nrf mesh mobile-app).
I also analyzed dbus system bus transaction. Please see attached dbus_log file.
[mesh-join-dbus.log](https://github.com/bluez/bluez/files/7594158/mesh-join-dbus.log)

From dbus transaction, i understood that mesh-service returns node information to the application though mesh advertising not started by adapter.

While checking debug log of bluetooth-mesh application, i found below log appears whenever I execute "join" call,

> v 24 08:19:19 centauri-imx6ull bluetooth-meshd[272]: Loading node configuration from /var/lib/bluetooth/mesh
> Nov 24 08:19:19 centauri-imx6ull bluetooth-meshd[272]: Loading configuration from /var/lib/bluetooth/mesh/fe1c1b5897d24c8085961540a7281173/nod
> e.json
> Nov 24 08:19:19 centauri-imx6ull bluetooth-meshd[272]: Failed to read IV index
> Nov 24 08:19:19 centauri-imx6ull bluetooth-meshd[272]: Loading configuration from /var/lib/bluetooth/mesh/fe1c1b5897d24c8085961540a7281173/nod
> e.json.bak
> Nov 24 08:19:19 centauri-imx6ull systemd[1]: Started Bluetooth mesh service.
> Nov 24 08:19:21 centauri-imx6ull bluetooth-meshd[272]: mesh_ready_callback
> Nov 24 08:19:21 centauri-imx6ull bluetooth-meshd[272]: Added Network Interface on /org/bluez/mesh
> Nov 24 08:19:21 centauri-imx6ull bluetooth-meshd[272]: Hci dev 0000 removed
> Nov 24 08:20:04 centauri-imx6ull bluetooth-meshd[272]: Hci dev 0001 removed
> Nov 24 08:21:23 centauri-imx6ull bluetooth-meshd[272]: **LE Scan disable failed (0x0c)**

From above log, I am suspecting about HCI transaction if HCI controller does not executed "disable scan" request and not starting advertising on "pb_adv_reg()" api call.

Please do needful to fix "Join" related issue on test-mesh python script as soon as  possible.

Regards,
Nirav Agrawal

inga-s, 2021-11-29:
Hi Nirav,

The "LE Scan disable failed" warning is pesky, but inconsequential and should not affect the mesh operation.

I believe the issue is that the NRF mesh scanner app is only looking for the unprovisioned nodes that support mesh proxy (GATT Provisioning bearer).
The BlueZ based implementation only supports Advertising Provisioning bearer so you won't be able to see the unprovisioned device beacon using the scanner app.
Assuming we are talking about the same app, the "Add node" option displays "Looking for unprovisioned proxy nodes..." which is an indication that the proviosioning is expected to be conducted over GATT.

niravagrawal552, 2021-11-30:
> Hi Nirav,
>
> The "LE Scan disable failed" warning is pesky, but inconsequential and should not affect the mesh operation.
>
> I believe the issue is that the NRF mesh scanner app is only looking for the [...]

Hi inga-s, 

Many thanks for your quick findings. 

I am using nordic's "nRF Mesh 3.1.4" mobile application where below screen appears at the time of initiating scan.
[image]

I did try some other mesh devices (sensors/light-controllers)to join through "nRF Mesh 3.1.4" app which provisioned successfully.  Does it mean those devices advertising beacons on gatt-proxy interface?

If Bluez5.62 does not support mesh-proxy (which will be required in our case to provision our device in mobile-app) then how we can enable our device to be scannable/provisionable on mobile-app ?

I found below advertising data using "hcidump -x" on device to capture advertise-data packet.
```
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  15 14 2B 00 DA 6B 95 6F E7 30 43 1F 8A 42 7C 6A 1F CC 0C 02 
  00 01 00 00 00 00 00 00 00 00 00 00 

```
In above data, 
0x2b = mesh_ad_type beacon
0x00= un provisioned beacon 
da6b956fe730431f8a427c6a1fcc0c02 = device uuid
0x0001 = oob data

I would be very grateful if you can give some insights to deal with this issue?

inga-s, 2021-12-01:
In order to provision a new device over APB, one can use mesh-cfgclient tool.
There's a brief README file in tools/mesh that describes what the tool does.
mesh-cfgclient can be used to generate a mesh network from scratch and add new devices. 
After the provisioning, the tool provides the full functionality of mesh configuration client model and can be used to configure a node.

niravagrawal552, 2021-12-01:
> 

Hi inga-s,

I highly appreciated your valuable inputs on this issue and seeking your help in understanding the pb-gatt support for mesh service.

In our case, we will be going to use mobile-application as provisioner (on PB-GATT bearer). Hence, we require to use PB-GATT bearer on bluez side for our mesh-application to advertise and join with provisioner.

I saw that meshctl is using pb-gatt bearer but it uses bluetoothd deamon (not bluetooth-meshd). is there any way to integrate pb-gatt bearer support in bluetooth-meshd deamon so that end application can connect to mesh service over dbus and we can leverage some part of code from bluetoothd ?

I apologies if i misinterpreted code or missing any configuration to add pb-gatt support in bluetooth-meshd service.

Regards,
Nirav Agrawal

inga-s, 2021-12-01:
Hi Nirav,

You are not misinterpreting. The current mesh implementation **does not** support proxy server: this is what would be required on the unprovisioned device side to be able to be provisioned by a mobile app such as NRF Connect.

meshctl tool implements proxy client and was initially used a quick way to create a mesh network when BT Mesh spec was just released and there were hardly any true PB ADV capable devices out there. Since then, the work on the tool essentially stopped and the efforts were concentrated on developing bluetooth-meshd daemon with "true" mesh support, i.e., over mesh advertising.

We'll consider raising the priority on adding proxy server support into bluetooth-meshd. It is predicated on other modifications that need to happen/propagate in BlueZ. 

Unfortunately, there's no immediate solution to the issue if your intention is to use a PB GATT based mobile app for a provisioner.

Regards,
Inga

niravagrawal552, 2021-12-02:
> Hi Nirav,
>
> You are not misinterpreting. The current mesh implementation **does not** support proxy server: this is what would be required on the unprovisioned device side to be able to be [...]

Hello Inga,

Thank you for your detailed information. I now understand that the support for "PB-GATT" will be added later in bluez for proxy-server in bluetooth-meshd deamon.

Now this seems blocker for us as we needed to use bluez in our project.

Is there any way to skip BLE mesh provisioning for our device but the device can still connect with mesh network using database (which can be imported from mobile-app provisioner). I intent to say here is , can device become a mesh-node using "import" command (If all required param. fetched from provisioning database which will have one dummy node entry)  and communicate with mesh network?

Please give your suggestion if it is possible to skip provisioning for a device but the device can still be a part of mesh-network by any mean?

Regards,
Nirav Agrawal

inga-s, 2021-12-03:
Hi Nirav,

Yes, a node can be added to mesh network using Import() command. The command is exactly for the situation when a node has been provisioned elsewhere. It's a "barebones" operation which adds a node to a network: if there's any node configuration performed by a config client, it's not being imported, and the node should be configured (e.g., ttl, subscriptions and such) from scratch.

Hopefully, this can work for you.

Regards,
Inga

niravagrawal552, 2021-12-11:
Hello Inga, 

Thank to so much for your kind support. It solved our provisioning issue. 

Now, I am facing some strange issue which i raised in #249. This is related to node-configuration and controlling other server nodes from our local-node. 

Can you please go through it and let me know if you came across or knows the reason behind the issue?

Again thanking you for your support.

Regards,
Nirav

inga-s, 2022-01-07:
Closing the issue: solved as indicated by the last entry.

closed 2022-01-07 by inga-s