bluez/bluez #241: Segmentation fault in `bluetoothctl select`

hurricup, 2021-11-30:
- Happens in Home Assistant docker container: https://github.com/home-assistant/core/pkgs/container/home-assistant

- Host: `uname -a`: 
```
Linux jethub-j80 5.9.16-jethubj80 #trunk.5 SMP PREEMPT Wed Apr 21 12:34:06 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

- Host: `lsb_release -a` 

```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
```

- `file core`

`ELF 64-bit LSB core file, ARM aarch64, version 1 (SYSV), SVR4-style, from 'bluetoothctl select F0:A3:B2:46:F0:4B', real uid: 0, effective uid: 0, real gid: 0, effective gid: 0, execfn: '/usr/bin/bluetoothctl', platform: 'aarch64'`

- core dumps: [core.zip](https://github.com/bluez/bluez/files/7623239/core.zip)

Let me know if you need more information

Vudentz, 2021-11-30:
What is segfauling? bluetoothctl or bluetoothd? Also try running with valgrind to capture the backtrace when the problem occurs.

hurricup, 2021-11-30:
I'm not very experienced with diagnostics of such problems, so I need some guidance here.

All I have, these core files. Doesn't `file core` output states that `bluetoothctl` is failing?

Any guidance on how to use the valgrind for this? To clarify: i can't reproduce this intentionally. It happens at some point of long running service (home assistant), which does a lot of things, including running `bluetoothctl`, obviously.

hurricup, 2021-12-08:
Most likely this is because of docker. In general or particular HASS image, not sure. But after I moved to direct running of the same stuff on the host machine - no core dumps or BT hangs so far. For few days.

closed 2021-12-21 by Vudentz