bluez/bluez #242: mesh-cfgclient: node-reset causes crash

LLangh, 2021-11-30:
I tried to remove a node with mesh-cfgclient:

```
[config: Target = 00ab]# node-reset
Received NodeResetStatus (len 0)
Node 00ab is reset
[config: Target = 00ab]# mesh-cfgclient: symbol lookup error: mesh-cfgclient: undefined symbol: json_object_array_del_idx
```

Then it crashes. 
C-json version 0.13

FBNH, 2021-12-13:
this happens to me with unbind & export- db as well.
I have the same JSON-C version.

Should I use a different version?

carminee, 2021-12-17:
I got the same error message, and then I tried different versions of c-json（version 0.15 and  master branch of c-json)，but the result did not change

inga-s, 2022-01-26:
Could you please make sure that there are no remnants of the previous version of json-c lingering on your system?
On raspberry Pi at least, I encountered ver 0.12 sticking around even though I explicitly installed ver 0.14.

closed 2025-08-24 by github-actions[bot]