bluez/bluez #245: 5.62: test suite is failing wioth SIGSEGV

kloczek, 2021-12-04:
```console
+ /usr/bin/make -O -j12 V=1 VERBOSE=1 check
/usr/bin/make --no-print-directory check-am
/usr/bin/make --no-print-directory check-TESTS
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o src/sdpd-database.o src/sdpd-database.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o src/textfile.o src/textfile.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-avctp.o unit/test-avctp.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/util.o unit/util.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o src/sdpd-request.o src/sdpd-request.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o src/sdpd-service.o src/sdpd-service.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-gobex-header.o unit/test-gobex-header.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o src/eir.o src/eir.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-gdbus-client.o unit/test-gdbus-client.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-gobex-packet.o unit/test-gobex-packet.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-hfp.o unit/test-hfp.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fdata-sections -ffunction-sections -flto=auto -flto-partition=none -Os -c -o unit/test-gobex-apparam.o unit/test-gobex-apparam.c
/usr/bin/gcc -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include  -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -pthread -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include  -I./lib        -O2 -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack- [... cut ...]
        data = {test_name = 0x55ab508adce0 "/TP/HGRF/RH/BV-01-I", pdu_list = 0x55ab508add00}
        data = {test_name = 0x55ab508ae300 "/TP/HGRF/RH/BV-08-I", pdu_list = 0x55ab508ae320}
        data = {test_name = 0x55ab508ae800 "/TP/HGRF/RH/BV-09-I", pdu_list = 0x55ab508ae820}
        data = {test_name = 0x55ab508aed60 "/TP/HGRF/RH/BV-06-I", pdu_list = 0x55ab508aed80}
        data = {test_name = 0x55ab508af560 "/TP/HGCF/RH/BV-01-I", pdu_list = 0x55ab508af580}
        data = {test_name = 0x55ab508afd20 "/TP/HGRF/RH/BV-02-I", pdu_list = 0x55ab508afd40}
(gdb)
```

Vudentz, 2021-12-08:
Im not able to reproduce with current git:

```
/TP/HGRF/RH/BV-01-I - init
/TP/HGRF/RH/BV-01-I - setup
/TP/HGRF/RH/BV-01-I - setup complete
/TP/HGRF/RH/BV-01-I - run
  hog: > 10 01 00 ff ff 00 28                             ......(         
  hog: < 11 06 01 00 04 00 12 18 05 00 08 00 12 18        ..............  
  hog: > 10 09 00 ff ff 00 28                             ......(         
  hog: < 01 10 09 00 0a                                   .....           
  hog: > 08 01 00 04 00 03 28                             ......(         
  hog: < 09 07 03 00 02 04 00 4b 2a                       .......K*       
  hog: > 08 01 00 04 00 02 28                             ......(         
  hog: < 01 08 01 00 0a                                   .....           
  hog: > 08 05 00 08 00 02 28                             ......(         
  hog: < 01 08 05 00 0a                                   .....           
  hog: > 08 05 00 08 00 03 28                             ......(         
  hog: < 09 07 07 00 02 08 00 4b 2a                       .......K*       
  hog: > 0a 04 00                                         ...             
  hog: < 0b 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13 14 15 16                             .......         
  hog: > 0c 04 00 16 00                                   .....           
  hog: < 0d 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13 14 15 16                             .......         
  hog: > 0c 04 00 2c 00                                   ...,.           
  hog: < 0d 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13                                      ....            
  hog: > 0a 08 00                                         ...             
  hog: < 0b 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13 14 15 16                             .......         
  hog: > 0c 08 00 16 00                                   .....           
  hog: < 0d 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13 14 15 16                             .......         
  hog: > 0c 08 00 2c 00                                   ...,.           
  hog: < 0d 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ................
  hog:   10 11 12 13                                      ....            
/TP/HGRF/RH/BV-01-I - test passed
```

kloczek, 2021-12-08:
Can I do something to try to diagnose that issue in my build env?

Vudentz, 2021-12-08:
@kloczek what version are you using btw, this looks like a mismatch of the unit test and gatt-client internals, so I wonder if you had cherry-picked some patches?

kloczek, 2021-12-08:
I'm usimg few patches from Fedora:
```spec
Patch00:        %{name}-obex-Use-GLib-helper-function-to-manipulate-paths.patch
Patch01:        %{name}-build-Always-define-confdir-and-statedir.patch
Patch02:        %{name}-systemd-Add-PrivateTmp-and-NoNewPrivileges-options.patch
Patch03:        %{name}-systemd-Add-more-filesystem-lockdown.patch
Patch04:        %{name}-systemd-More-lockdown.patch
Patch05:        %{name}-Use-g_memdup2-everywhere.patch
Patch06:        %{name}-Revert-attrib-Make-use-of-bt_att_resend.patch
```

kloczek, 2021-12-08:
(I should mention that on oppening the ticket .. sorry)

Vudentz, 2021-12-13:
@kloczek Revert-attrib-Make-use-of-bt_att_resend.patch that explains it, now why does the fedora has that patch I don't know.

kloczek, 2021-12-13:
Gosh .. good catch :)
Thank you for advise. Just tested without that patch and everything is OK now.
```console
============================================================================
Testsuite summary for bluez 5.62
============================================================================
# TOTAL: 26
# PASS:  26
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```
BTW did you have look on other patches used by fedora? https://src.fedoraproject.org/rpms/bluez/tree/rawhide
Maybe some of those patches can be merged into upstream?

One more time .. thank you.
I'm closing that ticket.

closed 2021-12-13 by kloczek