bluez/bluez #246: 5.64: autoconf 2.72 warnings

kloczek, 2021-12-04:
```console
+ autoreconf -fiv
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force
autoreconf: running: /usr/bin/autoconf --force
configure.ac:19: warning: The macro `AC_LANG_C' is obsolete.
configure.ac:19: You should run autoupdate.
./lib/autoconf/c.m4:72: AC_LANG_C is expanded from...
configure.ac:19: the top level
configure.ac:33: warning: The macro `AC_PROG_LIBTOOL' is obsolete.
configure.ac:33: You should run autoupdate.
aclocal.m4:121: AC_PROG_LIBTOOL is expanded from...
configure.ac:33: the top level
configure.ac:45: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:45: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
acinclude.m4:32: MISC_FLAGS is expanded from...
configure.ac:45: the top level
configure.ac:47: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:47: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:47: the top level
configure.ac:86: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:86: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:86: the top level
configure.ac:99: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:99: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:99: the top level
configure.ac:112: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:112: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:112: the top level
configure.ac:125: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:125: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:125: the top level
configure.ac:137: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:137: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:137: the top level
configure.ac:151: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:151: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:151: the top level
configure.ac:155: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:155: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:155: the top level
configure.ac:159: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:159: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:159: the top level
configure.ac:163: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:163: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:163: the top level
[... 108 lines cut ...]
configure.ac:363: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:363: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:363: the top level
configure.ac:367: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:367: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:367: the top level
configure.ac:401: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:401: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:401: the top level
configure.ac:423: warning: The macro `AC_HELP_STRING' is obsolete.
configure.ac:423: You should run autoupdate.
./lib/autoconf/general.m4:204: AC_HELP_STRING is expanded from...
configure.ac:423: the top level
configure.ac:440: warning: AC_OUTPUT should be used without arguments.
configure.ac:440: You should run autoupdate.
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:21: installing './compile'
configure.ac:5: installing './missing'
Makefile.am: installing './depcomp'
autoreconf: Leaving directory '.'
```

Vudentz, 2021-12-20:
@kloczek this is really bad call from autoconf then because it seems that some of its function got renamed/replaced but if we make the changes like:

https://github.com/curl/curl/commit/a59f04611629f0db9ad8e768b9def73b9b4d9423

But that means that we wouldn't be compatible with < 2.59, @holtmann any thoughts on this?

kloczek, 2021-12-27:
autoconf 2.59 has been released **+19 years ago**!!!
Why do you want to be compatible with something so ancient?

Vudentz, 2022-01-05:
@kloczek hmm good point, I haven't checked how far back was autoconf 2.59 if that is so far back then a simple s/AC_HELP_STRING/AS_HELP_STRING should make it go away.

kloczek, 2022-01-06:
> simple s/AC_HELP_STRING/AS_HELP_STRING should make it go away.

yep :)

closed 2022-01-12 by BluezTestBot