bluez/bluez #247: Double button click to activate function

Wiedzmin89, 2021-12-06:
Media functions play/pause need to be activated twice to actually play/pause.
Other function, like previous or next, working fine.

ERTM disabled
Worked OK in https://github.com/tedd-an/bluez/pull/1251

bluez-git 1:5.62.r78.gaa361e221-1
Operating System: Arch Linux
KDE Plasma Version: 5.23.4
KDE Frameworks Version: 5.88.0
Qt Version: 5.15.2
Kernel Version: 5.10.83-1-lts (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i5-8250U CPU @ 1.60GHz
Memory: 15.5 ГиБ of RAM
Graphics Processor: Mesa Intel® UHD Graphics 620

[bluetoothd.txt](https://github.com/bluez/bluez/files/7660547/bluetoothd.txt)
[btmon.txt](https://github.com/bluez/bluez/files/7660548/btmon.txt)

Vudentz, 2021-12-06:
For keys presses we just report them to the input subsystem so I wonder if there is anything wrong with the driver or something else is registering 2 events. Also btmon logs seem to be taken after the connection has been made, so it doesn't should the PSM (PSM 0) and because of that it is unable to decode the protocol properly.

Wiedzmin89, 2021-12-06:
Maybe you misunderstood me, the problem is not that the pause works twice, but that I need to press the button twice for the pause to work.
It seems that there is no separate driver for my headphones. (Sony WF-1000XM3)
I also mentioned what it worked properly in https://github.com/tedd-an/bluez/pull/1251

Vudentz, 2021-12-07:
Please capture the btmon logs from the start of the connection, also enable debug logs with bluetoothd -d.

Wiedzmin89, 2021-12-08:
Seems like only play event works, pause event is not processed by the system
[image]
[bluetoothd.txt](https://github.com/bluez/bluez/files/7677901/bluetoothd.txt)

closed 2021-12-08 by Wiedzmin89