bluez/bluez #250: Mesh friend node: SeqNum not increased for segmented messages to Low Power node

isakwestin, 2021-12-17:
Hi,

I am running mesh-cfgclient (BlueZ version 5.62) with friend feature enabled. I also have a Low Power node (nRF52840-DK board running the nordic LPN example, nRF5 SDK for Mesh v5.0.0). I can successfully establish a friendship between the two, so that the bluetooth-meshd starts using the friend queue for messages targeting the Low Power node.

In the logs I can notice that when I send a segmentet message to the Low Power node, the SeqNum of the Network PDU stays the same for all segments.

From `bluetooth-meshd -nd`:
```
mesh/node.c:dev_key_send_call() DevKeySend
mesh/net.c:enqueue_friend_pkt() enqueue_friend_pkt for 00ac from 0001 ttl: 7f (seq: 000411) (ctl: 0)
mesh/util.c:print_packet() 53738.924 RX: Network [enc] :: 23fe76fe53900fe63a1e9cfd8c1411a8bdde54
mesh/util.c:print_packet() 53738.924 RX: Network [clr] :: 238000e00500ac00010101
mesh/net.c:msg_in_cache() Add 00ac + 00e005 + 00e005ac
mesh/net.c:msg_in_cache() Remove 00ac + 00c016 + 00c016ac
mesh/net.c:packet_received() RX: Network 00ac -> 0001 : TTL 0x00 : IV : 00000000 SEQ 0x00e005
mesh/net.c:packet_received() RX: App 0x00ac -> 0x0001 : TTL 0x00 : SEQ 0x00e005
mesh/net.c:packet_received() CTL - 0000 RX
mesh/util.c:print_packet() 53738.925 Rx-NET_OP_FRND_POLL: 01
mesh/friend.c:friend_poll() POLL-RXED
mesh/friend.c:friend_delay_rsp() Fwd FRND pkt 000411
mesh/util.c:print_packet() 53739.025 Frnd-Msg: 634f7ffcd8cf5fbb27a25d09
mesh/net.c:mesh_net_send_seg() SEQ: 000411
mesh/net.c:mesh_net_send_seg() SEQ0: 000411
mesh/net.c:mesh_net_send_seg() segO: 0
mesh/net.c:mesh_net_send_seg() TX: Friend Seg-0 0001 -> 00ac : len 29) : TTL 127 : SEQ 000411
mesh/util.c:print_packet() 53739.026 TX: Friend: 23e3349212c9e3e78bd0cbfe3202466cabfe1e5a479245cb50377ce5fd
mesh/util.c:print_packet() 53739.027 RX: Network [enc] :: 23e3349212c9e3e78bd0cbfe3202466cabfe1e5a479245cb50377ce5fd
mesh/util.c:print_packet() 53739.027 RX: Network [clr] :: 237f000411000100ac80904401634f7ffcd8cf5fbb27a25d09
mesh/util.c:print_packet() 53739.199 RX: Network [enc] :: 23e64c5df4e6b53c6ce5889c7fcb59b7dadea2
mesh/util.c:print_packet() 53739.199 RX: Network [clr] :: 238000e00600ac00010100
mesh/net.c:msg_in_cache() Add 00ac + 00e006 + 00e006ac
mesh/net.c:msg_in_cache() Remove 00ac + 00c017 + 00c017ac
mesh/net.c:packet_received() RX: Network 00ac -> 0001 : TTL 0x00 : IV : 00000000 SEQ 0x00e006
mesh/net.c:packet_received() RX: App 0x00ac -> 0x0001 : TTL 0x00 : SEQ 0x00e006
mesh/net.c:packet_received() CTL - 0000 RX
mesh/util.c:print_packet() 53739.200 Rx-NET_OP_FRND_POLL: 00
mesh/friend.c:friend_poll() POLL-RXED
mesh/friend.c:friend_delay_rsp() Fwd FRND pkt 000411
mesh/util.c:print_packet() 53739.300 Frnd-Msg: 3ed4eb5292afbef2
mesh/net.c:mesh_net_send_seg() SEQ: 000412
mesh/net.c:mesh_net_send_seg() SEQ0: 000411
mesh/net.c:mesh_net_send_seg() segO: 1
mesh/net.c:mesh_net_send_seg() TX: Friend Seg-1 0001 -> 00ac : len 25) : TTL 127 : SEQ 000411
mesh/util.c:print_packet() 53739.301 TX: Friend: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.302 RX: Network [enc] :: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.302 RX: Network [clr] :: 237f000411000100ac809044213ed4eb5292afbef2
mesh/util.c:print_packet() 53739.554 RX: Network [enc] :: 238f90d8a0a8481edcfe979ca0f87c9c8b308e
mesh/util.c:print_packet() 53739.555 RX: Network [clr] :: 238000e00700ac00010100
mesh/net.c:msg_in_cache() Add 00ac + 00e007 + 00e007ac
mesh/net.c:msg_in_cache() Remove 00ac + 00c018 + 00c018ac
mesh/net.c:packet_received() RX: Network 00ac -> 0001 : TTL 0x00 : IV : 00000000 SEQ 0x00e007
mesh/net.c:packet_received() RX: App 0x00ac -> 0x0001 : TTL 0x00 : SEQ 0x00e007
mesh/net.c:packet_received() CTL - 0000 RX
mesh/util.c:print_packet() 53739.555 Rx-NET_OP_FRND_POLL: 00
mesh/friend.c:friend_poll() POLL-RXED
mesh/friend.c:friend_delay_rsp() Fwd FRND pkt 000411
mesh/util.c:print_packet() 53739.656 Frnd-Msg: 3ed4eb5292afbef2
mesh/net.c:mesh_net_send_seg() SEQ: 000412
mesh/net.c:mesh_net_send_seg() SEQ0: 000411
mesh/net.c:mesh_net_send_seg() segO: 1
mesh/net.c:mesh_net_send_seg() TX: Friend Seg-1 0001 -> 00ac : len 25) : TTL 127 : SEQ 000411
mesh/util.c:print_packet() 53739.657 TX: Friend: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.657 RX: Network [enc] :: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.658 RX: Network [clr] :: 237f000411000100ac809044213ed4eb5292afbef2
mesh/util.c:print_packet() 53739.904 RX: Network [enc] :: 234dadddf6c061b931e262132135b86666d40d
mesh/util.c:print_packet() 53739.904 RX: Network [clr] :: 238000e00800ac00010100
mesh/net.c:msg_in_cache() Add 00ac + 00e008 + 00e008ac
mesh/net.c:msg_in_cache() Remove 00ac + 00c019 + 00c019ac
mesh/net.c:packet_received() RX: Network 00ac -> 0001 : TTL 0x00 : IV : 00000000 SEQ 0x00e008
mesh/net.c:packet_received() RX: App 0x00ac -> 0x0001 : TTL 0x00 : SEQ 0x00e008
[... 82 lines cut ...]
mesh/util.c:print_packet() 53739.027 RX: Network [enc] :: 23e3349212c9e3e78bd0cbfe3202466cabfe1e5a479245cb50377ce5fd
mesh/util.c:print_packet() 53739.027 RX: Network [clr] :: 237f *000411* 000100ac80904401634f7ffcd8cf5fbb27a25d09
                                                                  ^ Seg-0
```

```
mesh/util.c:print_packet() 53739.200 Rx-NET_OP_FRND_POLL: 00
mesh/friend.c:friend_poll() POLL-RXED
mesh/friend.c:friend_delay_rsp() Fwd FRND pkt 000411
mesh/util.c:print_packet() 53739.300 Frnd-Msg: 3ed4eb5292afbef2
mesh/net.c:mesh_net_send_seg() SEQ: 000412
mesh/net.c:mesh_net_send_seg() SEQ0: 000411
mesh/net.c:mesh_net_send_seg() segO: 1
mesh/net.c:mesh_net_send_seg() TX: Friend Seg-1 0001 -> 00ac : len 25) : TTL 127 : SEQ 000411
mesh/util.c:print_packet() 53739.301 TX: Friend: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.302 RX: Network [enc] :: 2321b1bcc8a042e78bd0cbfe125fddf805b47ebb0e668dd52d
mesh/util.c:print_packet() 53739.302 RX: Network [clr] :: 237f *000411* 000100ac809044213ed4eb5292afbef2
                                                                  ^ Seg-1
```

I already [posted a question on the nordic forum about this](https://devzone.nordicsemi.com/f/nordic-q-a/82617/mesh-low-power-node-network-layer-dropping-segmented-mesh-packages). From the logs of the Low Power node they could verify that the segments are actually sent with the same SeqNum.

This eventually results in the Low Power node dropping the second segment (since it has the same SeqNum, according to Mesh specification) and the message never goes through.

I would be happy for any help to figure out where this issue might come from.

bgix, 2022-01-18:
I think I can identify the bug here...
In .../mesh/net.c : friend_packet_queue():
	if (IS_SEGMENTED(hdr)) {
		uint32_t seqAuth = seq_auth(seq, hdr >> SEQ_ZERO_HDR_SHIFT);
		uint8_t i;

		for (i = 0; i <= seg_max; i++) {
			memcpy(frnd_msg->u.s12[i].data, data, 12);
			frnd_msg->u.s12[i].hdr = hdr;
			frnd_msg->u.s12[i].seq = seqAuth + i;
			data += 12;
			hdr += (1 << SEGO_HDR_SHIFT);
		}

		**frnd_msg->u.s12[seg_max].seq = seq;**    <----- seqZero stored as last sequence number
		frnd_msg->cnt_in = seg_max;
		frnd_msg->last_len = size % 12;
		if (!frnd_msg->last_len)
			frnd_msg->last_len = 12;

I think the highlighted line can be entirely deleted, because the correct sequence number should have been stored within the preceding for loop.

isakwestin, 2022-01-19:
Hi, Thank you for looking into it!

Would you create a fix for this if it is a bug? Or do you prefer if I do it? (I have not contributed before but I would be willing to try).

closed 2022-06-29 by isakwestin