bluez/bluez #251: MPOW H19 handsfree head profile, no sound from video.

mauzil, 2021-12-18:
Hi all.
I connected the headphone to my pc.
The bt adapter is equiped with 

 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio

I start with H19 in a2dp_sink profile and I hear the sound of video.
When I switch to handsfree head profile the sound stops and the video slow down.
I set to a2dp_sink profile and all works

I'm using bluez 5.62, kernel 5.15.8 

This strange behaviour happens when I update the kernel (maybe from 5.14 to 5.15, but I don't remember)

From pavucontrol I get this
[image]

Best regards,
  MZ

Vudentz, 2021-12-20:
@mauzil what sound server is this? PulseAudio afaik doen't support mSBC and if you have support for that means code negotiation needs to be in place. Also please have btmon and bluetoothd logs attached.

mauzil, 2021-12-26:
Ok. I listen bluetooth with btmon
I attach a session where i switch from a2dp_sink to handsfree_head_unit
[btmon.log](https://github.com/bluez/bluez/files/7776842/btmon.log)

With command pacmd list-cards I get this for MPOW 

index: 2
	name: <bluez_card.34_75_63_01_21_05>
	driver: <module-bluez5-device.c>
	owner module: 26
	properties:
		device.description = "MPOW H19 IPO"
		device.string = "34:75:63:01:21:05"
		device.api = "bluez"
		device.class = "sound"
		device.bus = "bluetooth"
		device.form_factor = "headphone"
		bluez.path = "/org/bluez/hci0/dev_34_75_63_01_21_05"
		bluez.class = "0x240418"
		bluez.alias = "MPOW H19 IPO"
		device.icon_name = "audio-headphones-bluetooth"
		bluetooth.codec = "sbc"
	profiles:
		a2dp_sink: Riproduzione ad alta fedeltà (sink A2DP) (priority 40, available: yes)
		a2dp_source: Registrazione ad alta fedeltà (sorgente A2DP) (priority 20, available: no)
		handsfree_head_unit: Unità handsfree head (HFP) (priority 30, available: unknown)
		headset_head_unit: Unità headset head (HSP) (priority 30, available: no)
		off: Spento (priority 0, available: yes)
	active profile: <a2dp_sink>
	sinks:
		bluez_sink.34_75_63_01_21_05.a2dp_sink/#15: MPOW H19 IPO
	sources:
		bluez_sink.34_75_63_01_21_05.a2dp_sink.monitor/#23: Monitor of MPOW H19 IPO
	ports:
		headphone-output: Cuffie (priority 0, latency offset 0 usec, available: yes)
			properties:				
		headphone-input: Ingresso Bluetooth (priority 0, latency offset 130000 usec, available: unknown)
			properties:

Vudentz, 2021-12-27:
```
< HCI Command: Setup Synchronous... (0x01|0x0028) plen 17  #589 [hci0] 6.529912
        Handle: 3
        Transmit bandwidth: 8000
        Receive bandwidth: 8000
        Max latency: 13
        Setting: 0x0003
          Input Coding: Linear
          Input Data Format: 1's complement
          Input Sample Size: 8-bit
          # of bits padding at MSB: 0
          Air Coding Format: Transparent Data
        Retransmission effort: Optimize for link quality (0x02)
        Packet type: 0x0380
          3-EV3 may not be used
          2-EV5 may not be used
          3-EV5 may not be used
> HCI Event: Command Status (0x0f) plen 4                  #590 [hci0] 6.533079
      Setup Synchronous Connection (0x01|0x0028) ncmd 2
        Status: Success (0x00)
> HCI Event: Synchronous Connect Complete (0x2c) plen 17   #591 [hci0] 6.592081
        Status: Success (0x00)
        Handle: 11
        Address: 34:75:63:01:21:05 (SHENZHEN RF-LINK TECHNOLOGY CO.,LTD.)
        Link type: eSCO (0x02)
        Transmission interval: 0x0c
        Retransmission window: 0x04
        RX packet length: 60
        TX packet length: 60
        Air mode: Transparent (0x03)
< SCO Data TX: Handle 11 flags 0x00 dlen 60                #592 [hci0] 6.599305
> SCO Data RX: Handle 11 flags 0x00 dlen 72                #593 [hci0] 6.62308
```

I suspect the dlen is not quite right since it was supposed to be 60 bytes going either side but in fact it is doing 72 bytes, anyway the logs doesn't contain the decoded since the connection apparently was stablished before you had started the btmon thus why the PSM 0 appear on every packet, btw have you tried changing to another codec instead of mSBC?

mauzil, 2021-12-27:
I tried this codec too

[image]

I get the same behaviour

mauzil, 2021-12-30:
Here I start btmon before headphone connection
[sco_data.log](https://github.com/bluez/bluez/files/7793498/sco_data.log)

I switch from a2dp_sink to handsfree_head_unit and return.
I use blueman-manager to switch the profile

mauzil, 2022-01-02:
Hi
I installed the driver from MPOW support area and the problem is solved
This is the file I get.
https://mpow.s3-us-west-1.amazonaws.com/20201202_mpow_BH456A_driver+for+Linux.7z

The headset i MPOW H19 IPO, the usb dongle is MPOW BH456A

mauzil, 2022-01-04:
Sorry, but this is not true.
The mpow drivers doesn't solve the problem.

I'm investigating....

Vudentz, 2022-01-04:
@mauzil what does the driver does anyway?

mauzil, 2022-01-05:
Ok. 
Solved.

I downloaded the driver from MPow site and compiled it for my kernel version (5.15.12-artix1-1 amd64)

I blacklisted module btusb in /etc/modprobe.d/blacklist.conf

blacklist btusb

Now handsfree_head_unit profile works for MPOW BH456A dongle, an I can control it from pavucontrol.

The problem is in btusb,ko. Maybe some functions cannot manage mpow bh456a in the right way,

The compiled driver from MPOW source packet is named rtk_btusb.ko

From this link you get the source code.
[20201202_LINUX_BT_DRIVER.zip](https://github.com/bluez/bluez/files/7815162/20201202_LINUX_BT_DRIVER.zip)

closed 2022-01-05 by mauzil