bluez/bluez #252: Discovery is shared between applications (architectural issue)

weliem, 2021-12-20:
**Problem**
In its current implementation, all objects on the Bluez DBus tree are shared. This leads to issues when there are multiple applications that use Bluez. This is best explained by looking at the 'discovery' process. When two applications both start/stop the discovery, unexpected things will happen that are not solvable by any workaround. Consider the following scenario:

- Start bluetoothctl in one xterm and do 'scan on'
- Open bluetoothctl in another xterm and you will see that discovery results are pooring in although the application didn't do 'scan on'
- Do 'scan off' in the second xterm and the discovery will stop in both xterms.

In other words, there is only 1 discovery activity that both applications control. I understand that technically per adapter there can be only 1 scan on/off but Bluez should abstract that away (in my opinion)

This leads to problems where 2 (or more) applications will 'fight for the scanner', e.g.

- Application A wants to discover 10 seconds but application B makes this impossible because it wants to discovery only 5 seconds.
- Application A defines a discovery filter and application B also sets the discovery filter. Now both applications will get unwanted results!

**Possible workaround?**
I don't see how I can work around this in my application in a reliable way. Even if you implement application logic to check if some other applications are stopping your discovery, it will only lead to a fight between applications to control the discovery.

**Requested solution**
The 'bluetoothd' deamon should keep track of who is asking for a discovery and only deliver results to the application that asked for a discovery. Similarly, a discovery filter should be per application.

Alternatively, a new concept 'scanner' should be introduced that an application can request. It is then private for an application.

I realise this may be complicated to implement but I think it is the only way to make sure that multiple applications can use Bluez reliably. On iOS and Android, the 'Scanner' object is per application and the bluetooth stack handles the behavior towards the bluetooth hardware internally. Hence you can have multiple Bleutooth applications running independently without any issues!

Vudentz, 2021-12-20:
> **Problem** In its current implementation, all objects on the Bluez DBus tree are shared. This leads to issues when there are multiple applications that use Bluez. This is best explained by looking [...]

Have you actually tested that it worked that way? The scan off doesn't stop the discovery:

```
[bluetooth]# scan off
Discovery stopped
[NEW] Device 0B:CF:26:C8:74:61 0B-CF-26-C8-74-61
[NEW] Device C0:04:71:E1:ED:C0 004-CZ6804-001
[CHG] Device D4:4D:A4:2F:21:51 TxPower: 90
```
https://github.com/bluez/bluez/blob/master/src/adapter.c#L2272

> In other words, there is only 1 discovery activity that both applications control. I understand that technically per adapter there can be only 1 scan on/off but Bluez should abstract that away (in [...]

There is already some logic in place to merge filters:

https://github.com/bluez/bluez/blob/master/src/adapter.c#L2053
 
> **Possible workaround?** I don't see how I can work around this in my application in a reliable way. Even if you implement application logic to check if some other applications are stopping your discovery, it will only lead to a fight between applications to control the discovery.

Either you have a bug or you actually never tested this properly, each client is tracked separately and stopping one doesn't stop the other.
 
> **Requested solution** The 'bluetoothd' deamon should keep track of who is asking for a discovery and only deliver results to the application that asked for a discovery. Similarly, a discovery filter should be per application.

It already tracks clients just look at the code.
 
> Alternatively, a new concept 'scanner' should be introduced that an application can request. It is then private for an application.
>
> I realise this may be complicated to implement but I think it is [...]

Except for filtering the D-Bus Signals it already does that, D-Bus signals are unidirectional broadcast so we can't really indicate a client, or clients. That said the client/application can instead only subscribe for new devices while it is discovering, in that sense perhaps bluetoothctl is at fault since it is always listening for new devices.

weliem, 2021-12-22:
Thank your for your reply! I studied the code you pointed at and I now understand it works slightly different than I thought.

My understanding is that:

- If the adapter isn't discovering, the first call to StartDiscovery will start the discovery and the property 'discovering' becomes true.
- Each caller of StartDiscovery get's a 'session'. There is no way to find out if your app has a 'session'. The app needs to keep track of it itself.
- Calling StopDiscovery stops the session of the caller. If there are no other sessions then the adapter stops discoverying and the property 'discovering' becomes false.
- When the adapter is discovering, regardless if you were the one starting it, signals (interface-added and properties-changed) are broadcasted. The app needs to figure out which signals are because of the discovery session it started and whether it should ignore them or not.
- When multiple applications set discovery filters, the filters are 'merged'...effectively widening the search. So more signals are sent that one of the application might expect.

Is this correct? I now understand that my problems stem from the fact that the signals are broadcasted to all listeners and that I need to figure out which ones are the ones I am interested in. The 'bluetoothctl' doesn't do any filtering, hence all signals are shown...and my confusion.

My application scans continuously and needs to connect to devices once they are discovered.
I implemented the following logic in my app that seems to work:

- My app listens to 'interface-added' signals for the adapter and 'properties-change' for devices
- My app sets a discovery filter with some service UUIDs. My app stores a copy of the filter for later checks.
- When the method call 'StartDiscovery' succeeds, assume we have a 'session'. Ignore the adapter property 'discovering'.
- If we don't have a session, then ignore all 'interface-added' signals.
- If we don't have a session, then ignore all 'properties-change' signals from unconnected devices.
- Deal with 'properties-changed' signals only for devices that we connected ourselves.
- If we have a session and we get a 'interface-added' signal, we double-check if it matches the discovery filter because other apps may have 'widened' the filter by setting less restrictive filters. If it matches the discovery filter we have found a valid discovery result.
- If we have a session and get a 'properties-changed' signal, we check a bunch of conditions to see if it is a discovery result: 1) must be for a disconnected device, 2) must match the scan filter, 3) the property must be RSSI, ServiceData or ManufacturerData (as those are the only properties that can change during a discovery). If all conditions are satisfied, it is a valid discovery result.
- When we call 'StopDiscovery' and succeeds, we assume the session is stopped.

Not sure if I missed some other edge case, but initial testing seems to prove that this is more or less correct. Note that my application is a medical device gateway that typically scans continuously and connects to several devices simultaneously.

closed 2022-01-05 by Vudentz