bluez/bluez #255: obexd D-DBus activation does not work

nicolasfella, 2021-12-24:
When obexd is not running calling any method on its interface is supposed to start it. However when running `qdbus org.bluez.obex` I get:
```
Error: org.freedesktop.systemd1.NoSuchUnit
Unit dbus-org.bluez.obex.service not found.
```
`/usr/share/dbus-1/services/org.bluez.obex.service` refers to the systemd service `dbus-org.bluez.obex.service`, however no such service file exists. What does exist is `/usr/lib/systemd/user/obex.service` which contains `Alias=dbus-org.bluez.obex.service`. I'm not  sure whether that is supposed to work or it is a bug in DBus/systemd.

Arch Linux is working around this downstream by symlinking `/usr/lib/systemd/user/dbus-org.bluez.obex.service` to `/usr/lib/systemd/user/obex.service` (https://bugs.archlinux.org/task/45816)

Tested with bluez 5.62 on OpenSUSE Tumbleweed

nicolasfella, 2021-12-24:
See also https://linux.debian.bugs.dist.narkive.com/WSBHFy9L/bug-754675-obexd-is-not-started-automatically

HurricanePootis, 2023-02-28:
I have this problem, and even on 5.66, the Arch Linux symlinking stopped working

rparkins999, 2023-10-29:
<code>/usr/share/dbus-1/services/org.bluez.obex.service</code> does not exist at all on my machine (running OpenSUSE Leap 15.5). How do I find out what should be in it?

rparkins999, 2023-10-29:
Here is a version of <code>usr/share/dbus-1/services/org.bluez.obex.service</code> which works, at least for me, without putting in any symlinks anywhere else.
<code>
[D-BUS Service]
Name=org.bluez.obex
Exec=/usr/lib/bluetooth/obexd
SystemdService=obex.service
</code>

szw0407, 2024-03-13:
> When obexd is not running calling any method on its interface is supposed to start it. However when running `qdbus org.bluez.obex` I get:
>
> ```
> Error: org.freedesktop.systemd1.NoSuchUnit
> Unit [...]

on [openSUSE Wiki Bluetooth](https://en.opensuse.org/Bluetooth#File_transfer_not_working)

This can be done via:

```bash
systemctl --user start obex
systemctl --user enable obex
```

The openSUSE bug tracker:
[#1198626](https://bugzilla.opensuse.org/show_bug.cgi?id=1198626)

AND, what the commands have ACTUALLY done is to create a symlink as described above.