bluez/bluez #257: Sony WH-1000XM4 + DualSense Controller Disconnection

giddie, 2022-01-02:
When the WH-1000XM4 headset and DualSense controller are connected simultaneously via Bluetooth, and the audio stream to the headset is suspended, attempting to resume the audio stream causes the headset to disconnect.

A Wireshark capture is attached. This the sequence:

* WH-1000XM4 is connected
* I tap on the headset to start playing media (audio starts playing via LDAC)
* I tap on the headset to stop playing media (audio stream is suspended)
* I connect the DualShock controller
* I tap on the headset to start playing media (after several seconds, audio begins playing)
* I tap on the headset to stop playing media (audio stream is suspended)
* I tap on the headset to start playing media
* Nothing happens for several seconds
* The headset disconnects

[WH-1000XM4 + DualSense.zip](https://github.com/bluez/bluez/files/7799115/WH-1000XM4.%2B.DualSense.zip)

Let me know if there's any more information I can provide :)

Vudentz, 2022-01-26:
@giddie I suspect LDAC requires a lot of bandwidth which perhaps cannot work when there are other peripherals connected in parallel, try using another codec.

giddie, 2022-01-26:
Nope, that was also my first thought, but it's not that. I've tried with SBC, and I've taken bandwidth measurements. Also, I have had 0 issues playing games with the WH-1000XM4s + Dualsense _so long as_ I don't allow the audio stream to suspend and resume. In other words, I start playing music through the headphones, then connect the DualSense, launch a game, then pause the music. So long as I use that workaround, audio and controller latency are perfectly normal, and no drop-outs. If I close the game and don't immediately start playing music, the headphones will disconnect the next time any sound output is attempted.

urbenlegend, 2022-12-20:
I am running into this same issue with my Dualsense as well. Once the last active audio stream disappears, attempting to create another audio stream does not work. Here are my logs:

```
Dec 20 11:06:23 kernel: input: WH-1000XM4 (AVRCP) as /devices/virtual/input/input44
Dec 20 11:06:23 systemd-logind[934]: Watching system buttons on /dev/input/event9 (WH-1000XM4 (AVRCP))
Dec 20 11:06:23 wireplumber[1410]: RFCOMM receive command but modem not available: AT+NREC=0
Dec 20 11:06:34 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:34 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:44 kernel: playstation 0005:054C:0CE6.000A: unknown main item tag 0x0
Dec 20 11:06:44 kernel: playstation 0005:054C:0CE6.000A: hidraw7: BLUETOOTH HID v1.00 Gamepad [Wireless Controller] on 50:e0:85:bd:f8:57
Dec 20 11:06:44 kernel: input: Wireless Controller as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:09:00.3/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:512/0005:054C:0CE6.000A/input/input45
Dec 20 11:06:44 kernel: input: Wireless Controller Motion Sensors as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:09:00.3/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:512/0005:054C:0CE6.000A/input/input46
Dec 20 11:06:44 kernel: input: Wireless Controller Touchpad as /devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:09:00.3/usb3/3-2/3-2:1.0/bluetooth/hci0/hci0:512/0005:054C:0CE6.000A/input/input47
Dec 20 11:06:44 kernel: playstation 0005:054C:0CE6.000A: Registered DualSense controller hw_version=0x00000413 fw_version=0x01030002
Dec 20 11:06:44 kded5[1184]: Touchpad reset
Dec 20 11:06:44 kcminit[4874]: Initializing  "/usr/lib/qt/plugins/plasma/kcms/systemsettings/kcm_mouse.so"
Dec 20 11:06:51 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:51 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:53 plasmashell[4881]: [2022-12-20T19:06:53Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:06:53 plasmashell[4881]: [2022-12-20T19:06:53Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:06:53 plasmashell[4881]: [2022-12-20T19:06:53Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:06:53 plasmashell[4881]: [2022-12-20T19:06:53Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:06:53 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:53 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:53 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:53 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:06:53 rtkit-daemon[1212]: Successfully made thread 4974 of process 4881 owned by '1000' RT at priority 10.
Dec 20 11:06:53 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:06:54 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:06:54 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:06:56 plasmashell[1249]: file:///usr/share/plasma/plasmoids/org.kde.plasma.mediacontroller/contents/ui/CompactRepresentation.qml:77: TypeError: Cannot read property 'implicitWidth' of null
Dec 20 11:06:56 plasmashell[1249]: file:///usr/share/plasma/plasmoids/org.kde.plasma.taskmanager/contents/ui/ToolTipInstance.qml:248:9: QML Image: Error decoding: https://i.ytimg.com/vi/vMoUUrojlpg/hqdefault.jpg?sqp=-oaymwEcCNACELwBSFXyq4qpAw4IARUAAIhCGAFwAcABBg==&rs=AOn4CLBKYn7CJHu2bDse-y-RNVgEs-XBsg: Unsupported image format
Dec 20 11:07:00 bluetoothd[931]: /org/bluez/hci0/dev_94_DB_56_AE_8E_20/sep6/fd3: fd(41) ready
Dec 20 11:07:15 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:07:15 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:07:22 plasmashell[1249]: file:///usr/lib/qt/qml/org/kde/plasma/components.3/ScrollView.qml:45:27: QML ScrollBar: Binding loop detected for property "visible"
Dec 20 11:07:22 plasmashell[1249]: file:///usr/lib/qt/qml/org/kde/plasma/components.3/ScrollView.qml:45:27: QML ScrollBar: Binding loop detected for property "visible"
Dec 20 11:07:32 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:32 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:33 plasmashell[5076]: [2022-12-20T19:07:33Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:07:33 plasmashell[5076]: [2022-12-20T19:07:33Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:07:33 plasmashell[5076]: [2022-12-20T19:07:33Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:07:33 plasmashell[5076]: [2022-12-20T19:07:33Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 1 threads of 1 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Successfully made thread 5170 of process 5076 owned by '1000' RT at priority 10.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
Dec 20 11:07:33 rtkit-daemon[1212]: Supervising 2 threads of 2 processes of 1 users.
[... cut ...]
Dec 20 11:09:47 plasmashell[1249]: org.kde.plasma.pulseaudio: No object for name "bluez_input.94_DB_56_AE_8E_20.0"
Dec 20 11:09:48 bluetoothd[931]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Dec 20 11:10:08 bluetoothd[931]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Connection timed out
Dec 20 11:10:13 bluetoothd[931]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)
```

System Info:
Intel AX200 Bluetooth
Kernel 6.0.12-arch1-1
bluez 5.66-1
pipewire 0.3.63-1
wireplumber 0.4.13-1

Vudentz, 2022-12-20:
> I am running into this same issue with my Dualsense as well. Once the last active audio stream disappears, attempting to create another audio stream does not work. Here are my logs:
>
> ```
> Dec 20 [...]

Do you have the btmon traces when this occurs?

urbenlegend, 2022-12-20:
@Vudentz I ran `sudo btmon > btmon.log` and did the following steps:
1. Pair WH-1000XM4
2. Play a youtube video to test audio is working
3. Paused video
4. Paired Dualsense controller
5. Tried to resume video
6. Video spinner spins and does not resume
7. Waited for about a minute before terminating logs

Here is the result: [btmon.tar.gz](https://github.com/bluez/bluez/files/10273312/btmon.tar.gz)

tkachen, 2023-07-25:
Have same problem with DualSense controller and WH-1000XM2 headset. As soon as I stop playing music headset disconnects. Sometimes it connects back for ~5 seconds, but not listed as output device and disconnects again

System Info:
bluetooth adapter: Intel AX200
kernel: 6.4.5-1-MANJARO
bluez: 5.68-1
pulseaudio: 16.1-6

yann-boyer, 2023-12-20:
I have the same issue but not on the same headset(it's a KDIVIDIO one)... I think the issue is clearly the same.
I hope this will get fixed quickly, it's really annoying, I have to connect my DualSense with a cable...

gabriele2000, 2024-02-01:
I have the same issue.
Today I tried to connect simultaneously:
- My girlfriend's WF-C500 (they require a lot of bandwitch, they're two separate units and I had to upgrade from the Intel 3168 to the Intel AX210 to make them have the correct latency)
- My WH-1000XM5

It all worked, when they go to idle, nothing happens.

Then I connected the Dualsense, then one of the device goes to idle, that device disconnects.
I have to disconnect the Dualsense to be able to reconnect that device, otherwise it won't work anymore.

If I connect the Dualsense before everything else, I'm not able to connect both devices anymore, not even one at a time.
I'll reference my findings in other issues:
- https://github.com/pop-os/linux-firmware/issues/13#issuecomment-1920231748
- https://github.com/bluez/bluez/issues/728#issuecomment-1920374274

urbenlegend, 2024-02-12:
Before I was testing with a WH-1000XM4 + Dualsense and I chucked it up to what @Vudentz said regarding LDAC bandwidth, but I recently tested with my Creative Outlier Air V3 in SBC mode and it was also getting disconnected by the Dualsense. There's definitely some weirdness going on with this specific controller.

gabriele2000, 2024-02-12:
> There's definitely some weirdness going on with this specific controller.

Exactly. The Dualsense is definitely the culprit.
Apparently it never goes into sleep mode, so it definitely hijacks something but since I don't know anything about how the underlying system works, I've no clue.

Vudentz, 2024-02-12:
> > There's definitely some weirdness going on with this specific controller.
> 
> Exactly. The Dualsense is definitely the culprit. Apparently it never goes into sleep mode, so it definitely hijacks something but since I don't know anything about how the underlying system works, I've no clue.

Im afraid this is down at link layer thus why we cannot see what is going on before it disconnects, we may need a sniffer airtrace to check what traffic it is generating.

gabriele2000, 2024-02-13:
> Im afraid this is down at link layer thus why we cannot see what is going on before it disconnects, we may need a sniffer airtrace to check what traffic it is generating.

Ok, give me instructions and I'll collect everything this evening.

gabriele2000, 2024-03-10:
I think the issue is not fixed, I just installed v5.73 and IT WORKS, YAY!

tkachen, 2024-03-16:
Still got same problem with 5.73 installed. Also now I have error messages like in this issue https://github.com/bluez/bluez/issues/777

urbenlegend, 2024-03-28:
I can confirm that 5.73 also does NOT fix the issue for me. I am also seeing the log spam after disconnecting the Dualsense.