bluez/bluez #258: Composite HOG device only register 1 interface

rixnco, 2022-01-05:
Hi,
I'm developping HOG device wich expose a composite HID device (Dual gamepad). 
I can connect and bond, but only one gamepad is registered. (On windows system, both gamepad are correctly registered).
Is there a plan to support composite HOG devices ?

Vudentz, 2022-01-13:
@rixnco Does the composite HID should in the report map or there are multiple instances of HoG service? Not sure if the uhid handles composite report map or perhaps it is missing a kernel driver to do that properly.

rixnco, 2022-01-14:
Hi @Vudentz ,

Yes, this is what I figured out. I had a composite HID in report map. When I changed to multiple instances of HoG it started working with UHID.
Thanks for the hint anyway.

Vudentz, 2022-01-15:
@rixnco perhaps we could split the descriptors and create multiple devices but that would probably require parsing the report map.

Vudentz, 2022-01-15:
@dvdhrm perhaps you have some inputs on how composite HID device shall be created with UHID?

dvdhrm, 2022-01-16:
> @dvdhrm perhaps you have some inputs on how composite HID device shall be created with UHID?

I have no clue what "composite HID devices" are, sorry. UHID forwards the hid-descriptor to the HID core, which then takes care of registrering required input devices. I don't think UHID is in any way special in that regard.

Vudentz, 2022-01-18:
@rixnco do you have an example of how the composite HID device report looks like? From what I understand Linux doesn't understand it but Windows does so perhaps that is something to extend on HID core.

Vudentz, 2022-01-19:
@dvdhrm looks like we never did convert to use UHID_CREATE2 which is actually quite useful given that it does generate UHID_START so we can tell when the kernel does prefix the date with report number https://github.com/BluezTestBot/bluez/pull/1117

closed 2022-02-02 by Vudentz

bam80, 2022-02-18:
Hi,
@Vudentz does UHID_CREATE2 patch above intend to solve the problem here, or it's still not enough to support composite HID devices?

Lenovo Legion M600 wireless bluetooth mouse I have probably has composite design too as some others gaming mice:
https://github.com/kyokenn/rogdrv#rogdrv
The mouse doesn't work on Linux being BT-connected, but works on Windows and Android, here is btmon logs:
https://www.spinics.net/lists/linux-bluetooth/msg96904.html

I tried BlueZ master but it didn't help.
Can we conclude for sure the problem is because of lack of composite devices support in HID core, or the problem might be in something else?