bluez/bluez #259: `bluetoothd` oops sporadically with burst typing on BLE

crides, 2022-01-07:
I have a embedded stenography system on a BLE keyboard, so the firmware would press and release the keys as fast as possible to deliver the keys to the host. On Linux with BlueZ, this sporadically produce kernel oops, i.e. after some random period of typing the Bluetooth system stops responding anymore and needs to be restarted. Other tested hosts are Android 8.1.1 and 12, and they work fine.

I haven't looked into any of the details, but the trace I obtained from `dmesg` is as follows:

```
[  +5.444862] ------------[ cut here ]------------
[  +0.000003] WARNING: CPU: 2 PID: 664 at kernel/workqueue.c:1633 __queue_delayed_work+0x61/0x90
[  +0.000005] Modules linked in: tun ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device uhid xt_conntrack xt_MASQUERADE nf_conntrack_netl ink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overl ay cmac algif_hash algif_skcipher af_alg uvcvideo bnep videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc btusb btrtl btbcm btintel bluetooth rtsx_usb_ms memstick ecdh_generic ecc snd_soc_skl snd_soc_hdac_hda snd_hda_ext_core snd_soc_sst_ipc snd_soc_sst_dsp snd_so c_acpi_intel_match snd_soc_acpi snd_soc_core snd_hda_codec_hdmi snd_compr ess intel_tcc_cooling snd_ctl_led ac97_bus x86_pkg_temp_thermal joydev sn d_pcm_dmaengine snd_hda_codec_realtek snd_hda_codec_generic ee1004 moused ev intel_powerclamp iTCO_wdt intel_pmc_bxt iTCO_vendor_support kvm_intel hid_multitouch kvm mei_hdcp snd_hda_intel snd_intel_dspcfg dell_laptop ir qbypass snd_intel_sdw_acpi i915 crct10dif_pclmul
[  +0.000040]  iwlmvm crc32_pclmul mac80211 coretemp ghash_clmulni_intel dell_wmi ledtrig_audio intel_rapl_msr wmi_bmof dell_smbios aesni_intel cr ypto_simd i2c_algo_bit libarc4 snd_hda_codec cryptd dell_wmi_descriptor s parse_keymap snd_hda_core rapl dcdbas ttm iwlwifi intel_cstate intel_unco re snd_hwdep snd_pcm psmouse drm_kms_helper pcspkr snd_timer r8169 snd ce c realtek i2c_i801 processor_thermal_device_pci_legacy processor_thermal_ device cfg80211 mdio_devres soundcore i2c_smbus libphy intel_gtt processo r_thermal_rfim mei_me processor_thermal_mbox processor_thermal_rapl agpga rt intel_lpss_pci syscopyarea intel_lpss sysfillrect intel_xhci_usb_role_ switch mei sysimgblt intel_rapl_common fb_sys_fops roles intel_pch_therma l idma64 intel_soc_dts_iosf i2c_hid_acpi wmi i2c_hid int3403_thermal dell _smo8800 dell_rbtn int3402_thermal tpm_crb mac_hid video int340x_thermal_ zone tpm_tis tpm_tis_core acpi_pad int3400_thermal rfkill tpm acpi_therma l_rel rng_core vmmon(OE) vmw_vmci vboxnetflt(OE)
[  +0.000041]  vboxnetadp(OE) vboxdrv(OE) ipmi_devintf ipmi_msghandler de ll_smm_hwmon sg crypto_user drm fuse bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 rtsx_usb_sdmmc mmc_core rtsx_usb hid_lo gitech_hidpp hid_logitech_dj usbhid serio_raw atkbd libps2 xhci_pci i8042 crc32c_intel xhci_pci_renesas serio
[  +0.000015] CPU: 2 PID: 664 Comm: bluetoothd Tainted: G          IOE 5.14.16-arch1-1 #1 ad87b876fa2ab6fdbd995dc1c9aab0ad8f767b2c
[  +0.000003] Hardware name: Dell Inc. Vostro 5468/0RRMTH, BIOS 1.1.1 02/ 14/2017
[  +0.000001] RIP: 0010:__queue_delayed_work+0x61/0x90
[  +0.000003] Code: 57 95 01 48 89 72 48 89 7a 50 48 01 c8 48 89 42 30 81 ff 40 01 00 00 75 2c 4c 89 cf e9 48 b9 07 00 e9 73 fb ff ff 0f 0b eb cd <0f> 0b 48 81 7a 38 b0 21 8b ac 74 ab 0f 0b 48 8b 42 28 48 85 c0 74
[  +0.000002] RSP: 0018:ffffafcd4088fcf0 EFLAGS: 00010046
[  +0.000001] RAX: 0000000000000282 RBX: 0000000000000282 RCX: 0000000000 000000
[  +0.000002] RDX: ffff919091e498f0 RSI: 0000000000000000 RDI: 0000000000 000140
[  +0.000001] RBP: ffff919091e49800 R08: 0000000000000000 R09: ffff919091 e49910
[  +0.000001] R10: 0000000000000008 R11: ffff918e4c5b0310 R12: ffff919091 e498f0
[  +0.000001] R13: 0000000000000000 R14: 0000000000000000 R15: ffff918e46 13c2f8
[  +0.000001] FS:  0000000000000000(0000) GS:ffff91911f500000(0000) knlGS :0000000000000000
[  +0.000002] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0.000001] CR2: 00007ffcde68c6b8 CR3: 000000010250e001 CR4: 0000000000 3706e0
[  +0.000004] Call Trace:
[  +0.000003]  queue_delayed_work_on+0x31/0x50
[  +0.000005]  l2cap_chan_del+0x227/0x2c0 [bluetooth 3654cb44fc46aeba175d 052a4f5eb93cc56c3217]
[  +0.000046]  l2cap_sock_shutdown+0x11c/0x4f0 [bluetooth 3654cb44fc46aeb a175d052a4f5eb93cc56c3217]
[  +0.000034]  l2cap_sock_release+0x49/0xe0 [bluetooth 3654cb44fc46aeba17 5d052a4f5eb93cc56c3217]
[  +0.000039]  __sock_release+0x3d/0xa0
[  +0.000005]  sock_close+0x11/0x20
[  +0.000003]  __fput+0x8c/0x230
[  +0.000004]  task_work_run+0x5c/0x90
[  +0.000004]  do_exit+0x361/0xac0
[  +0.000004]  do_group_exit+0x33/0xa0
[  +0.000002]  __x64_sys_exit_group+0x14/0x20
[  +0.000003]  do_syscall_64+0x5c/0x80
[  +0.000005]  ? exc_page_fault+0x72/0x170
[  +0.000002]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  +0.000004] RIP: 0033:0x7f38324bff41
[  +0.000003] Code: Unable to access opcode bytes at RIP 0x7f38324bff17.
[  +0.000001] RSP: 002b:00007fff40e43ac8 EFLAGS: 00000246 ORIG_RAX: 00000 000000000e7
[  +0.000004] RAX: ffffffffffffffda RBX: 00007f38325b6470 RCX: 00007f3832 4bff41
[  +0.000002] RDX: 000000000000003c RSI: 00000000000000e7 RDI: 0000000000 000000
[  +0.000002] RBP: 0000000000000000 R08: ffffffffffffff80 R09: 0000000000 000000
[  +0.000001] R10: 0000000000000008 R11: 0000000000000246 R12: 00007f3832 5b6470
[  +0.000003] R13: 0000000000000002 R14: 00007f38325b6948 R15: 0000000000 000000
[  +0.000004] ---[ end trace 52605a349970365c ]---
[  +0.000003] BUG: kernel NULL pointer dereference, address: 000000000000 0102
[  +0.000004] #PF: supervisor read access in kernel mode
[  +0.000002] #PF: error_code(0x0000) - not-present page
[  +0.000002] PGD 0 P4D 0
[  +0.000003] Oops: 0000 [#1] PREEMPT SMP PTI
[  +0.000003] CPU: 2 PID: 664 Comm: bluetoothd Tainted: G        W IOE 5.14.16-arch1-1 #1 ad87b876fa2ab6fdbd995dc1c9aab0ad8f767b2c
[  +0.000005] Hardware name: Dell Inc. Vostro 5468/0RRMTH, BIOS 1.1.1 02/ 14/2017
[  +0.000002] RIP: 0010:__queue_work+0x20/0x410
[  +0.000004] Code: 46 5b a1 00 e9 db bb 9b 00 90 0f 1f 44 00 00 41 57 41 56 49 89 d6 41 55 41 54 41 89 fc 55 48 89 f5 53 48 83 ec 10 89 7c 24 04 <f6> 86 02 01 00 00 01 0f 85 12 03 00 00 e8 1e 9d 06 00 49 c7 c5 e8
[  +0.000003] RSP: 0018:ffffafcd4088fcb0 EFLAGS: 00010082
[  +0.000003] RAX: ffff919091e498f8 RBX: 0000000000000282 RCX: 0000000000 000000
[  +0.000003] RDX: ffff919091e498f0 RSI: 0000000000000000 RDI: 0000000000 000140
[  +0.000002] RBP: 0000000000000000 R08: ffff919091e498f8 R09: ffff919091 e49910
[  +0.000002] R10: 0000000000000008 R11: ffff918e4c5b0310 R12: 0000000000 000140
[  +0.000002] R13: 0000000000000000 R14: ffff919091e498f0 R15: ffff918e46 13c2f8
[  +0.000002] FS:  0000000000000000(0000) GS:ffff91911f500000(0000) knlGS :0000000000000000
[  +0.000003] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0.000002] CR2: 0000000000000102 CR3: 000000010250e001 CR4: 0000000000 3706e0
[  +0.000003] Call Trace:
[  +0.000002]  queue_delayed_work_on+0x31/0x50
[  +0.000005]  l2cap_chan_del+0x227/0x2c0 [bluetooth 3654cb44fc46aeba175d 052a4f5eb93cc56c3217]
[  +0.000035]  l2cap_sock_shutdown+0x11c/0x4f0 [bluetooth 3654cb44fc46aeb a175d052a4f5eb93cc56c3217]
[  +0.000036]  l2cap_sock_release+0x49/0xe0 [bluetooth 3654cb44fc46aeba17 5d052a4f5eb93cc56c3217]
[... 11 lines cut ...]
[  +0.000002] Code: Unable to access opcode bytes at RIP 0x7f38324bff17.
[  +0.000002] RSP: 002b:00007fff40e43ac8 EFLAGS: 00000246 ORIG_RAX: 00000 000000000e7
[  +0.000003] RAX: ffffffffffffffda RBX: 00007f38325b6470 RCX: 00007f3832 4bff41
[  +0.000002] RDX: 000000000000003c RSI: 00000000000000e7 RDI: 0000000000 000000
[  +0.000002] RBP: 0000000000000000 R08: ffffffffffffff80 R09: 0000000000 000000
[  +0.000002] R10: 0000000000000008 R11: 0000000000000246 R12: 00007f3832 5b6470
[  +0.000002] R13: 0000000000000002 R14: 00007f38325b6948 R15: 0000000000 000000
[  +0.000003] Modules linked in: tun ccm rfcomm snd_seq_dummy snd_hrtimer snd_seq snd_seq_device uhid xt_conntrack xt_MASQUERADE nf_conntrack_netl ink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc overl ay cmac algif_hash algif_skcipher af_alg uvcvideo bnep videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc btusb btrtl btbcm btintel bluetooth rtsx_usb_ms memstick ecdh_generic ecc sn [... cut ...]
[  +0.000001] RBP: 0000000000000000 R08: ffff919091e498f8 R09: ffff919091 e49910
[  +0.000002] R10: 0000000000000008 R11: ffff918e4c5b0310 R12: 0000000000 000140
[  +0.000002] R13: 0000000000000000 R14: ffff919091e498f0 R15: ffff918e46 13c2f8
[  +0.000002] FS:  0000000000000000(0000) GS:ffff91911f500000(0000) knlGS :0000000000000000
[  +0.000003] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0.000002] CR2: 0000000000000102 CR3: 000000010250e001 CR4: 0000000000 3706e0
[  +0.000002] Fixing recursive fault but reboot is needed!
```

Vudentz, 2022-01-26:
@crides what kernel version? From the looks of it that is not caused by fast input or anything like that but rather a disconnection or did bluetoothd crashed in the process? It would be great to have access to the HCI traces (btmon) and bluetoothd logs.

crides, 2022-01-26:
Surprisingly I didn't include that in the original issue, and I can't remember what I used then. Now I'm on `5.15.12-arch1-1`, and I think I'm still running into the same issue. However, now there's no kernel dumps in `dmesg` on `bluetoothd` when it stops working. For accessing the logs I'm assuming I'd need to run `btmon` and `bluetoothd -d -n` continuously and monitor/log the outputs?

Vudentz, 2022-01-26:
> Surprisingly I didn't include that in the original issue, and I can't remember what I used then. Now I'm on `5.15.12-arch1-1`, and I think I'm still running into the same issue. However, now there's [...]

Yep