bluez/bluez #260: AirPods Level of Battery

chodorenko, 2022-01-07:
Please add support for AirPods battery levels indications as described on manual [Accessory Design Guidelines](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf) section 22. Bluetooth Headset Battery Level Indication

chodorenko, 2022-01-10:
I'm talking about the fact that the AirPods series headphones do not reflect the charge level in any of the Linux distributions (https://drive.google.com/file/d/1IJYZWgOBVtVXEWyhTSeW_Ii5okKbEzst/view), the standard monitor of power  does not see the charge level of this device. For example Xiaomi mi Bluetooth speaker show level of battery   https://drive.google.com/file/d/1Sl0v-dBb7uU7qpIegid203dn0O3swVUc/view

ShadowJonathan, 2022-02-16:
This repo seems to successfully fetch airpods battery status when bluetooth is in `dual` mode; https://github.com/delphiki/AirStatus

andirsun, 2023-02-23:
@chodorenko I am thinking of create a Gnome extention for this https://github.com/andirsun/gnome-shell-extention-airpods-battery/blob/main/airpods_battery.py

Actually dont know why bluez dont support to report the battery over DBus  API
[image]

cenunix, 2023-07-04:
I find this script to be quite inaccurate at times, I have the airpods pro 2 and had to manually add support for them myself. On top of that the script is seemingly quite inefficient. Is there any way support for this could be added to bluez?

andirsun, 2023-09-27:
@cenunix What do you think about to create a PR on BlueZ, I have airpods pro 1 and you have version 2.

Vudentz, 2023-10-03:
@chodorenko afaik apple use HFP/AT-commands for accessing the battery status, which is not under control of bluetoothd.

andirsun, 2023-10-04:
@Vudentz what do you think will be the solution here ?

pv, 2023-10-04:
If you are running pulseaudio/pipewire, they will parse the HFP commands and update battery level. This requires you enable experimental features in BlueZ (`Experimental=true` in main.conf), to have the battery reporting interface.

Vudentz, 2023-10-04:
> If you are running pulseaudio/pipewire, they will parse the HFP commands and update battery level. This requires you enable experimental features in BlueZ (`Experimental=true` in main.conf), to have the battery reporting interface.
> 
> The above script however parses the battery level from LE advertisement ManufacturerData, so it's probably available in several places in airpods.

I don't think we would like to continuously scan to gather the battery info though and I suspect apple only do that over AD so the central can see the battery level before connecting. That said perhaps it is time to make the Battery interface stable if that is what is preventing to get the battery status.

andirsun, 2023-10-05:
@pv Is not working.
I change Experimenta=true on ```/etc/bluetooth/main.conf```
Then restart the system but the my Airpods 1 are not reporting the battery level.
[image]

[image]

pv, 2023-10-05:
@andirsun: pipewire or pulseaudio? does the device have HFP audio profile visible? did you also remove the `#` comment character in front of the line?

andirsun, 2023-10-05:
Pipewire
Yes I remove the #
And I don't know how can I find the HFP profile available