bluez/bluez #261: Media buttons non-working and Logitech MX Maser connect issue

infabo, 2022-01-07:
After a upgrade from bluez 5.61 to 5.62 I have these issues:

- Logitech MX Master says it is connected. But the cursor does not move. I have to switch off/on the mouse, so it reconnects and gets working.
- Media keys on my BT headphones (Grado GW100) do not work in 5.62. mpris-proxy is running. It works with 5.61
- Bluetooth is unavailable after a standby. I have a Dell laptop where I have a hardware killswitch for wifi/bluetooth. I hit that, it recovers then.

What info can I provide from my system to help you debug the issue?

infabo, 2022-01-07:
Mouse trouble may be related to #220

infabo, 2022-01-07:
The media control stuff is more interesting.

I execute mpris-proxy and the media keys do not respond.
Then I restart the bluetooth service. `$ sudo systemctl restart bluetooth`
Now the media control buttons work on my headphones. Surprise.

infabo, 2022-01-07:
```
mpris-proxy 
org.bluez appeared
Bluetooth Adapter /org/bluez/hci0 found
player org.mpris.MediaPlayer2.firefox.instance3266 at :1.52 found
Bluetooth Transport /org/bluez/hci0/dev_5C_EB_68_82_46_80/sep5/fd0 found
Bluetooth Transport /org/bluez/hci0/dev_5C_EB_68_82_46_80/sep5/fd0 removed
Bluetooth Adapter /org/bluez/hci0 removed
org.bluez disappeared
org.bluez appeared
Bluetooth Adapter /org/bluez/hci0 found
player org.mpris.MediaPlayer2.firefox.instance3266 at :1.52 found
Bluetooth Transport /org/bluez/hci0/dev_5C_EB_68_82_46_80/sep5/fd0 found
```

infabo, 2022-01-07:
```
Jän 07 16:21:47 HOST bluetoothd[10034]: Player unregistered: sender=:1.142 path=/_1_52
Jän 07 16:21:48 HOST bluetoothd[10034]: Player registered: sender=:1.147 path=/_1_52
Jän 07 16:22:11 HOST bluetoothd[10034]: Terminating
Jän 07 16:22:11 HOST systemd[1]: Stopping Bluetooth service...
Jän 07 16:22:11 HOST bluetoothd[10034]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/ldac
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSink/aptx_hd
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_hd
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSink/aptx
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aac
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSink/sbc
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/sbc
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSink/sbc_xq
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/sbc_xq
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/faststream
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jän 07 16:22:11 HOST bluetoothd[10034]: Endpoint unregistered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jän 07 16:22:11 HOST bluetoothd[10034]: Stopping SDP server
Jän 07 16:22:11 HOST bluetoothd[10034]: Exit
Jän 07 16:22:11 HOST systemd[1]: bluetooth.service: Deactivated successfully.
Jän 07 16:22:11 HOST systemd[1]: Stopped Bluetooth service.
Jän 07 16:22:11 HOST systemd[1]: Starting Bluetooth service...
Jän 07 16:22:11 HOST bluetoothd[10527]: Bluetooth daemon 5.62
Jän 07 16:22:11 HOST systemd[1]: Started Bluetooth service.
Jän 07 16:22:11 HOST bluetoothd[10527]: Starting SDP server
Jän 07 16:22:11 HOST bluetoothd[10527]: Bluetooth management interface 1.21 initialized
Jän 07 16:22:11 HOST bluetoothd[10527]: Player registered: sender=:1.147 path=/_1_52
Jän 07 16:22:11 HOST bluetoothd[10527]: Player registered: sender=:1.44 path=/media_player5
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/ldac
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSink/aptx_hd
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_hd
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSink/aptx
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aac
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSink/sbc
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/sbc
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSink/sbc_xq
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/sbc_xq
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/faststream
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jän 07 16:22:11 HOST bluetoothd[10527]: Endpoint registered: sender=:1.44 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jän 07 16:22:14 HOST bluetoothd[10527]: profiles/input/hog-lib.c:set_report_cb() Error setting Report value: Unexpected error code
Jän 07 16:22:32 HOST bluetoothd[10527]: /org/bluez/hci0/dev_5C_EB_68_82_46_80/sep5/fd0: fd(43) ready
```

stacyharper, 2022-01-08:
I relate similar issues on postmarketos aarch64. Restarting bluez made it to works one time but is not a consistent workarround. Downgrading to 5.60 did not solved the issue. I got the issue with two of my device (pinephone a pinebook pro) but my last computer got the jbl bluetooth speakerphone media buttons working without issue (with bluez 5.62 and everything as up to date than other machines). The issue could then probably be related to drivers ?

Vudentz, 2022-02-28:
> Mouse trouble may be related to #220

If it is related to #220 could you please test with the proposed fix?

infabo, 2022-03-01:
Mouse works now on bluez 6.63-2 on Manjaro Linux, as it includes the patch for #220. See https://github.com/archlinux/svntogit-packages/commit/0ca004cf8b75166a29ce6c96a463e2f80142a760

But the other problems reported here still apply.