bluez/bluez #262: Is there a Bluetooth LE EIR (Extended Inquiry Response)/Scan Response example?

PutAForkInMe, 2022-01-08:
Hello,

I am struggling to understand how I would set the EIR to a custom value, are there any examples that I may have missed?

Vudentz, 2022-02-03:
You looking for Advertising or Extended Inquiry, for the former there is the Advertising API:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt

For classic is not common to have user define values.

PutAForkInMe, 2022-02-04:
I'm looking to customize the response from the server, which I have seen absolutely no reference to, anywhere.

I've been through those API docs many times now, and unless I'm just outright missing it, I can't spot anything that would allow me to control what my server responds for the Extended Inquiry Response for a BLE peripheral server.

Vudentz, 2022-02-07:
@PutAForkInMe Extended Inquiryy Response is used by classing BLE uses advertising reports, their format is similar but that all they share.

You can create your own advertisements with the API Ive mentioned on https://github.com/bluez/bluez/issues/262#issuecomment-1028522559

PutAForkInMe, 2022-02-08:
I already have advertisements completed.

I'm looking to customise the GATT server's _response_ to a client device's active scan, which my client devices have dubbed an "Extended Inquiry Response".

I can find nothing on such behaviour in the API docs, nor is there an [example of it](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test), that I'm aware of.

Again, did I miss this information somewhere? All I've found to date is that the hcitool can apparently do this, so I simply assumed that I'm somehow missing that information in the docs/examples.

Vudentz, 2022-02-08:
@PutAForkInMe 

> I already have advertisements completed.
> 
> I'm looking to customise the GATT server's _response_ to a client device's active scan, which my client devices have dubbed an "Extended Inquiry Response".

Does sound something from Bluetooth Spec, there is no response from GATT that uses EIR.

> I can find nothing on such behaviour in the API docs, nor is there an [example of it](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/test), that I'm aware of.
>
> Again, did I miss this [...]

Can do what? What command from hcitool are you talking about?

PutAForkInMe, 2022-02-09:
Ah, this Windows client is calling it a "ScanResponse".

I'm trying to return a customised "ScanResponse" packet, to listening/requesting clients.

Vudentz, 2022-02-10:
> Ah, this Windows client is calling it a "ScanResponse".
> 
> I'm trying to return a customised "ScanResponse" packet, to listening/requesting clients.

It starting to make more sense, ScanResponse is typically where things like device name appears, it is set with use of the advertising-api, so is the client scanning for a specific manufacturer data perhaps?

PutAForkInMe, 2022-02-11:
No, it's just doing an active scan.

I just need to set the data portion of the scan response, but the device name is typically not what I set in the initial advertisement packet anyway, as I recall, it defaults to the device's hostname.

PutAForkInMe, 2022-02-22:
@Vudentz 
I happen to be working on this again, so I have some relevant data from a Bluez server, captured on a Windows client:
This is the initial advertisement. Everything working as expected here:
```
Status: Success
Device name: Device_Name
ConnectableUndirected
Data section: 0 1 2 3
```

This is the ScanResponse, from an active scan:
```
Status: Unreachable
Device name: Device_Name
ScanResponse
Data section: 44 65 76 69 63 65 5F 4E 61 6D 65
```

As you can see, the ScanResponse data reverts to the device's name, whilst the original advert has the expected service UUID short in bytes.

I'm unclear why the GattCommunicationStatus is "Unreachable", but as that's then followed up by connecting to the server and reading information, I'm going to assume there's some console printing error/lag.

I'd like to be able to set the data segments in the "ScanResponse", but since it defaults back to the name of the server, in hex, I have no idea how to do so.
I can set whatever I want in the initial advert, replace the UUID, add data, etc, but the ScanResponse is _always_ the device name, in hex.

kchaves-lumiradx, 2022-10-26:
I'd actually like this added to the dbus interface, about a year ago I was trying to set something up and it looked like the scan response is based on the advertising data that is set. I had to use the hcitool to circumvent the advertising-api which has turned into a very painful trying to manage the advertising in conjunction with my org.bluez.GattService1. 

You can set the scan response data with

hcitool -i hci0 cmd 0x08 0x0009 SOMEHEXSTRING

If there is another way to do this or I'm misunderstanding the advertising-api, some guidance would be appreciated.