bluez/bluez #266: Bluetooth and dual-booting between Linux and Windows

pguerin3, 2022-01-09:
I own bluetooth earbuds, and bluetooth headphones.
Both worked fine in Windows, and I could switch between devices and connect without a problem.

However, now I dual-boot between Windows, and Fedora 35.
I can use the earbuds on Fedora 35, but when I try the headphones they no longer connect.
I can use the headphones in Windows, but the earbuds won't connect.

I was reading in another forum that there is a problem with connecting bluetooth devices when dual-booting.

Using bluetoothctl in Fedora, there isn't an obvious sequence of commands to rectify the situation.
So I'm wondering if there is a procedure, using bluetoothctl, or other tools, that I can use to allow me to use either bluetooth device in Fedora?

Vudentz, 2022-01-09:
Im afraid this is not supported since the windows and the linux don't share any information such as link-keys, etc, so once you boot another OS it has no idea that it had been paired, or in case it was previously paired that its link-keys are no longer valid, so all you can do is to remove the device and attempt to do a new pairing.

closed 2022-01-09 by Vudentz

pguerin3, 2022-01-15:
No problem - I have worked out that the commands to get the device to work again, and they are as follows:
```
remove <device>
scan on
```
Once the device has been scanned, then do the following:
```
scan off
pair <device>
connect <device>
```
Your device should now work.

Tyson319, 2026-11-09:
## Solved `le-connection-abort-by-local` & `br-connection-not-supported` loop for BLE Keyboards (Dual-Boot Windows/Linux)

If you are trying to share Bluetooth LE (Low Energy) pairing keys from Windows to Linux for keyboards like **JetAccess Slim Line K4 BT** (or similar BLE HID devices), and you are trapped in an endless loop of `le-connection-abort-by-local` or `br-connection-not-supported` (the keyboard blinks red/blue and drops the link), here is the easy step-by-step fix.

### Why this happens
1. **The IRK Missing Link:** Unlike classic Bluetooth, BLE devices require a full cryptographic key-set. Most dual-boot guides tell you to copy only the `LTK`. However, without the real **`IRK` (Identity Resolving Key)** from Windows, the Linux kernel (BlueZ) fails to resolve the randomized private BLE addresses, leading to a host-initiated disconnect (`le-connection-abort-by-local`).
2. **The AddressType Trap:** BlueZ often caches these devices with a `public` address transport. If the device uses a `static random` address in Windows, Linux gets confused and defaults to a classic BR/EDR link, causing the `br-connection-not-supported` error.

---

### Step-by-Step Solution

#### Step 1: Export Key Registry directly to a file in Windows
*Crucial Rule:* Pair the keyboard in Linux first (just to find out its MAC addresses), then pair it **last** in Windows.

1. Boot into Windows and pair your keyboard normally.
2. Download the official Microsoft **PsExec** utility (needed to bypass SYSTEM registry permissions).
3. Open **CMD as Administrator** and launch the Registry Editor as SYSTEM:
   ```cmd
   psexec.exe -s -i regedit
   ```
4. To avoid manual typing, export the entire Bluetooth key folder straight to your Desktop by running this command in the same Administrator CMD:
   ```cmd
   reg export "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\BTHPORT\Parameters\Keys" "%USERPROFILE%\Desktop\ble_keys.reg"
   ```
5. Open the `ble_keys.reg` file from your Desktop with Notepad. Find your keyboard's MAC address folder and look for these 4 parameters:
   * **`LTK`** -> Copy the long string of characters from the `"LTK"=hex:...` line. Combine them into one solid line, remove commas, and make sure all letters are **UPPERCASE**.
   * **`IRK`** -> Copy the string from the `"IRK"=hex:...` line. **Warning:** This is a different key, do not duplicate your LTK here!
   * **`EDIV`** -> Look at `"EDIV"=dword:0000ca4b`. Take the last 4 characters (`ca4b`), open your OS Calculator in **Programmer** mode, select **HEX**, type `ca4b`, and look at what number it shows in **DEC** mode (for `ca4b` it will be **`51787`**). Write this number down.
   * **`ERand`** -> Look at the `"ERand"=hex(b):b7,b5,66,12,97,76,d3,42` line. Put these numbers into a single solid string with **UPPERCASE** letters and no commas: **`B7B566129776D342`**.

#### Step 2: Set up folders in Linux Mint / Ubuntu
Boot into Linux. For BLE Random devices, BlueZ **strictly requires** the folder name to be prefixed with `1:` followed by the lowercase MAC address, otherwise it forces a classic connection.

Open your terminal and run these commands (replace `[Adapter_MAC]` and `[Keyboard_MAC]` with your actual addresses in lowercase, without colons):

```bash
# Clean up old stale connection attempts
sudo rm -rf /var/lib/bluetooth/[Adapter_MAC]/[Keyboard_MAC]
sudo rm -rf /var/lib/bluetooth/[Adapter_MAC]/cache/*

# Create the correct BLE Random folder
sudo mkdir -p /var/lib/bluetooth/[Adapter_MAC]/1:[Keyboard_MAC]
```

#### Step 3: Write the absolute `info` config file
Create and open the configuration file:
```bash
sudo nano /var/lib/bluetooth/[Adapter_MAC]/1:[Keyboard_MAC]/info
```

Paste the following structure, substituting your clean Windows keys from Step 1 (all keys must be **UPPERCASE, no spaces, no commas, no colons**):

```ini
[General]
Name=JetAccess Slim Line K4 BT
Appearance=0x03c1
AddressType=random
SupportedTechnologies=LE;
Trusted=true
Blocked=false
Services=00001800-0000-1000-8000-00805f9b34fb;00001801-0000-1000-8000-00805f9b34fb;00001812-0000-1000-8000-00805f9b34fb;

[LongTermKey]
Key=[Your_Windows_LTK]
Authenticated=1
EncSize=16
EDiv=[Your_Decimal_EDIV]
Rand=[Your_Windows_ERand]

[IdentityResolvingKey]
Key=[Your_Windows_IRK]

[Device2_[Inverted_Keyboard_MAC]]
UUIDs=00001800-0000-1000-8000-00805f9b34fb;00001801-0000-1000-8000-00805f9b34fb;00001812-0000-1000-8000-00805f9b34fb;
ServicesResolved=true

[ConnectionParameters]
MinInterval=6
MaxInterval=9
Latency=0
Timeout=300
```

> 💡 **How to properly fill out the `[Device2_...]` line:**
> Take your keyboard's MAC address in lowercase without colons, and **reverse it pair-by-pair**, separating them with underscores instead of colons.
> *Example:* If your MAC is `d6:11:8c:37:be:1c`, the symbol pairs are `d6`, `11`, `8c`, `37`, `be`, `1c`. Reversing them pair-by-pair gives you `1c_be_37_8c_11_d6`. Therefore, your line must look exactly like this: `[Device2_1c_be_37_8c_11_d6]`. This forces BlueZ to bypass discovery and immediately activate the keys.

Save the file (press **`Ctrl + O`**, then **`Enter`**) and exit nano (**`Ctrl + X`**).

#### Step 4: Apply and Enjoy
Restart the Bluetooth service to reload the configuration file:
```bash
sudo systemctl restart bluetooth
```

Turn the keyboard physical power switch off and on, press the spacebar a few times, and it will instantly connect, blink blue twice, and start typing. The pairing keys will now seamlessly work across both OS reboot cycles.