bluez/bluez #267: bluez-5.63: patch to allow compilation with musl

pacho2, 2022-01-09:
From:
https://bugs.gentoo.org/812797
```
From e5dcac6abd6129829c4983d6635916cee8f1a627 Mon Sep 17 00:00:00 2001
From: Jory Pratt <anarchy@gentoo.org>
Date: Sun, 12 Sep 2021 09:46:11 -0500
Subject: [PATCH] include limits.h for PATH_MAX for musl support

---
 tools/mesh-cfgtest.c | 1 +
 1 file changed, 1 insertion(+)

diff --git a/tools/mesh-cfgtest.c b/tools/mesh-cfgtest.c
index fa0474f..bfc8e1b 100644
--- a/tools/mesh-cfgtest.c
+++ b/tools/mesh-cfgtest.c
@@ -26,6 +26,7 @@
 #include <time.h>
 #include <unistd.h>
 #include <sys/stat.h>
+#include <limits.h>
 
 #include <ell/ell.h>
 
-- 
2.32.0
```

Vudentz, 2022-01-10:
@pacho2 can we get a proper patch sent to linux-bluetooth?

inga-s, 2022-03-15:
The following commit addressed the issue (patch applied Wed, 16 Feb 2022):

commit ec8c8f22efb66ccae533fbd55a236570ffcf756c
Author: Michael Nosthoff <bluez@heine.tech>

  tools/mesh-cfgtest: include limits.h
  mesh-cfgtest.c uses MAX_PATH so it should include limits.h.
  fixes compilation error with musl-based toolchains.

closed 2022-03-15 by inga-s