bluez/bluez #269: bluetoothd 5.63 crashes

UweSauter, 2022-01-11:
On Arch Linux with bluez 5.63 I had several crashes in the last days:

<details>
<summary>Expand to view coredump</summary>

```
           PID: 3965 (bluetoothd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Tue 2022-01-11 09:09:13 CET (3h 8min ago)
  Command Line: /usr/lib/bluetooth/bluetoothd
    Executable: /usr/lib/bluetooth/bluetoothd
 Control Group: /system.slice/bluetooth.service
          Unit: bluetooth.service
         Slice: system.slice
       Boot ID: 679cad728dd8462e97a097144e29b84a
    Machine ID: fbf3b39530714781a43e127ea27ed859
      Hostname: hpcsaute-t14
       Storage: /var/lib/systemd/coredump/core.bluetoothd.0.679cad728dd8462e97a097144e29b84a.3965.1641888553000000.zst (present)
     Disk Size: 168.6K
       Message: Process 3965 (bluetoothd) of user 0 dumped core.

                Module linux-vdso.so.1 with build-id 035c7da8b451e8d19aed30c8ca3970d363e14060
                Module libgpg-error.so.0 with build-id 3801187d42c4955bd93c605451430cdf1b164e63
                Module libgcc_s.so.1 with build-id 7f8508bb914546ada778809b64b99d234337d835
                Module libgcrypt.so.20 with build-id db45f5d5e0f7af1e77324fea1885f974619ad268
                Module libcap.so.2 with build-id f56c6cd6ad4d35053340d7ff2f8f954498796739
                Module liblz4.so.1 with build-id e63600ab23b2f6997f42fac2fa56e1f02ce159a1
                Module libzstd.so.1 with build-id 6214183688018e3fba6d66db10c2c8d0b45e0578
                Module liblzma.so.5 with build-id 8b615460aa230708c5183f16bede67aa0437d95e
                Module librt.so.1 with build-id 75484da2d6f1515189eefa076e0a40328834cd16
                Module libm.so.6 with build-id 2b8fd1f869ecab4e0b55e92f2f151897f6818acf
                Module ld-linux-x86-64.so.2 with build-id 040cc3dd10461562f177df39e3be2f3704258c3c
                Module libsystemd.so.0 with build-id b86eb3142e9f930340715a0f090da03edbb24f9b
                Module libpthread.so.0 with build-id 07c8f95b4f3251d08550217ad8a1f31066229996
                Module libpcre.so.1 with build-id 845483dd0acba86de9f0313102bebbaf3ce52767
                Module libc.so.6 with build-id 4b406737057708c0e4c642345a703c47a61c73dc
                Module libasound.so.2 with build-id b17fc1317a484842433c772360376623b64ec3d8
                Module libdl.so.2 with build-id 5abc547e7b0949f89f3c0e21ab0c8331a7440a8a
                Module libdbus-1.so.3 with build-id 74f2ab9c60512f3a93c932c3f627564d42e0b11e
                Module libglib-2.0.so.0 with build-id 220436a4e27f39372710df014d2517c0387eefa4
                Module bluetoothd with build-id 5525c0571ff90afd826ab20e87cbda7a2c47061c
                Stack trace of thread 3965:
                #0  0x00007f49719dd547 __strlen_avx2 (libc.so.6 + 0x160547)
                #1  0x00007f4971b6c246 _dbus_string_init_const (libdbus-1.so.3 + 0x34246)
                #2  0x00007f4971b57be9 n/a (libdbus-1.so.3 + 0x1fbe9)
                #3  0x00007f4971b5a699 dbus_message_iter_append_basic (libdbus-1.so.3 + 0x22699)
                #4  0x00007f4971b5b22f dbus_message_append_args_valist (libdbus-1.so.3 + 0x2322f)
                #5  0x00007f4971b5b4ea dbus_message_append_args (libdbus-1.so.3 + 0x234ea)
                #6  0x0000559f90c409f5 n/a (bluetoothd + 0x359f5)
                #7  0x0000559f90c311b8 n/a (bluetoothd + 0x261b8)
                #8  0x00007f4971bfe740 g_slist_foreach (libglib-2.0.so.0 + 0x71740)
                #9  0x0000559f90c39438 n/a (bluetoothd + 0x2e438)
                #10 0x0000559f90c395bd n/a (bluetoothd + 0x2e5bd)
                #11 0x0000559f90cc8231 n/a (bluetoothd + 0xbd231)
                #12 0x00007f4971be27c8 n/a (libglib-2.0.so.0 + 0x557c8)
                #13 0x00007f4971be1fd3 g_main_context_dispatch (libglib-2.0.so.0 + 0x54fd3)
                #14 0x00007f4971c38049 n/a (libglib-2.0.so.0 + 0xab049)
                #15 0x00007f4971be1523 g_main_loop_run (libglib-2.0.so.0 + 0x54523)
                #16 0x0000559f90c2b742 n/a (bluetoothd + 0x20742)
                #17 0x00007f49718a4b25 __libc_start_main (libc.so.6 + 0x27b25)
                #18 0x0000559f90c2d1ae n/a (bluetoothd + 0x221ae)
                ELF object binary architecture: AMD x86-64
```
</details>

Vudentz, 2022-01-11:
@UweSauter try running it under valgrind, perhaps there is a string that uninitialized.

UweSauter, 2022-01-12:
I had to rebuild bluez 5.63 with debugging symbols.
When running bluetoothd as systemd service with
```
[Unit]
Description=Bluetooth service
Documentation=man:bluetoothd(8)
ConditionPathIsDirectory=/sys/class/bluetooth

[Service]
Type=dbus
BusName=org.bluez
ExecStart=/usr/bin/valgrind --tool=memcheck --leak-check=yes --trace-children=yes --time-stamp=yes --log-file=/var/log/valgrind/bluetoothd.log /usr/lib/bluetooth/bluetoothd
NotifyAccess=main
#WatchdogSec=10
#Restart=on-failure
CapabilityBoundingSet=CAP_NET_ADMIN CAP_NET_BIND_SERVICE
LimitNPROC=1
ProtectHome=true
ProtectSystem=full

[Install]
WantedBy=bluetooth.target
Alias=dbus-org.bluez.service
```
I get this Valgrind output:
<details>
<summary>Click to expand</summary>

[Click to download log file](https://github.com/bluez/bluez/files/7853003/bluetoothd.log)

```
==00:00:00:00.000 2111== Memcheck, a memory error detector
==00:00:00:00.000 2111== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==00:00:00:00.000 2111== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==00:00:00:00.000 2111== Command: /usr/lib/bluetooth/bluetoothd
==00:00:00:00.000 2111== Parent PID: 1
==00:00:00:00.000 2111== 
==00:00:01:00.841 2111== Invalid read of size 8
==00:00:01:00.842 2111==    at 0x143524: media_transport_get_path (transport.c:896)
==00:00:01:00.842 2111==    by 0x13FC50: clear_configuration (media.c:258)
==00:00:01:00.842 2111==    by 0x13FC50: clear_endpoint (media.c:272)
==00:00:01:00.842 2111==    by 0x139C47: avdtp_sep_set_state (avdtp.c:1037)
==00:00:01:00.842 2111==    by 0x48E073F: g_slist_foreach (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.842 2111==    by 0x138548: connection_lost (avdtp.c:1158)
==00:00:01:00.842 2111==    by 0x13885C: disconnect_timeout (avdtp.c:1191)
==00:00:01:00.842 2111==    by 0x1DF760: timeout_callback (timeout-glib.c:25)
==00:00:01:00.842 2111==    by 0x48C47C7: ??? (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.842 2111==    by 0x48C3FD2: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.842 2111==    by 0x491A048: ??? (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.842 2111==    by 0x48C3522: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.842 2111==    by 0x1DFA09: mainloop_run (mainloop-glib.c:66)
==00:00:01:00.842 2111==  Address 0x550d0a0 is 0 bytes inside a block of size 120 free'd
==00:00:01:00.842 2111==    at 0x484127F: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==00:00:01:00.842 2111==    by 0x1C7B7B: remove_interface (object.c:654)
==00:00:01:00.842 2111==    by 0x1C7FFB: g_dbus_unregister_interface (object.c:1388)
==00:00:01:00.842 2111==    by 0x14325F: media_transport_destroy (transport.c:171)
==00:00:01:00.843 2111==    by 0x13FEFE: endpoint_reply (media.c:308)
==00:00:01:00.843 2111==    by 0x49C1FA1: dbus_connection_dispatch (in /usr/lib/libdbus-1.so.3.19.13)
==00:00:01:00.843 2111==    by 0x1C5088: message_dispatch (mainloop.c:59)
==00:00:01:00.843 2111==    by 0x48C3FD2: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.843 2111==    by 0x491A048: ??? (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.843 2111==    by 0x48C3522: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.843 2111==    by 0x1DFA09: mainloop_run (mainloop-glib.c:66)
==00:00:01:00.843 2111==    by 0x1DFE9B: mainloop_run_with_signal (mainloop-notify.c:188)
==00:00:01:00.843 2111==  Block was alloc'd at
==00:00:01:00.843 2111==    at 0x4843A83: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==00:00:01:00.843 2111==    by 0x48CD381: g_malloc0 (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.843 2111==    by 0x1432BB: media_transport_create (transport.c:855)
==00:00:01:00.843 2111==    by 0x141E96: set_configuration (media.c:490)
==00:00:01:00.843 2111==    by 0x141E96: set_config (media.c:624)
[... 999 lines cut ...]
==00:00:01:00.880 2111==    by 0x491A048: ??? (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.880 2111==    by 0x48C3522: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.7000.2)
==00:00:01:00.880 2111== 
==00:00:01:37.315 2111== 
==00:00:01:37.315 2111== HEAP SUMMARY:
==00:00:01:37.315 2111==     in use at exit: 59,164 bytes in 354 blocks
==00:00:01:37.315 2111==   total heap usage: 21,810 allocs, 21,456 frees, 2,042,215 bytes allocated
==00:00:01:37.315 2111== 
==00:00:01:37.316 2111== LEAK SUMMARY:
==00:00:01:37.316 2111==    definitely lost: 0 bytes in 0 blocks
==00:00:01:37.316 2111==    indirectly lost: 0 bytes in 0 blocks
==00:00:01:37.316 2111==      possibly lost: 0 bytes in 0 blocks
==00:00:01:37.316 2111==    still reachable: 59,164 bytes in 354 blocks
==00:00:01:37.316 2111==         suppressed: 0 bytes in 0 blocks
==00:00:01:37.316 2111== Reachable blocks (those to which a pointer was found) are not shown.
==00:00:01:37.316 2111== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==00:00:01:37.316 2111== 
==00:00:01:37.316 2111== For lists of detected and suppressed errors, rerun with: -s
==00:00:01:37.316 2111== ERROR SUMMARY: 224 errors from 23 contexts (suppressed: 0 from 0)
```

</details>

but while running with Valgrind I sometimes have trouble connecting my headset.
`blueman-manager` reports `br-connection-page-timeout` when the headset was previuosly connected to my cell phone but when the headset was connected to my laptop (bluez without valgrind) then the connection is established.

Vudentz, 2022-01-12:
@UweSauter please check if the above patch solves the crashes. Note that it is likely that your sound server is replying with an error to SetConfiguration, so perhaps something is not quite right with that as well.

UweSauter, 2022-01-13:
The proposed patch solves the observed memory leaks. I cannot yet comment if it also resolves the crashes as I don't have a reproducer for the crash.

If bluetoothd runs with Valgrind, I cannot connect my headset. On the first try `blueman-manager`reports `br-connection-page-timeout` while on the following tries it reports `br-connection-unknown`. I suspect that Valgrind slows memory accesses down too much but this is just speculation.

Even with this patch Valgrind reports
```
==00:00:07:55.692 12648== LEAK SUMMARY:
==00:00:07:55.692 12648==    definitely lost: 0 bytes in 0 blocks
==00:00:07:55.692 12648==    indirectly lost: 0 bytes in 0 blocks
==00:00:07:55.692 12648==      possibly lost: 0 bytes in 0 blocks
==00:00:07:55.692 12648==    still reachable: 56,991 bytes in 355 blocks
==00:00:07:55.692 12648==         suppressed: 0 bytes in 0 blocks
```

I can upload the whole trace if you like.

closed 2022-01-13 by BluezTestBot