bluez/bluez #270: Resuming server notification sessions

mxk, 2022-01-11:
I'm implementing battery and HID peripheral services, which require caching of CCCD state across client connections. The client should be able to enable notifications after bonding and then continue receiving notifications on all subsequent connections without having to update CCCDs. This is the specific behavior I'm interested in:

1. Server configures the app and starts advertising.
2. Client connects and establishes a new bond.
3. Client writes to CCCDs to enable notifications.
4. The server PC is restarted.
5. Client re-connects and starts receiving notifications without having to re-enable them.

I thought this would be controlled by the `Cache` option in `main.conf`, but that doesn't seem to be the case, at least not across bluetoothd restarts. Right now, both `StartNotify` and `AcquireNotify` calls persist across client connections, but they are not resumed if bluetoothd itself is restarted. Is there any way to accomplish the desired behavior with the current API?

Here's what the [spec](https://www.bluetooth.com/specifications/specs/battery-service-1-0/) says:

> 4.1.1 Notification Behavior
>
> If a client has configured notification on the Battery Level characteristic, no notifications
of this characteristic shall be sent when disconnected. The Battery service shall not
initiate a connection if a notification would have been sent when connected.
>
> Upon reconnection, the server shall send a notification if the value of the Battery Level
characteristic has changed while the service has been disconnected from a bonded
client.
>
> If a client has configured notification on the Battery Level characteristic and the Battery
Level characteristic value changes while in a connection, then the characteristic shall be
notified.

Vudentz, 2022-01-11:
@mxk The problem is that the service are not persisted when you restart bluetoothd, in order to do that we would likely need to reserve handles while restarting and somehow match the services after they are registered again, so some changes would be required. Also note that we do send a ServiceChanged in that case so the remote device would need to rediscover so it is not like we are not compliant with the specs.

mxk, 2022-01-11:
But if the remote device does not re-enable notifications after Service Changed, then the only current workaround is to delete the device/bond information on the server and force the client to perform bonding again, correct?

Also, when debugging my application with nRF Connect on Android, I frequently had to manually refresh services in order to see changes. Is there a way to control the Service Changed characteristic or to remove it altogether to prevent caching?

Vudentz, 2022-01-12:
> But if the remote device does not re-enable notifications after Service Changed, then the only current workaround is to delete the device/bond information on the server and force the client to [...]

Android should be able to detect and rediscover then, perhaps nRF Connect is not handling Service Changed but I assume a proper application would rediscover and resubscribe for the notifications, at least BlueZ would do that when handling ServiceChanged.

mxk, 2022-01-13:
After some more testing, it seems that macOS handles Service Changed indication correctly by resubscribing to the relevant characteristics. Android, however, does not do this. It seems to receive the Service Changed indication both when the server app is shutting down and when it's started up again, but it does not re-enable indications. I haven't been able to test a Windows client yet.

Any ideas why Android 11 may not be doing what it's supposed to?

closed 2025-08-22 by github-actions[bot]