bluez/bluez #272: [bluez 5.63] bluetoothd segfaults when manually reconnecting Bose QC35II after suspend

spielkind, 2022-01-13:
As stated in #220 my Bose QC35II headset doesn't reconnect automatically after a suspend and bluetoothd even segfaults after a manual  reconnect attempt.

bluez 5.61 just works fine, no options in main.conf set

Let me know if you need any detailed information.

(Linux: Arch, Hardware: ThinkPad X280)

```
Jan 13 22:23:17 xpaddy dbus-daemon[1745]: [session uid=1000 pid=1745] Activating via systemd: service name='org.bluez.obex' unit='dbus-org.bluez.obex.service' requested by ':1.78' (uid=1000 pid=8502 comm="gnome-control-center bluetooth ")
Jan 13 22:23:17 xpaddy dbus-daemon[1745]: [session uid=1000 pid=1745] Successfully activated service 'org.bluez.obex'
Jan 13 22:23:17 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:18 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:18 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:18 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:18 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:19 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:19 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:22 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:22 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:22 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:23 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:23 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 13 22:23:25 xpaddy pulseaudio[2071]: Device doesn't exist for /org/bluez/hci0/dev_4C_87_5D_7E_A7_04
Jan 13 22:23:25 xpaddy bluetoothd[1168]: src/profile.c:new_conn_reply() Hands-Free Voice gateway replied with an error: org.bluez.Error.InvalidArguments, Unable to handle new connection
Jan 13 22:23:25 xpaddy pulseaudio[2071]: Information about device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04 is invalid
Jan 13 22:23:25 xpaddy bluetoothd[1168]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Jan 13 22:23:26 xpaddy audit[1168]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1168 comm="bluetoothd" exe="/usr/lib/bluetooth/bluetoothd" sig=11 res=1
Jan 13 22:23:26 xpaddy kernel: bluetoothd[1168]: segfault at 55d626c60 ip 00007f7cab9e9547 sp 00007ffd6c931ff8 error 4 in libc-2.33.so[7f7cab8af000+14b000]
Jan 13 22:23:26 xpaddy kernel: audit: type=1701 audit(1642109006.665:171): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1168 comm="bluetoothd" exe="/usr/lib/bluetooth/bluetoothd" sig=11 res=1
Jan 13 22:23:26 xpaddy systemd-coredump[9073]: Process 1168 (bluetoothd) of user 0 dumped core.
                                               Module bluetoothd with build-id 5525c0571ff90afd826ab20e87cbda7a2c47061c
                                               #6  0x000055d625c4e9f5 n/a (bluetoothd + 0x359f5)
                                               #7  0x000055d625c3f1b8 n/a (bluetoothd + 0x261b8)
                                               #9  0x000055d625c47438 n/a (bluetoothd + 0x2e438)
                                               #10 0x000055d625c475bd n/a (bluetoothd + 0x2e5bd)
                                               #11 0x000055d625cd6231 n/a (bluetoothd + 0xbd231)
                                               #16 0x000055d625c39742 n/a (bluetoothd + 0x20742)
                                               #18 0x000055d625c3b1ae n/a (bluetoothd + 0x221ae)
Jan 13 22:23:26 xpaddy systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Jan 13 22:23:26 xpaddy systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Jan 13 22:23:26 xpaddy audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
Jan 13 22:23:26 xpaddy kernel: audit: type=1131 audit(1642109006.988:177): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=bluetooth comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'

```

Vudentz, 2022-01-13:
Looks quite similar to: https://github.com/bluez/bluez/issues/269

Vudentz, 2022-01-13:
@spielkind looks like pulseaudio is loosing track of the device for some reason:

`Jan 13 22:48:08 xpaddy pulseaudio[2065]: Device doesn't exist for /org/bluez/hci0/dev_4C_87_5D_7E_A7_04`

So it doesn't seem to be bluetoothd that is rejecting the connection but PulseAudio.

spielkind, 2022-01-13:
@Vudentz bluez-5.61 works without any issues (with no changes to pulse audio) for the headset and for my mx anywhere 3 mouse. So may it's PulseAudio that rejects the connection but then its because sth. changed in bluez 5.61 to 5.62?

Any suggestions what to test/try/debug?

spielkind, 2022-01-14:
@Vudentz not sure what you mean with "always show" :)

Here is what I did with the patched version from #269, whats also weird: that I've to power on the bluetooth controller after the resume.

```
❯ bluetoothctl list
Controller 74:70:FD:4F:03:CF xpaddy [default]
❯ bluetoothctl devices
Device FC:37:BC:B8:BE:29 MX Anywhere 3
Device 4C:87:5D:7E:A7:04 Nachtschatten
Device E8:36:17:C4:BB:8E AirPods von Reik
❯ bluetoothctl info FC:37:BC:B8:BE:29
Device FC:37:BC:B8:BE:29 (random)
	Name: MX Anywhere 3
	Alias: MX Anywhere 3
	Appearance: 0x03c2
	Icon: input-mouse
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-0000-1000-8000-011f2000046d)
	Modalias: usb:v046DpB025d0014
	Battery Percentage: 0x32 (50)
❯ bluetoothctl info 4C:87:5D:7E:A7:04
Device 4C:87:5D:7E:A7:04 (public)
	Name: Nachtschatten
	Alias: Nachtschatten
	Class: 0x00240418
	Icon: audio-headphones
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (81c2e72a-0591-443e-a1ff-05f988593351)
	UUID: Vendor specific           (931c7e8a-540f-4686-b798-e8df0a2ad9f7)
	UUID: Vendor specific           (f8d1fbe4-7966-4334-8024-ff96c9330e15)
	Modalias: bluetooth:v009Ep4020d0481
❯ # supend / resume
❯ bluetoothctl show
Controller 74:70:FD:4F:03:CF (public)
	Name: xpaddy
	Alias: xpaddy
	Class: 0x00000000
	Powered: no
	Discoverable: yes
	DiscoverableTimeout: 0x00000000
	Pairable: no
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
[... 259 lines cut ...]
Jan 14 08:07:12 xpaddy bluetoothd[213323]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Jan 14 08:07:25 xpaddy pulseaudio[188907]: Device doesn't exist for /org/bluez/hci0/dev_4C_87_5D_7E_A7_04
Jan 14 08:07:25 xpaddy bluetoothd[213323]: src/profile.c:new_conn_reply() Hands-Free Voice gateway replied with an error: org.bluez.Error.InvalidArguments, Unable to handle new connection
Jan 14 08:07:25 xpaddy pulseaudio[188907]: Information about device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04 is invalid
Jan 14 08:07:25 xpaddy bluetoothd[213323]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Jan 14 08:07:27 xpaddy pulseaudio[188907]: Information about device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04 is invalid
Jan 14 08:07:27 xpaddy bluetoothd[213323]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArgumentsJan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSink/sbc
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSource/sbc
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint unregistered: sender=:1.170 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSink/sbc
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSource/sbc
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSource/sbc_xq_512
[... cut ...] registered: sender=:1.232 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 14 08:10:32 xpaddy bluetoothd[213323]: Endpoint registered: sender=:1.232 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 14 08:11:10 xpaddy bluetoothd[213323]: /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep1/fd7: fd(43) ready
Jan 14 08:11:10 xpaddy pulseaudio[345860]: org.freedesktop.DBus.Properties.Get /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep1/fd7 Volume failed: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Volume'
```

Vudentz, 2022-01-14:
@spielkind it doesn't seem we ever remove the device so I wonder why PulseAudio is rejecting the connection, try enabling PulseAudio debug logs as well (-vvvv or something like that).

spielkind, 2022-01-17:
Suspend & Resume:
```
I: [pulseaudio] backend-native.c: Lost RFCOMM connection.
D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/fd21 state: idle -> disconnected
D: [pulseaudio] card.c: Setting card bluez_card.4C_87_5D_7E_A7_04 profile handsfree_head_unit to availability status no
D: [pulseaudio] device-port.c: Setting port headphone-input to status no
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04
D: [pulseaudio] bluez5-util.c: dbus: path=/MediaEndpoint/A2DPSource/sbc, interface=org.bluez.MediaEndpoint1, member=ClearConfiguration
D: [pulseaudio] bluez5-util.c: Clearing transport /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep1/fd4 profile a2dp_sink
D: [pulseaudio] bluez5-util.c: Transport /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep1/fd4 state: idle -> disconnected
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Device removed from object /org/pulseaudio/core1/sink1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Sink removed from object /org/pulseaudio/core1/sink1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.DevicePort removed from object /org/pulseaudio/core1/sink1/port0
I: [pulseaudio] core.c: default_sink: bluez_sink.4C_87_5D_7E_A7_04.a2dp_sink -> alsa_output.pci-0000_00_1f.3.analog-stereo
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Device removed from object /org/pulseaudio/core1/source2
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Source removed from object /org/pulseaudio/core1/source2
D: [pulseaudio] source.c: bluez_sink.4C_87_5D_7E_A7_04.a2dp_sink.monitor: state: SUSPENDED -> UNLINKED
D: [pulseaudio] sink.c: bluez_sink.4C_87_5D_7E_A7_04.a2dp_sink: state: SUSPENDED -> UNLINKED
D: [bluetooth] module-bluez5-device.c: IO thread shutdown requested, stopping cleanly
D: [bluetooth] module-bluez5-device.c: IO thread shutting down
I: [pulseaudio] sink.c: Freeing sink 1 "bluez_sink.4C_87_5D_7E_A7_04.a2dp_sink"
I: [pulseaudio] source.c: Freeing source 2 "bluez_sink.4C_87_5D_7E_A7_04.a2dp_sink.monitor"
D: [pulseaudio] card.c: bluez_card.4C_87_5D_7E_A7_04: active_profile: a2dp_sink -> off
D: [pulseaudio] card.c: Setting card bluez_card.4C_87_5D_7E_A7_04 profile a2dp_sink to availability status no
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
D: [pulseaudio] device-port.c: Setting port headphone-output to status no
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.
D: [pulseaudio] module-bluez5-discover.c: Unregistering module for /org/bluez/hci0/dev_4C_87_5D_7E_A7_04
D: [pulseaudio] module-bluez5-device.c: Unloading module for device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04
I: [pulseaudio] module.c: Unloading "module-bluez5-device" (index: #23).
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Module removed from object /org/pulseaudio/core1/module23
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.Card removed from object /org/pulseaudio/core1/card1
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.CardProfile removed from object /org/pulseaudio/core1/card1/profile0
D: [pulseaudio] ratelimit.c: 508 events suppressed
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.CardProfile removed from object /org/pulseaudio/core1/card1/profile1
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] flist.c: ../pulseaudio/src/pulsecore/hashmap.c: entries flist is full (don't worry)
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.CardProfile removed from object /org/pulseaudio/core1/card1/profile2
D: [pulseaudio] protocol-dbus.c: Interface org.PulseAudio.Core1.CardProfile removed from object /org/pulseaudio/core1/card1/profile3
I: [pulseaudio] card.c: Freed 1 "bluez_card.4C_87_5D_7E_A7_04"
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to remove event.
I: [pulseaudio] module.c: Unloaded "module-bluez5-device" (index: #23).
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep3 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep51 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep19 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep49 was removed
[... cut ...]
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep49 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep17 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep33 was removed
D: [pulseaudio] bluez5-util.c: Remote endpoint /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep1 was removed
D: [pulseaudio] bluez5-util.c: Properties changed in device /org/bluez/hci0/dev_4C_87_5D_7E_A7_04

```

spielkind, 2022-01-17:
out of curiosity I replace pulseaudio by pipewire, here  bluez-5.61 also works, but the patched 5.62 results in a segfault of pipewire:

```
Jan 17 12:30:55 xpaddy pipewire-media-session[64446]: spa.bluez5.sink.a2dp: 0x5652b77044c8: error 24
Jan 17 12:30:55 xpaddy bluetoothd[34250]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jan 17 12:30:55 xpaddy bluetoothd[34250]: src/profile.c:ext_io_disconnected() Unable to get io data for Phone Book Access: getpeername: Transport endpoint is not connected (107)
Jan 17 12:30:55 xpaddy pipewire-media-session[64446]: spa.bluez5: Failed to release transport /org/bluez/hci0/dev_4C_87_5D_7E_A7_04/sep3/fd2: Method "Release" with signature "" on interface "org.bluez.MediaTransport1" doesn't exist
Jan 17 12:30:55 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="method_return", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/ldac
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSink/aptx_hd
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx_hd
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSink/aptx
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aac
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSink/sbc
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/sbc
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSink/sbc_xq
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/sbc_xq
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/faststream
Jan 17 12:31:10 xpaddy bluetoothd[34250]: Endpoint unregistered: sender=:1.135 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
Jan 17 12:31:10 xpaddy dbus-daemon[714]: [system] Rejected send message, 0 matched rules; type="error", sender=":1.135" (uid=1000 pid=64446 comm="/usr/bin/pipewire-media-session ") interface="(unset)" member="(unset)" error name="org.bluez.MediaEndpoint1.Error.NotImplemented" requested_reply="0" destination=":1.99" (uid=0 pid=34250 comm="/usr/lib/bluetooth/bluetoothd ")
[... cut ...]
Jan 17 12:31:13 xpaddy bluetoothd[34250]: Endpoint registered: sender=:1.151 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jan 17 12:31:13 xpaddy bluetoothd[34250]: Endpoint registered: sender=:1.151 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jan 17 12:31:13 xpaddy bluetoothd[34250]: Endpoint registered: sender=:1.151 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jan 17 12:31:13 xpaddy bluetoothd[34250]: Endpoint registered: sender=:1.151 path=/MediaEndpoint/A2DPSource/faststream
Jan 17 12:31:13 xpaddy bluetoothd[34250]: Endpoint registered: sender=:1.151 path=/MediaEndpoint/A2DPSource/faststream_duplex
```

yeezy69, 2022-01-18:
Same here with Sony WH-1000XM4 on Archlinux with bluez-5.63

```
systemd-coredump[37622]: [🡕] Process 653 (bluetoothd) of user 0 dumped core.
Module linux-vdso.so.1 with build-id f4c0c67a495415ccf56ef21d6eaadfebc34186ec
Module libgpg-error.so.0 with build-id 3801187d42c4955bd93c605451430cdf1b164e63
Module libgcc_s.so.1 with build-id 7f8508bb914546ada778809b64b99d234337d835
Module libgcrypt.so.20 with build-id db45f5d5e0f7af1e77324fea1885f974619ad268
Module libcap.so.2 with build-id f56c6cd6ad4d35053340d7ff2f8f954498796739
Module liblz4.so.1 with build-id e63600ab23b2f6997f42fac2fa56e1f02ce159a1
Module libzstd.so.1 with build-id 6214183688018e3fba6d66db10c2c8d0b45e0578
Module liblzma.so.5 with build-id 8b615460aa230708c5183f16bede67aa0437d95e
Module librt.so.1 with build-id 75484da2d6f1515189eefa076e0a40328834cd16
Module libm.so.6 with build-id 2b8fd1f869ecab4e0b55e92f2f151897f6818acf
Module ld-linux-x86-64.so.2 with build-id 040cc3dd10461562f177df39e3be2f3704258c3c
Module libsystemd.so.0 with build-id 69dfd360ef432adc8911dd446411765f1922fb48
Module libpthread.so.0 with build-id 07c8f95b4f3251d08550217ad8a1f31066229996
Module libpcre.so.1 with build-id 845483dd0acba86de9f0313102bebbaf3ce52767
Module libc.so.6 with build-id 4b406737057708c0e4c642345a703c47a61c73dc
Module libasound.so.2 with build-id b17fc1317a484842433c772360376623b64ec3d8
Module libdl.so.2 with build-id 5abc547e7b0949f89f3c0e21ab0c8331a7440a8a
Module libdbus-1.so.3 with build-id 74f2ab9c60512f3a93c932c3f627564d42e0b11e
Module libglib-2.0.so.0 with build-id 220436a4e27f39372710df014d2517c0387eefa4
Module bluetoothd with build-id 5525c0571ff90afd826ab20e87cbda7a2c47061c
Stack trace of thread 653:
#0  0x00007f4ae193a547 __strlen_avx2 (libc.so.6 + 0x160547)
#1  0x00007f4ae1ac9246 _dbus_string_init_const (libdbus-1.so.3 + 0x34246)
#2  0x00007f4ae1ab4be9 n/a (libdbus-1.so.3 + 0x1fbe9)
#3  0x00007f4ae1ab7699 dbus_message_iter_append_basic (libdbus-1.so.3 + 0x22699)
#4  0x00007f4ae1ab822f dbus_message_append_args_valist (libdbus-1.so.3 + 0x2322f)
#5  0x00007f4ae1ab84ea dbus_message_append_args (libdbus-1.so.3 + 0x234ea)
#6  0x0000562a48fe29f5 n/a (bluetoothd + 0x359f5)
#7  0x0000562a48fd31b8 n/a (bluetoothd + 0x261b8)
#8  0x00007f4ae1b5b740 g_slist_foreach (libglib-2.0.so.0 + 0x71740)
#9  0x0000562a48fdb438 n/a (bluetoothd + 0x2e438)
#10 0x0000562a48fdb5bd n/a (bluetoothd + 0x2e5bd)
#11 0x0000562a4906a231 n/a (bluetoothd + 0xbd231)
#12 0x00007f4ae1b3f7c8 n/a (libglib-2.0.so.0 + 0x557c8)
#13 0x00007f4ae1b3efd3 g_main_context_dispatch (libglib-2.0.so.0 + 0x54fd3)
#14 0x00007f4ae1b95049 n/a (libglib-2.0.so.0 + 0xab049)
#15 0x00007f4ae1b3e523 g_main_loop_run (libglib-2.0.so.0 + 0x54523)
#16 0x0000562a48fcd742 n/a (bluetoothd + 0x20742)
#17 0x00007f4ae1801b25 __libc_start_main (libc.so.6 + 0x27b25)
#18 0x0000562a48fcf1ae n/a (bluetoothd + 0x221ae)
ELF object binary architecture: AMD x86-64
```

After restart bluetooth.service i could connect.

Vudentz, 2022-01-19:
I cannot reproduce with master:

```
bluetoothd[1428452]: Controller resume with wake event 0x0
bluetoothd[1428452]: plugins/policy.c:reconnect_set_timer() attempt 1/7 2 seconds
bluetoothd[1428452]: plugins/policy.c:reconnect_timeout() Reconnecting profiles
bluetoothd[1428452]: src/service.c:btd_service_ref() 0x6e77e10: ref=3
bluetoothd[1428452]: src/service.c:change_state() 0x6e77e10: device 94:DB:56:F7:F2:88 profile Hands-Free Voice gateway state changed: disconnected -> connecting (0)
bluetoothd[1428452]: src/adapter.c:connected_callback() hci0 device 94:DB:56:F7:F2:88 connected eir_len 12
bluetoothd[1428452]: src/profile.c:ext_connect() Hands-Free Voice gateway connected to 94:DB:56:F7:F2:88
bluetoothd[1428452]: src/service.c:change_state() 0x6e77e10: device 94:DB:56:F7:F2:88 profile Hands-Free Voice gateway state changed: connecting -> connected (0)
bluetoothd[1428452]: src/device.c:device_profile_connected() Hands-Free Voice gateway Success (0)
bluetoothd[1428452]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_94_DB_56_F7_F2_88
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_ref() 0x7e77870: ref=1
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_ref() 0x7e77870: ref=2
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_ref() 0x7e7a960: ref=1
bluetoothd[1428452]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_94_DB_56_F7_F2_88: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
bluetoothd[1428452]: profiles/audio/sink.c:sink_connect() stream creation in progress
bluetoothd[1428452]: src/service.c:change_state() 0x6215dd0: device 94:DB:56:F7:F2:88 profile a2dp-sink state changed: disconnected -> connecting (0)
bluetoothd[1428452]: plugins/policy.c:service_cb() Added Hands-Free Voice gateway reconnect 1
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 94:DB:56:F7:F2:88
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=895
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting true
bluetoothd[1428452]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_94_DB_56_F7_F2_88/sep1
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 3 type 1 media 0 delay_reporting true
bluetoothd[1428452]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_94_DB_56_F7_F2_88/sep3
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 4 type 1 media 0 delay_reporting true
bluetoothd[1428452]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_94_DB_56_F7_F2_88/sep4
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 2 type 1 media 0 delay_reporting true
bluetoothd[1428452]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_94_DB_56_F7_F2_88/sep2
bluetoothd[1428452]: profiles/audio/a2dp.c:load_remote_sep() LastUsed: lseid 2 rseid 4
bluetoothd[1428452]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_parse_resp() DISCOVER request succeeded
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 1 type 1 media 0 in use 0
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 2 type 1 media 0 in use 0
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 3 type 1 media 0 in use 0
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 4 type 1 media 0 in use 0
bluetoothd[1428452]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
bluetoothd[1428452]: profiles/audio/sink.c:discovery_complete() Discovery complete
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_ref() 0x7e7a960: ref=2
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_ref() 0x7e7a960: ref=3
bluetoothd[1428452]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectConfiguration: name = :1.2839 path = /MediaEndpoint/A2DPSource/aac
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_unref() 0x7e7a960: ref=2
bluetoothd[1428452]: profiles/audio/a2dp.c:a2dp_config() a2dp_config: selected SEP 0x6f09310
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_ref() 0x7e7a960: ref=3
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_set_configuration() 0x7e77870: int_seid=2, acp_seid=4
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_unref() 0x7e7a960: ref=2
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_unref() 0x7e7a960: ref=1
bluetoothd[1428452]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_parse_resp() SET_CONFIGURATION request succeeded
bluetoothd[1428452]: profiles/audio/a2dp.c:setconf_cfm() Source 0x6f09310: Set_Configuration_Cfm
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_ref() 0x7e7a960: ref=2
bluetoothd[1428452]: profiles/audio/media.c:media_endpoint_async_call() Calling SetConfiguration: name = :1.2839 path = /MediaEndpoint/A2DPSource/aac
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: IDLE -> CONFIGURED
bluetoothd[1428452]: profiles/audio/a2dp.c:setup_unref() 0x7e7a960: ref=1
bluetoothd[1428452]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[1428452]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received DELAY_REPORT_CMD
bluetoothd[1428452]: profiles/audio/a2dp.c:endpoint_delayreport_ind() Source 0x6f09310: DelayReport_Ind
bluetoothd[1428452]: profiles/audio/avdtp.c:session_cb() 
[... cut ...]
bluetoothd[1428452]: profiles/audio/avrcp.c:avrcp_connect() path /org/bluez/hci0/dev_94_DB_56_F7_F2_88
bluetoothd[1428452]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting
bluetoothd[1428452]: src/service.c:change_state() 0x6205120: device 94:DB:56:F7:F2:88 profile avrcp-controller state changed: disconnected -> connecting (0)
bluetoothd[1428452]: plugins/policy.c:service_cb() Added a2dp-sink reconnect 1
bluetoothd[1428452]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_94_DB_56_F7_F2_88: SINK_STATE_CONNECTING -> SINK_STATE_CONNECTED
```

osoroed, 2022-01-20:
`Jan 13 22:23:23 xpaddy bluetoothd[1168]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
`
I don't know if related but I'm getting a my system logs filled with that same message. Also getting random disconnects from my JBL Charge 4. This started after an upgrade from 5.62

(Linux Arch linux) inxi -F output below:

```
[edd@ryzenpc >~] sudo inxi -F
System:
  Host: ryzenpc Kernel: 5.16.1-arch1-1 x86_64 bits: 64 Desktop: Xfce 4.16.0
    Distro: ArchLinux
Machine:
  Type: Laptop System: HP product: HP Laptop 15s-eq1xxx v: N/A
    serial: 123456788
  Mobo: HP model: 8707 v: 37.18 serial: 123456789ABCDE UEFI: AMI v: F.31
    date: 01/04/2021
Battery:
  ID-1: BAT0 charge: 38.8 Wh (100.0%) condition: 38.8/38.8 Wh (100.0%)
CPU:
  Info: 6-core model: AMD Ryzen 5 4500U with Radeon Graphics bits: 64
    type: MCP cache: L2: 3 MiB
  Speed (MHz): avg: 1562 min/max: 1400/2375 cores: 1: 1400 2: 1400 3: 1400
    4: 1400 5: 1400 6: 2375
Graphics:
  Device-1: AMD Renoir driver: amdgpu v: kernel
  Device-2: IMC Networks HP TrueVision HD Camera type: USB driver: uvcvideo
  Display: server: X.Org 1.21.1.3 driver: loaded: amdgpu,ati
    unloaded: fbdev,modesetting,vesa resolution: 1920x1080~60Hz
  Message: Unable to show advanced data. Required tool glxinfo missing.
Audio:
  Device-1: AMD Renoir Radeon High Definition Audio driver: snd_hda_intel
  Device-2: AMD Raven/Raven2/FireFlight/Renoir Audio Processor
    driver: snd_rn_pci_acp3x
  Device-3: AMD Family 17h HD Audio driver: snd_hda_intel
  Sound Server-1: ALSA v: k5.16.1-arch1-1 running: yes
  Sound Server-2: PulseAudio v: 15.0 running: yes
  Sound Server-3: PipeWire v: 0.3.43 running: yes
Network:
  Device-1: Realtek RTL8821CE 802.11ac PCIe Wireless Network Adapter
    driver: rtw_8821ce
  IF: wlo1 state: up mac: ZZ:ZZ:ZZ:ZZ:ZZ
Bluetooth:
  Device-1: Realtek Realtek Bluetooth 4.2 Adapter type: USB driver: btusb
  Report: bt-adapter ID: hci0 state: up address: ZZ:ZZ:ZZ:ZZ:ZZ
Drives:
  Local Storage: total: 476.94 GiB used: 58.48 GiB (12.3%)
  ID-1: /dev/nvme0n1 vendor: Western Digital
    model: PC SN730 SDBPNTY-512G-1006 size: 476.94 GiB
Partition:
  ID-1: / size: 274.43 GiB used: 58.42 GiB (21.3%) fs: ext4
    dev: /dev/nvme0n1p5
  ID-2: /boot/efi size: 256 MiB used: 62.8 MiB (24.5%) fs: vfat
    dev: /dev/nvme0n1p1
Swap:
  Alert: No swap data was found.
Sensors:
  System Temperatures: cpu: 42.0 C mobo: N/A gpu: amdgpu temp: 40.0 C
  Fan Speeds (RPM): cpu: 0 fan-2: 0
Info:
  Processes: 277 Uptime: 28m Memory: 7.2 GiB used: 2.96 GiB (41.2%)
  Shell: Bash inxi: 3.3.11
```

macmpi, 2022-01-21:
same on 5.63 / Alpine 3.15: can't connect anymore to Amazon FireTV remote.
Pairing works but connect fails as below:
```
[]# pair XX:XX:XX:XX:XX:XX
Attempting to pair with XX:XX:XX:XX:XX:XX
[CHG] Device XX:XX:XX:XX:XX:XX Connected: yes
[CHG] Device XX:XX:XX:XX:XX:XX Modalias: usb:v04E8p7021d0001
[CHG] Device XX:XX:XX:XX:XX:XX UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device XX:XX:XX:XX:XX:XX UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: yes
[CHG] Device XX:XX:XX:XX:XX:XX Paired: yes
Pairing successful
[CHG] Device XX:XX:XX:XX:XX:XX WakeAllowed: yes
[CHG] Device XX:XX:XX:XX:XX:XX ServicesResolved: no
[CHG] Device XX:XX:XX:XX:XX:XX Connected: no
[]# connect XX:XX:XX:XX:XX:XX
Attempting to connect to XX:XX:XX:XX:XX:XX
[CHG] Device XX:XX:XX:XX:XX:XX Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
```
Journal says:
`Jan 21 15:14:57 tiny-radiocd daemon.err bluetoothd[1854]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed`

ratijas, 2022-01-21:
I am getting similar issue while trying to connect Apple Magic Keyboard. At first, it connected and worked alright (welp, at least after I figured that I needed to turn of the NumLock), but subsequent reconnects and even after reboot — it doesn't work, and `bluetooth.service` spams these red lines in log:

```
Jan 21 21:00:03 orange systemd[1]: Starting Bluetooth service...
Jan 21 21:00:03 orange bluetoothd[2302]: Bluetooth daemon 5.63
Jan 21 21:00:03 orange systemd[1]: Started Bluetooth service.
Jan 21 21:00:03 orange bluetoothd[2302]: Starting SDP server
Jan 21 21:00:03 orange bluetoothd[2302]: Bluetooth management interface 1.21 initialized
Jan 21 21:00:48 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 21 21:00:48 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 21 21:00:48 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
[...]
```

and sometimes also these lines:
```
Jan 21 21:17:52 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 21 21:17:53 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 21 21:17:53 orange bluetoothd[2302]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 21 21:18:43 orange bluetoothd[2302]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
Jan 21 21:18:43 orange bluetoothd[2302]: profiles/input/server.c:confirm_event_cb() Refusing connection from 90:84:0D:E3:16:CD: unknown device
```

ratijas, 2022-01-21:
Update: it works again after cold reboot (turn off the laptop completely, wait few seconds, and only then turn it back on). I have to admit, this is not the only piece of hardware in my laptop that suffers from poor drivers and warm boot problems. My Ethernet driver would not reset to normal once it disappears after hybernation; and my sound card emits static high-pitch noise, and also won't reset on simple warm restart.

### Versions and system info

Operating System: Arch Linux
KDE Plasma Version: 5.24.80
KDE Frameworks Version: 5.91.0
Qt Version: 5.15.2
Kernel Version: 5.16.2-arch1-1 (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-6700HQ CPU @ 2.60GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 970M/PCIe/SSE2

### Hardware

Laptop's model: ASUS ROG G752VT
Bluetooth module: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Network controller: Intel Corporation Wireless 7265 (rev 59)
Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 10)

kaimast, 2022-01-29:
I'm not sure if I have the same issue but my bluetoothd seems to segfault every 15 minutes or so. I never had major bluetooth issues until about two weeks ago. 

I am also on Arch, but this does happen even after a cold boot. Can investigate further if desired.

rusins, 2022-01-29:
Also not sure if it's the same problem, but timeframe is about the same. After suspending my laptop bluetoothd segfaults when connecting to my earphones. I've found that running `sudo systemctl restart bluetooth` and then doing `power on` in `bluetoothctl` fixes it and I can connect again.

nebulade, 2022-01-30:
Came here since I hit a similar issue on Archlinux, looks like after some time the daemon segfaults and systemd for some reason would not restart it automatically. A simple `systemctl restart bluetoothd` brings it backup.
```
Jan 29 20:42:19 lunar systemd[1]: Started Bluetooth service.
Jan 29 20:42:19 lunar bluetoothd[430]: Starting SDP server
Jan 29 20:42:19 lunar bluetoothd[430]: Bluetooth management interface 1.21 initialized
Jan 29 20:42:19 lunar bluetoothd[430]: Failed to set mode: Blocked through rfkill (0x12)
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 29 20:44:19 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 30 15:29:28 lunar bluetoothd[430]: Endpoint unregistered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 30 15:29:29 lunar bluetoothd[430]: Failed to set mode: Blocked through rfkill (0x12)
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_453
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_453
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_512
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_512
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink/sbc_xq_552
Jan 30 15:29:29 lunar bluetoothd[430]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource/sbc_xq_552
Jan 30 15:57:57 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:58 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:58 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:58 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:58 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:59 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:59 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:57:59 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:58:00 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:58:04 lunar bluetoothd[430]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Jan 30 15:58:04 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:58:05 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:58:05 lunar bluetoothd[430]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Jan 30 15:58:06 lunar systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Jan 30 15:58:06 lunar systemd[1]: bluetooth.service: Failed with result 'core-dump'.
```

For reference about the segfault, here the related system logs, it was mostly already posted here, but maybe someone with more insight here could use more log data.
```
[... cut ...]
                                              #15 0x00007f2647e9b463 g_main_loop_run (libglib-2.0.so.0 + 0x54463)
                                              #16 0x000055f81f976742 n/a (bluetoothd + 0x20742)
                                              #17 0x00007f2647b5eb25 __libc_start_main (libc.so.6 + 0x27b25)
                                              #18 0x000055f81f9781ae n/a (bluetoothd + 0x221ae)
                                              ELF object binary architecture: AMD x86-64
```

bastianplettner, 2022-02-03:
I have the same issue, connecting to Bose 700 Headphones after suspend does not work and bluetoothd crashes. A workaround is to do `systemctl restart bluetooth` after every suspend.

I have bisected the issue and I believe the commit causing this issue is d36c45c55a412204c9a14ba2ebd15319b3e51eda.

The current master is working correctly after reverting that commit.

g33kex, 2022-02-03:
I can reproduce on Arch with Sennheiser Momentum TW 2 ear buds. After suspend, bluez segfaults when trying to connect manually. 

Here is the core dump:
```
Feb 03 10:55:44 Mercury systemd-coredump[2483]: [🡕] Process 1064 (bluetoothd) of user 0 dumped core.

                                                Module linux-vdso.so.1 with build-id bd401e5bd0d642686e3a27e85c9cfb15e287f348
                                                Module libgpg-error.so.0 with build-id 82524ee3d1c4c2244d7cfdcc1e6eea5f9855f6c6
                                                Module libgcc_s.so.1 with build-id 7f8508bb914546ada778809b64b99d234337d835
                                                Module libgcrypt.so.20 with build-id db45f5d5e0f7af1e77324fea1885f974619ad268
                                                Module libcap.so.2 with build-id c86eeb29471e24e4dcce8bdba133919582424cef
                                                Module liblz4.so.1 with build-id e63600ab23b2f6997f42fac2fa56e1f02ce159a1
                                                Module libzstd.so.1 with build-id 72f3511cba7db578f6a2647925f35664da6c838b
                                                Module liblzma.so.5 with build-id 8b615460aa230708c5183f16bede67aa0437d95e
                                                Module librt.so.1 with build-id 75484da2d6f1515189eefa076e0a40328834cd16
                                                Module libm.so.6 with build-id 2b8fd1f869ecab4e0b55e92f2f151897f6818acf
                                                Module ld-linux-x86-64.so.2 with build-id 040cc3dd10461562f177df39e3be2f3704258c3c
                                                Module libsystemd.so.0 with build-id eca1e10a62234b73a4e32d1b52598ce39cb186d4
                                                Module libpthread.so.0 with build-id 07c8f95b4f3251d08550217ad8a1f31066229996
                                                Module libpcre.so.1 with build-id 845483dd0acba86de9f0313102bebbaf3ce52767
                                                Module libc.so.6 with build-id 4b406737057708c0e4c642345a703c47a61c73dc
                                                Module libasound.so.2 with build-id b17fc1317a484842433c772360376623b64ec3d8
                                                Module libdl.so.2 with build-id 5abc547e7b0949f89f3c0e21ab0c8331a7440a8a
                                                Module libdbus-1.so.3 with build-id 74f2ab9c60512f3a93c932c3f627564d42e0b11e
                                                Module libglib-2.0.so.0 with build-id bd949a4794ad959019499c6afcb8f416dcff9810
                                                Module bluetoothd with build-id 5525c0571ff90afd826ab20e87cbda7a2c47061c
                                                Stack trace of thread 1064:
                                                #0  0x00007f4906b6b515 __strlen_avx2 (libc.so.6 + 0x160515)
                                                #1  0x00007f4906cfa246 _dbus_string_init_const (libdbus-1.so.3 + 0x34246)
                                                #2  0x00007f4906ce5be9 n/a (libdbus-1.so.3 + 0x1fbe9)
                                                #3  0x00007f4906ce8699 dbus_message_iter_append_basic (libdbus-1.so.3 + 0x22699)
                                                #4  0x00007f4906ce922f dbus_message_append_args_valist (libdbus-1.so.3 + 0x2322f)
                                                #5  0x00007f4906ce94ea dbus_message_append_args (libdbus-1.so.3 + 0x234ea)
                                                #6  0x000055bbbf5ea9f5 n/a (bluetoothd + 0x359f5)
                                                #7  0x000055bbbf5db1b8 n/a (bluetoothd + 0x261b8)
                                                #8  0x00007f4906d8c680 g_slist_foreach (libglib-2.0.so.0 + 0x71680)
                                                #9  0x000055bbbf5e3438 n/a (bluetoothd + 0x2e438)
                                                #10 0x000055bbbf5e35bd n/a (bluetoothd + 0x2e5bd)
                                                #11 0x000055bbbf672231 n/a (bluetoothd + 0xbd231)
                                                #12 0x00007f4906d70708 n/a (libglib-2.0.so.0 + 0x55708)
                                                #13 0x00007f4906d6ff13 g_main_context_dispatch (libglib-2.0.so.0 + 0x54f13)
                                                #14 0x00007f4906dc60d9 n/a (libglib-2.0.so.0 + 0xab0d9)
                                                #15 0x00007f4906d6f463 g_main_loop_run (libglib-2.0.so.0 + 0x54463)
                                                #16 0x000055bbbf5d5742 n/a (bluetoothd + 0x20742)
                                                #17 0x00007f4906a32b25 __libc_start_main (libc.so.6 + 0x27b25)
                                                #18 0x000055bbbf5d71ae n/a (bluetoothd + 0x221ae)
                                                ELF object binary architecture: AMD x86-64
```

Vudentz, 2022-02-04:
> I have the same issue, connecting to Bose 700 Headphones after suspend does not work and bluetoothd crashes. A workaround is to do `systemctl restart bluetooth` after every suspend.
> 
> I have bisected the issue and I believe the commit causing this issue is [d36c45c](https://github.com/bluez/bluez/commit/d36c45c55a412204c9a14ba2ebd15319b3e51eda).
> 
> The current master is working correctly after reverting that commit.

Perhaps it is related to: https://github.com/bluez/bluez/issues/284

Vudentz, 2022-02-04:
> I can reproduce on Arch with Sennheiser Momentum TW 2 ear buds. After suspend, bluez segfaults when trying to connect manually.
>
> Here is the core dump:
>
> ```
> Feb 03 10:55:44 Mercury [...]

The crash itself was fixed in https://github.com/bluez/bluez/issues/269, but I think the actual problem is something like https://github.com/bluez/bluez/issues/284

bastianplettner, 2022-02-04:
> Perhaps it is related to: #284

Looks like this is the same issue, the patch linked in #284  fully fixes the issue for me.

closed 2022-02-04 by BluezTestBot

ratijas, 2022-03-13:
Got myself a Sony headset WH-1000XM4. Can't pair or connect to it even with the https://github.com/bluez/bluez/commit/19448b9cb852b8c38f15362e185e6b18d634859d patch applied (Arch Linux bluez-5.63-2):

```
Mar 13 14:36:04 orange bluetoothd[592]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Mar 13 14:36:07 orange bluetoothd[592]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Mar 13 14:36:07 orange bluetoothd[592]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Mar 13 14:36:14 orange bluetoothd[592]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for F8:4E:17:A1:53:6C: Protocol not available
Mar 13 14:36:14 orange bluedevil-wizard[7304]: kf.bluezqt: PendingCall Error: "br-connection-profile-unavailable"
```

The device appears in the list of paired devices after all, but no matter what I do, it can't connect. Works perfectly with Android 10 though.

ratijas, 2022-03-13:
Note: I also bought an external Bluetooth 5.0 adapter in a hope that it'll fix the issue (maybe built-in adapter has a shitty driver?  who knows). Unfortunately, no matter which adapter is used, the log is flashing with red error messages, and devices still fail to connect.

kaimast, 2022-03-13:
Are you on arch too? At least for me 5.63-2 fixed this issue. I think they back-ported the commit from above.

ratijas, 2022-03-14:
> Are you on arch too? At least for me 5.63-2 fixed this issue. I think they back-ported the commit from above.

Yes, I am. Yes, they did. Double checked PKGBUILD and repositories. Manager is still NULL, and the sh¡t doesn't connect, unfortunately ):

class101, 2022-11-21:
@Vudentz 

Do you think the following errors are connected to this ticket ?

```
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed
```

Seen on Steam Deck, SteamOS 3.3.2, I think they are using a vulnerable kernel `5.13.0-valve21.3-1-neptune` because when I connect the WF100XM4 headphone + gamepad , the gamepad  disconnects after 2 min and the journal is showing the errors

Unfortunately, the Valve developers don't respond to my [ticket](https://steamcommunity.com/app/1675200/discussions/1/3541546590703351528/) so I've to dig myself to find the responsible parties :(

Vudentz, 2022-11-21:
> @Vudentz
>
> Do you think the following errors are connected to this ticket ?
>
> ```
> Nov 21 12:52:45 steamdeck bluetoothd[921]: src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get [...]

Nope, afaik these are just warnings ADV monitor is not supported.

> Seen on Steam Deck, SteamOS 3.3.2, I think they are using a vulnerable kernel `5.13.0-valve21.3-1-neptune` because when I connect the WF100XM4 headphone + gamepad , the gamepad disconnects after 2 min and the journal is showing the errors
> 
> Unfortunately, the Valve developers don't respond to my [ticket](https://steamcommunity.com/app/1675200/discussions/1/3541546590703351528/) so I've to dig myself to find the responsible parties :(
> 
> ```
> (deck@steamdeck ~)$ /usr/lib/bluetooth/bluetoothd -v
> 5.63
> (deck@steamdeck ~)$ uname -r
> 5.13.0-valve21.3-1-neptune
> (deck@steamdeck ~)$ 
> ```

class101, 2022-11-21:
Thanks a lot @Vudentz, do you think I can open a ticket for my issue ? 

Briefly explaining to you the issue :

I connect on Bluetooth, a gamepad + keyboard + mouse , no problems

I connect on Bluetooth, a gamepad + headphone = the gamepad can't stay connected more than 2 min 

I use Bluetooth connection sharing, the service crashes

I suspect a traffic jam is the Bluetooth service, several users are reporting Bluetooth instabilities on Steam Deck to with a gamepad and headphone, but nothing has been officially aggregated yet