bluez/bluez #275: (Maybe) #220 was fixed after all

Flowdalic, 2022-01-21:
I run net-wireless/bluez-5.63-r1 on my Gentoo system. This version has the patches for #220 included (see https://github.com/gentoo/gentoo/commit/277ef13c7bb383657ee4628b45a37b62f66eb20e). However I still have to `systemctl restart bluetooth` for my Logitech MX 3 mouse to connect. Before restarting the bluetooth service, `bluetoothctl` shows that devices is connected then disconnected and then connected again as soon as I start moving the mouse. I've also attached the btmon capture while this happens: [btmon.gz](https://github.com/bluez/bluez/files/7911203/btmon.gz)

Vudentz, 2022-01-21:
@Flowdalic does it should any error on bluetoothd logs?

Vudentz, 2022-01-21:
@Flowdalic it seems to be a different problem:

```
@ MGMT Command: User Confirmation Negative Reply (0x001d) plen 7                                                                                 {0x0001} [hci0] 10.855634
        LE Address: E9:F9:E3:08:AC:BA (Static)
@ MGMT Event: Authentication Failed (0x0011) plen 8                                                                                              {0x0001} [hci0] 10.855640
        LE Address: E9:F9:E3:08:AC:BA (Static)
        Status: Authentication Failed (0x05)
< ACL Data TX: Handle 16 flags 0x00 dlen 6                                                                                                           #173 [hci0] 10.855642
      SMP: Pairing Failed (0x05) len 1
        Reason: Numeric comparison failed (0x0c)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                  {0x0001} [hci0] 10.855644
      User Confirmation Negative Reply (0x001d) plen 7
        Status: Success (0x00)
        LE Address: E9:F9:E3:08:AC:BA (Static)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                               #174 [hci0] 10.862767
        Num handles: 1
        Handle: 16
        Count: 1
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                       #175 [hci0] 10.960760
        Status: Success (0x00)
        Handle: 16
        Reason: Remote User Terminated Connection (0x13)
```

It is trying to pair and fail, now how come it is not paired and reconnecting seems to be the real culprit here.

Flowdalic, 2022-01-22:
> It is trying to pair and fail, now how come it is not paired and reconnecting seems to be the real culprit here.

Hmm, I think the mouse is paired, because I paired it (ha) and after `systemctl restart bluetooth` it starts working. But maybe I am missing something?

After a GDM login, I see the following when starting to move the mouse in `bluetoothctl`:

```
$ bluetoothctl                                                                                                                                                                                    
Agent registered
[CHG] Controller E8:48:B8:C8:20:00 Pairable: yes
[CHG] Device E9:F9:E3:08:AC:BA Connected: yes
[CHG] Device E9:F9:E3:08:AC:BA Connected: no
[CHG] Device E9:F9:E3:08:AC:BA Connected: yes
[CHG] Device E9:F9:E3:08:AC:BA Connected: no
[CHG] Device E9:F9:E3:08:AC:BA Connected: yes
[CHG] Device E9:F9:E3:08:AC:BA Connected: no
[CHG] Device E9:F9:E3:08:AC:BA Connected: yes
[CHG] Device E9:F9:E3:08:AC:BA Connected: no
```
obviously the mouse (E9:F9:E3:08:AC:BA) is not working. Then, after restarting the bluetooth systemd service:

```CHG] Controller E8:48:B8:C8:20:00 Powered: no
[CHG] Controller E8:48:B8:C8:20:00 Discovering: no
[CHG] Controller E8:48:B8:C8:20:00 Class: 0x00000000
Agent unregistered
[DEL] Device E9:F9:E3:08:AC:BA MX Master 3
[DEL] Device 38:18:4C:D2:CF:49 WH-1000XM3
[DEL] Controller E8:48:B8:C8:20:00 neo-pc.sch [default]
Agent registered
[CHG] Controller E8:48:B8:C8:20:00 Class: 0x00000104
[CHG] Controller E8:48:B8:C8:20:00 Powered: yes
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 Name: neo-pc.sch
[CHG] Controller E8:48:B8:C8:20:00 Alias: neo-pc.sch
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 Alias: BlueZ 5.63
[CHG] Controller E8:48:B8:C8:20:00 Class: 0x00200104
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 Class: 0x00600104
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
[CHG] Controller E8:48:B8:C8:20:00 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[... 77 lines cut ...]
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E9_F9_E3_08_AC_BA/service001d/char001e
	00002a19-0000-1000-8000-00805f9b34fb
	Battery Level
[NEW] Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_E9_F9_E3_08_AC_BA/service001d/char001e/desc0020
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_E9_F9_E3_08_AC_BA/service0041
	00010000-0000-1000-8000-011f2000046d
	Vendor specific
[NEW] Characteristic (Handle 0x0000)
	/org/bluez/hci0/dev_E9_F9_E3_08_AC_BA/service0041/char0042
	00010001-0000-1000-8000-011f2000046d
	Vendor specific
[NEW] Descriptor (Handle 0x0000)
	/org/bluez/hci0/dev_E9_F9_E3_08_AC_BA/service0041/char0042/desc0044
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device E9:F9:E3:08:AC:BA ServicesResolved: yes
[CHG] Device E9:F9:E3:08:AC:BA Paired: yes
```

Does that help?

Vudentz, 2022-01-24:
So it is repairing:

`[CHG] Device E9:F9:E3:08:AC:BA Paired: yes`

But looking into the bluetoothd there is the following line:

`Jan 22 18:18:55 neo-pc.sch bluetoothd[1105]: Loading LTKs timed out for hci0`

So it looks like the link keys failed to be reloaded which means your kernel wont have access to them which explain why it attempts to pair, that is weird given it does work when you restart bluetoothd, check with btmon if there is something different when Load LTK fails compared to when it succeeds.

Vudentz, 2022-01-24:
I suspect the system is busy resuming and with many commands queued it may take more than 2 seconds to load the LTKs, instead perhaps we should start the timer only when the command has actually been sent.

Vudentz, 2022-01-25:
@Flowdalic any chance to test the above changes?

Flowdalic, 2022-01-25:
Just re-emerged net-wireless/bluez-9999, i.e., the live ebuild from the current master, with 2ca87617bcb70bdfe97d744043942d6559b0dea4 and 9ba27547df6991b47f507fce362bfe4561b3718b applied. I will report if this fixes the issue for me.

Vudentz, 2022-01-26:
@Flowdalic were you able to test it?

Flowdalic, 2022-01-27:
@Vudentz I wanted to test this more thorough after [it previously appeared to be fixed but then issues re-appeared](https://bugs.gentoo.org/827706#c9). I further have to do a full-powercycle of my workhorse PC to confirm that it works as expected. What I can say, as interim report maybe, is that I booted my PC yesterday and today, and the mouse started working automatically a few seconds after I first moved it. I did not have to restart `bluetoothd`. So it looks very promising! Thanks for your work. :)

Vudentz, 2022-01-28:
@Flowdalic I intend to merge it this week, you can reopen it if you find that it doesn't actually fix your problems.

closed 2022-01-31 by Vudentz