bluez/bluez #276: simple-agent causes Seg Fault and breaks onboard RPi BT

gearhead, 2022-01-22:
Trying to pair without input and since I tried to pair using `simple-agent -c NoInputNoOutput` or `bt-agent --capability=NoInputNoOutput` I get a seg fault. 

This is on a RPi 3B+ running Arch Linux aarch64 linux 5.10.92 and bluez 5.63.

The error is always:
```
~# /usr/lib/bluetooth/bluetoothd -n -d
bluetoothd[5934]: Bluetooth daemon 5.63
...
bluetoothd[5934]: src/adapter.c:connected_callback() hci0 device F8:1F:32:XX:XX:XX connected eir_len 17
bluetoothd[5934]: src/device.c:device_create() dst F8:1F:32:XX:XX:XX
bluetoothd[5934]: src/device.c:device_new() address F8:1F:32:XX:XX:XX
bluetoothd[5934]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_F8_1F_32_39_FE_31
bluetoothd[5934]: src/device.c:device_set_class() /org/bluez/hci0/dev_F8_1F_32_XX_XX_XX 0x5A020C
bluetoothd[5934]: src/adapter.c:new_link_key_callback() hci0 new key for F8:1F:32:XX:XX:XX type 7 pin_len 0 store_hint 1
bluetoothd[5934]: src/adapter.c:store_link_key() Unable to load key file from /var/lib/bluetooth/43:45:C0:XX:XX:XX/F8:1F:32:XX:XX:XX/info: (No such file or directory)
bluetoothd[5934]: GLib: g_file_set_contents_full: assertion 'error == NULL || *error == NULL' failed
bluetoothd[5934]: src/adapter.c:store_link_key() Unable set contents for /var/lib/bluetooth/43:45:C0:XX:XX:XX/F8:1F:32:XX:XX:XX/info: (7)
bluetoothd[5934]: GLib: (../glib/glib/gkeyfile.c:3877):g_key_file_remove_key_value_pair_node: runtime check failed: (pair->value != NULL)
bluetoothd[5934]: src/device.c:device_set_bonded()
bluetoothd[5934]: src/device.c:btd_device_set_temporary() temporary 0
bluetoothd[5934]: src/device.c:device_bonding_complete() bonding (nil) status 0x00
bluetoothd[5934]: src/device.c:device_bonding_complete() setting timer for reverse service discovery
bluetoothd[5934]: src/adapter.c:resume_discovery()
bluetoothd[5934]: src/adapter.c:add_accept_list_complete() F8:1F:32:XX:XX:XX added to kernel accept list
Segmentation fault (core dumped)
```
Before I tried this experiment, I could pair my phone and connect it as an audio source using bluetoothctl interactive but not any more and not with either of these scripts. Now when I try to connect my phone via bluetoothctl interactive, I get the same result where it seg faults.
It appears that trying to use simple-agent broke bluetooth?! 
Before I started this experiment, I was listening to audio streaming from my phone to the RPi. I removed the phone from BT and re-paired it and then tried this experiment with simple-agent and it started seg faulting. Tried to re-pair using bluetoothctl and it also seg faulted. I made a clean new image and tried the same thing and can no longer pair my phone or another phone to this hardware with either image after it started segfaulting. This device also lost wifi after trying simple-agent (which is very strange). To restore wifi, I reinstalled all the firmware packages and now wifi is back functional, but bluetoothd still seg faults when I try to pair any phone. I can pair and use headphones (as an audio sink) but not my phone.

gearhead, 2022-01-22:
Well I did a bit more digging and changed the title of this issue. Apparently, because my phone was paired with an earlier version of Bluez, it was able to connect. When I removed it from the 'devices' list and tried to re-pair it is when the seg fault happens. This happens on a RPi with a BT dongle and also with another with onboard BT. It appears that Blues 5.6.3 is the culprit? I will attempt to downgrade to a previous version of bluez and see if it resolves.

gearhead, 2022-01-22:
Verified that reverting to Bluez 5.62 fixes pairing. on both dongles and onboard BT on RPi armv7 and aarch64. Now back to experimenting with simple pairing.

Vudentz, 2022-01-25:
@gearhead would you be able to provide any backtrace?

corrupt-stack, 2022-02-03:
I've got the same problem, and am including backtraces and ASAN output (in logs pasted below).

I'm on a different architecture (a x86_64 desktop, not a RPi ARM), and a slightly later Linux kernel:

```
$ uname -a
Linux moarbits 5.16.3-arch1-1 #1 SMP PREEMPT Thu, 27 Jan 2022 14:18:25 +0000 x86_64 GNU/Linux
```

Steps: I checked out tag 5.63 from bluez; configured the build with debugging, backtraces, ASAN, etc. enabled; and ran `make src/bluetoothd`. To reproduce the problem, I temporarily swapped my /usr/lib/bluetooth/bluetoothd binary with the debug one, so it would be run from systemd an an identical environment. Ran pairing initiated from my cellphone, using a DisplayYesNo-capable agent locally.

The logs (virtually identical errors that OP had are in there, plus backtraces included):

```
Feb 03 01:18:35 moarbits systemd[1]: Starting Bluetooth service...
Feb 03 01:18:35 moarbits bluetoothd[734241]: Bluetooth daemon 5.63
Feb 03 01:18:35 moarbits systemd[1]: Started Bluetooth service.
Feb 03 01:18:35 moarbits bluetoothd[734241]: Starting SDP server
Feb 03 01:18:35 moarbits bluetoothd[734241]: Bluetooth management interface 1.21 initialized
Feb 03 01:18:35 moarbits bluetoothd[734241]: Admin Policy Set interface registered
Feb 03 01:18:35 moarbits bluetoothd[734241]: Admin Policy Status interface registered
Feb 03 01:18:35 moarbits bluetoothd[734241]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Feb 03 01:18:35 moarbits bluetoothd[734241]: sap-server: Operation not permitted (1)
Feb 03 01:21:20 moarbits bluetoothd[734241]: src/adapter.c:store_link_key() Unable to load key file from /var/lib/bluetooth/80:C5:F2:63:E4:94/F0:5C:77:CE:43:B3/info: (No such file or directory)
Feb 03 01:21:20 moarbits bluetoothd[734241]: GLib: g_file_set_contents_full: assertion 'error == NULL || *error == NULL' failed
Feb 03 01:21:20 moarbits bluetoothd[734241]: ++++++++ backtrace ++++++++
Feb 03 01:21:20 moarbits bluetoothd[734241]: #1  btd_backtrace+0x27a (src/backtrace.c:59) [0x5603f52a2a5a]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #2  g_logv+0x22d (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3964e5a3d]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #3  g_log+0x94 (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3964e5d44]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #4  g_file_set_contents_full+0x67 (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3964c9b17]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #5  store_link_key+0x2fd (src/adapter.c:8369) [0x5603f5320f3d]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #6  new_link_key_callback+0x474 (src/adapter.c:8419) [0x5603f5336884]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #7  queue_foreach+0x163 (src/shared/queue.c:203) [0x5603f54334e3]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #8  can_read_data+0x597 (src/shared/mgmt.c:310) [0x5603f543e0c7]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #9  watch_callback+0x10e (src/shared/io-glib.c:162) [0x5603f549c98e]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #10 g_main_context_dispatch+0x193 (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3964dcf13]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #11 ?? (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3965330d9]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #12 g_main_loop_run+0x73 (/usr/lib/libglib-2.0.so.0.7000.3) [0x7fb3964dc463]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #13 mainloop_run+0x15 (src/shared/mainloop-glib.c:68) [0x5603f549e695]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #14 mainloop_run_with_signal+0x203 (src/shared/mainloop-notify.c:190) [0x5603f549efe3]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #15 main+0x1229 (src/main.c:1241) [0x5603f5138839]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #16 __libc_start_main+0xd5 (/usr/lib/libc-2.33.so) [0x7fb395820b25]
Feb 03 01:21:20 moarbits bluetoothd[734241]: #17 _start+0x2e (/usr/lib/bluetooth/bluetoothd) [0x5603f513b8de]
Feb 03 01:21:20 moarbits bluetoothd[734241]: +++++++++++++++++++++++++++
Feb 03 01:21:20 moarbits bluetoothd[734241]: src/adapter.c:store_link_key() Unable set contents for /var/lib/bluetooth/80:C5:F2:63:E4:94/F0:5C:77:CE:43:B3/info: ((null))
Feb 03 01:21:21 moarbits bluetoothd[734241]: AddressSanitizer:DEADLYSIGNAL
Feb 03 01:21:21 moarbits bluetoothd[734241]: =================================================================
Feb 03 01:21:21 moarbits bluetoothd[734241]: ==734241==ERROR: AddressSanitizer: SEGV on unknown address (pc 0x7fb3964f95c3 bp 0x60400002b790 sp 0x7ffcf667d9b8 T0)
Feb 03 01:21:21 moarbits bluetoothd[734241]: ==734241==The signal is caused by a READ memory access.
Feb 03 01:21:21 moarbits bluetoothd[734241]: ==734241==Hint: this fault was caused by a dereference of a high value address (see register values below).  Dissassemble the provided pc to learn which register was used.
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #0 0x7fb3964f95c3 in g_slist_last (/usr/lib/libglib-2.0.so.0+0x715c3)
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #1 0x7fb3964fdf84 in g_slist_append (/usr/lib/libglib-2.0.so.0+0x75f84)
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #2 0x5603f53b28e0 in update_record src/device.c:5133
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #3 0x5603f53b28e0 in update_bredr_services src/device.c:5219
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #4 0x5603f53b3622 in browse_cb src/device.c:5409
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #5 0x5603f52e89bf in search_completed_cb src/sdp-client.c:204
[... cut ...]
Feb 03 01:21:22 moarbits bluetoothd[734241]:     #15 0x5603f513b8dd in _start (/usr/lib/bluetooth/bluetoothd+0x4f98dd)
Feb 03 01:21:22 moarbits bluetoothd[734241]: AddressSanitizer can not provide additional info.
Feb 03 01:21:22 moarbits bluetoothd[734241]: SUMMARY: AddressSanitizer: SEGV (/usr/lib/libglib-2.0.so.0+0x715c3) in g_slist_last
Feb 03 01:21:22 moarbits bluetoothd[734241]: ==734241==ABORTING
Feb 03 01:21:22 moarbits systemd[1]: bluetooth.service: Main process exited, code=exited, status=1/FAILURE
Feb 03 01:21:22 moarbits systemd[1]: bluetooth.service: Failed with result 'exit-code'.
```

corrupt-stack, 2022-02-03:
I believe I've found the bug. Note the diff between tag 5.62 and 5.63 on the file src/adapter.c (showing only changes for the `store_link_key()` function for explanation purposes):

```
$ git diff 5.62..5.63 -- src/adapter.c
index 5846f0396..c49f42cfb 100644
--- a/src/adapter.c
+++ b/src/adapter.c
...similar changes in many places before store_link_key()...
@@ -8222,6 +8338,7 @@ static void store_link_key(struct btd_adapter *adapter,
        char device_addr[18];
        char filename[PATH_MAX];
        GKeyFile *key_file;
+       GError *gerr = NULL;
        gsize length = 0;
        char key_str[33];
        char *str;
@@ -8232,7 +8349,11 @@ static void store_link_key(struct btd_adapter *adapter,
        snprintf(filename, PATH_MAX, STORAGEDIR "/%s/%s/info",
                        btd_adapter_get_storage_dir(adapter), device_addr);
        key_file = g_key_file_new();
-       g_key_file_load_from_file(key_file, filename, 0, NULL);
+       if (!g_key_file_load_from_file(key_file, filename, 0, &gerr)) {
+               error("Unable to load key file from %s: (%s)", filename,
+                                                               gerr->message);
+               g_error_free(gerr);
+       }

        for (i = 0; i < 16; i++)
                sprintf(key_str + (i * 2), "%2.2X", key[i]);
@@ -8245,7 +8366,11 @@ static void store_link_key(struct btd_adapter *adapter,
        create_file(filename, 0600);

        str = g_key_file_to_data(key_file, &length, NULL);
-       g_file_set_contents(filename, str, length, NULL);
+       if (!g_file_set_contents(filename, str, length, &gerr)) {
+               error("Unable set contents for %s: (%s)", filename,
+                                                               gerr->message);
+               g_error_free(gerr);
+       }
        g_free(str);

        g_key_file_free(key_file);

...similar changes after store_link_key()...
```

GLib's assertion error in the logs points directly at the issue: `gerr` is correctly initialized to NULL at the top of the function. Then, it is used by `g_key_file_load_from_file()`, which is setting `gerr` to something not-NULL. The code uses `gerr->message` and then calls `g_error_free(gerr)` but does *not* also set `gerr = NULL` afterwards. This is important because `gerr` is then later re-used in the call to `g_file_set_contents()`, which sees `gerr` is not NULL and thus tries dereferencing the pointer to free'd memory.

Fix/Confirmation: `gerr` must be set back to NULL before it is re-used by the second glib call. I added `gerr = NULL` after every `g_error_free(gerr)` in my local build, and the crashes stopped. :-)

This code pattern occurs in many places throughout src/adapter.c. `git blame` indicates this was introduced by the following commit:

```
commit 2287c517ca1bde4a642ea2ddbed852bcf37e7bb9
Author: Tedd Ho-Jeong An <tedd.an@intel.com>
Date:   Mon Oct 18 10:28:26 2021 -0700

    adapter: Fix unchecked return value

    This patch fixes the unchecked return value(CWE-252) issues reported by
    the Coverity.
```

The issue is present in other files as well (e.g., in src/device.c). It seems someone needs to do a sweeping check throughout the code base for how GError variables are being used w.r.t. this pattern everywhere.

tedd-an, 2022-02-11:
I submitted the patches after reviewing all the related codes.