bluez/bluez #277: Support of bluetooth MIDI

EFeru, 2022-01-22:
Hello,

I followed the description from [here](https://neuma.studio/rpi-midi-complete.html) to create a Bluetooth MIDI server on Raspberry Pi 0 which uses an old fork https://github.com/oxesoft/bluez

But then I run into the following issue:
If I delete/forget the connection on the PC and try to reconnect. Then the PC (Windows10 based) has a hard time reconnecting. Sometimes it connects shortly, then disconnects, also the raspberry py bluetooth appears dissapears often from the list when I search for it.
Below you see the messages:
[image]

I believe the issue could be in the fork of https://github.com/oxesoft/bluez which is using a very old version (2017) from this official bluez repository.

I also tryied then to compile the the bluez version from here but then the build failed here, when running:
```c
./bootstrap
./configure --enable-midi --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var
make
sudo make install
```
[image]

Any suggestions on how to fix this issue? Any indications into where to look or inverstigate are helpful. Thanks in advance

EFeru, 2022-01-23:
Please also see this issue: https://github.com/Mylab6/PiBluetoothMidSetup/issues/3

I think is clear there are some robustness/stability issues. I am not sure if it is only with Raspberry Pi Zero or is a general issue.

Vudentz, 2022-01-25:
> [image]
> 
> Any suggestions on how to fix this issue? Any indications into where to look or inverstigate are helpful. Thanks in advance

Have you tried installing rst2man? Or disable the manpages with --disable-manpages. As for the connection instability we will probably need some HCI (e.g. use btmon) to check why it is disconnecting, but I suspect this is down in the link layer which is part of the controller firmware.

EFeru, 2022-01-25:
Hi @Vudentz, thank you for your reply. Regarding the connection instability, when you say "controller firmware" you mean the Bluetooth module firmware on Raspberry pi?

If yes, any idea what can we do there? Contact developers, check if there are new versions, try to re-flash etc.

Vudentz, 2022-01-25:
@EFeru Yes, you shall probably contact the manufacturer and include HCI traces with the disconnect reason. You can also try using a USB controller to check if that has better interroperability.

closed 2022-03-29 by EFeru